stereolabs/zed-ros2-wrapper #1: [RTPS_WRITER Error] Cannot add RTPS submesage to the CDRMessage. Buffer too small -> Function insert_submessage

YuehChuan, 2019-04-19:
Thanks for being the pioneer make ros2 example available on internet.
some issue I tested on cuda10.0  nvidia-410

here is the report 
https://hackmd.io/s/rkRj5iG5N
1. missing dependency image-transport
2. the point cloud freeze after start 2 seconds, recieved frames are not stable
3. roslaunch cannot do nested launch
4. [RTPS_WRITER Error] Cannot add RTPS submesage to the CDRMessage. Buffer too small -> Function insert_submessage

wish someone get what I mean.

Myzhar, 2019-04-19:
Hi @YuehChuan 
thank you for the full report, the ROS2 wrapper is still a beta version so the more we get feedback, the more it will be improved.

1. You can install `image_transport` with the following command:
`$ sudo apt install ros-crystal-image-transport`
as you correctly reported.
2. If you speak about "RVIZ2" it's a known problem and it is reported in the "Known Issues". RVIZ2 is still missing `message_filters` implementation, so topics synchronization is not working very well. We hope that Rviz2 will get `message_filters` support soon.
[For more info see [this issue](https://github.com/ros2/rviz/issues/359) and this [pull request](https://github.com/ros2/rviz/pull/375)]
3. I'm going to investigate and fix the `ros2 launch` problem.
4. We never got that error message. Are you using a not default configuration for RTPS middleware?

YuehChuan, 2019-04-19:
@Myzhar  
thanks for your quick response 
4. using the default RTPS middleware

closed 2019-04-19 by YuehChuan