stereolabs/zed-ros2-wrapper #2: Dashing Support

clungzta, 2019-08-27:
Hi,

I see that this package supports Crystal at this stage.

I did make an attempt at building it on my Dashing system. I successfully got through  most of the build steps, until:
`colcon build --symlink-install --packages-select stereolabs_zed --cmake-args=-DCMAKE_BUILD_TYPE=Release`

Where the following issue occurred:
```
Starting >>> stereolabs_zed
--- stderr: stereolabs_zed                         
CMake Error at CMakeLists.txt:155 (rclcpp_components_register_nodes):
  Unknown CMake command "rclcpp_components_register_nodes".

---
Failed   <<< stereolabs_zed	[ Exited with code 1 ]
```

Is Dashing support planned?

Best Regards

Myzhar, 2019-08-27:
Hi @clungzta 
Porting the ZED node to Dashing requires a lot of work since it introduced a lot of novelties. 
We are working on it and it will be ready soon.

clungzta, 2019-08-27:
Hi

Thank you for the status update and active development.
I look forward to using it upon release.

Best Regards

estefaniatc, 2019-11-12:
Hi,

Is there any update on the expected support for Dashing?

EunSungYang, 2020-01-15:
@Myzhar 
Hi.

I have a same issue.
Can you tell me when the problem will be solved?

mhaboali, 2020-04-29:
Any updates on the Dashing release?

Thanks!

cardboardcode, 2020-05-04:
@Myzhar Thank you for the ROS2 wrapper of Zed Camera.

Same question as the rest. Any updates on the Dashing or later releases?

Myzhar, 2020-05-04:
We are going to work to release an update for the new ROS2 LTS (Foxy) that will be released in May:
https://index.ros.org/doc/ros2/Releases/Release-Foxy-Fitzroy/

james-yoo, 2020-05-07:
@Myzhar SDK update for the ROS2 Foxy will be compatible Dashing as well?

Myzhar, 2020-05-08:
@james-yoo I must verify what's different between the two versions

mehmetkillioglu, 2020-07-22:
> We are going to work to release an update for the new ROS2 LTS (Foxy) that will be released in May:
> https://index.ros.org/doc/ros2/Releases/Release-Foxy-Fitzroy/

Hi,

Any updates for Foxy release? 

Best,
Mehmet

Myzhar, 2020-07-22:
Work in progress 
Lot of changes and many quality checks to do.
It will take yet a bit of time before the final release

annb3, 2020-09-30:
Hi, 
Any updates for Dashing release?

Best regards,
Anna

Myzhar, 2020-09-30:
No Dashing release, Eloquent release is instead almost ready:
https://github.com/stereolabs/zed-ros2-wrapper/tree/eloquent

mehmetkillioglu, 2020-09-30:
> Hi,
> Any updates for Dashing release?
> 
> Best regards,
> Anna

Hi,

From this [repository](https://github.com/mehmetkillioglu/zed-ros2-wrapper), you can find my custom made version of zed wrapper for Dashing. The tests were not done thoroughly, and I cannot claim it will work fine as I am not affiliated with Stereolabs. I do not accept any liability. Might not be suitable for production-level applications.

Although if you use and find some problems with it, you can create an issue on my repository though. I can try to fix it.

Best,
Mehmet

Myzhar, 2020-09-30:
FYI the `eloquent` version of the ZED ROS2 wrapper will not be based on LifeCycle anymore, but it will be based on the "straight" `Node` model.
This is because the LifeCycle model is not yet totally supported by ROS2 and many packages and nodes like Image Transport and TF do not handle it yet correctly.

Myzhar, 2020-10-29:
Eloquent version has been released

closed 2020-10-29 by Myzhar

Rayman, 2021-03-16:
@Myzhar can you elaborate a little more about the issues you encountered with LifeCycle? What exactly is not supported? I'm just getting into ROS2 and I thought LifeCycle was a big improvement.

Myzhar, 2021-03-16:
Hi @Rayman 
Lifecycle is really interesting, but the many important features for us were not available, mainly `image_transport` and `TF broadcasting`, but there were problems also with other features.
As far as I know `image_transport` is not yet available with Lifecycle nodes even with ROS2 Foxy

Rayman, 2021-03-16:
@Myzhar thank you!

Strange that this is still not supported. I'm linking the issue here for future reference:
https://github.com/ros-perception/image_common/issues/108
https://github.com/ros-perception/image_common/pull/167