stereolabs/zed-ros2-wrapper #5: Jetson Nano shuts off when ZED camera plugged in

calvertdw, 2020-01-03:
If the ZED camera is plugged into the Jetson when I power on, I can see the system loading but around when the lock screen would show up the system crashes.

Similarly, if I start the Jetson without the ZED plugged in, I can use the Jetson normally until I plug in the ZED camera, in which the Jetson hard crashes like I pulled the power plug.

Any ideas how I would debug this sort of thing? Are there any special steps required for Jetson operation? Maybe I should reflash the Jetson SD card and try again.

jacobculley, 2020-01-03:
First thing I would check is the Jetson's power supply, if you are using usb-c maybe try the barrel jack? Sounds like the Jetson and the camera combined is drawing too many amps.

Another option that probably isn't recommended is using a powered usb 3 hub, that should take the load of the camera off the Jetson.

Myzhar, 2020-01-03:
Hi @calvertdw,
What @jacobculley said is correct.
The problem you are describing seems related to a  power issue.
Another good solution is using the two pins on the Jetson Nano expansion header to provide more power.
I suggest you to read this good post on Jetsonhack blog:
https://www.jetsonhacks.com/2019/04/10/jetson-nano-use-more-power/

closed 2020-10-29 by Myzhar