stereolabs/zed-ros2-wrapper #7: Could not find a package configuration file provided by "image_transport"

Demonaxx, 2020-04-27:
i am getting  one error when i execute the following command to install the zed-ros2-wrapper into my work space of ros2 dashing:

"colcon build --symlink-install --packages-select stereolabs_zed --cmake-args=-DCMAKE_BUILD_TYPE=Release"

error: 
Starting >>> stereolabs_zed
[Processing: stereolabs_zed]                         
--- stderr: stereolabs_zed                           
CMake Error at CMakeLists.txt:58 (find_package):
  By not providing "Findimage_transport.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "image_transport", but CMake did not find one.

  Could not find a package configuration file provided by "image_transport"
  with any of the following names:

    image_transportConfig.cmake
    image_transport-config.cmake

  Add the installation prefix of "image_transport" to CMAKE_PREFIX_PATH or
  set "image_transport_DIR" to a directory containing one of the above files.
  If "image_transport" provides a separate development package or SDK, be
  sure it has been installed.

i try to source  the setup file(local and main) but don't work.

Myzhar, 2020-04-27:
Hi @Demonaxx 
have you tried to install the `image-transport` pakcage?
`$ sudo apt install ros-dashing-image-transport ros-dashing-image-transport-plugins`

Demonaxx, 2020-04-27:
Hi @Myzhar 
I install the ros-dashing-image-transport package as you say, but the error persists. it some possibility that a have wrong configure the CMAKE_MODULE_PATH ?

Myzhar, 2020-04-27:
Are you compiling it inside QtCreator or similar IDE?

Demonaxx, 2020-04-27:
I just used the  shell of linux

Demonaxx, 2020-04-27:
i think, that the problem is that this two files
    image_transportConfig.cmake
    image_transport-config.cmake

are not recognized  but they are in the /opt/ros/dashing/share/image_transport/cmake

[image]

Myzhar, 2020-04-27:
Try to clean the CMake cache deleting the build and the devel folder in the catkin workspace and try to compile again.
Have you correctly configured your environment with the ROS2 paths?

Demonaxx, 2020-04-27:
i think that, the problem was ros2 install from source get me a bug in the opt/ros/dashing folder ,i install ros2 from Binary packages . but the command give me other error. 
[image]

Myzhar, 2020-04-27:
Have you installed the Dashing binaries? The ZED2 wrapper is not compatible with newer ROS2 versions

Demonaxx, 2020-04-27:
the zed wrapper work  with ROS 2 Eloquent Elusor binary
?

Demonaxx, 2020-04-27:
ok  i going to try install cystal binaries

Myzhar, 2020-04-27:
Ss you can read in the README file, the ROS2 wrapper is designed for ROS2 Crystal.
A lot of changes have been made in ROS2 since the Crystal version.
We will update the ROS2 wrapper with the next ROS2 LTS version that will be released in May:
https://index.ros.org/doc/ros2/Releases/Release-Foxy-Fitzroy/

Demonaxx, 2020-04-27:
ok thx

closed 2020-04-27 by Demonaxx

fairuskhalid, 2020-07-17:
locate your image transport: find . -name image_transport*
in my case:
ubuntu@ubuntu-OptiPlex-7070:/opt/ros/eloquent$ find . -name image_transport*
./share/image_transport
./share/image_transport/cmake/image_transportConfig.cmake
./share/image_transport/cmake/image_transportConfig-version.cmake

add path into CMAKE_PREFIX_PATH:
export CMAKE_PREFIX_PATH=$CMAKE_PREFIX_PATH:/opt/ros/eloquent/share

Myzhar, 2020-07-17:
Hi @fairuskhalid 
please do not reply to a closed issue. Opening a new one is always better to keep tracks of user problems.

However, as I said before, the ZED-ROS2-WRAPPER only supports ROS2 Crystal, we are working to support the new ROS2 release and a new version of the wrapper will be ready soon.