stereolabs/zed-ros2-wrapper #12: ROS Eloquent - ZED_compoenet build error

harisiva1, 2020-09-23:
[image]
Hello,
I get an std error(please see the attached image) for zed_component while trying to build the package in ros eloquent.Could you please support?

Also i don't see any node or launch files to view camera output in rviz. When is it expected to release here / any release planned?

Thanks in advance !

Regards,
Hari.s

Myzhar, 2020-09-23:
Hi @harisiva1 
the Eloquent branch is under active development and it has not yet been officially released in a stable version

However your problem seems related to CUDA that has not been correctly installed in your system.

About RVIZ, currently we have not yet added a launch file for it because we are adding new features continuously and we are simply manually starting RVIZ using the `rviz2` command and we are selecting to view only the topic that are under testing.

harisiva1, 2020-09-23:
Hi,
Thanks for your quick reply.
yeah it was a problem with CUDA, "libnppicom.so" seems to be missing in CUDA 11 for Ubuntu 18.

So i tried with CUDA10.2  and corresponding ZDK version in ros eloquent.It was build successfully however with many warnings and atlast stderr output(zed_compoenent).
Please see the attached image.

I am waiting for the stable version release for eloquent.
Thanks for the update!

[image]
[image]

Myzhar, 2020-09-23:
OK, you can ignore those warnings. If you compile forcing the Release mode they will disappear:
`colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release`

The main functionalities of the node are working: image,depth,pointcloud,positional tracking, TF, dynamic parameters.
We are currently working to add sparial mapping and object detection support

harisiva1, 2020-10-09:
hello,
i am unable to find and run nodes which gives image,depth and point cloud etc., in eloquent.can you support me?

Myzhar, 2020-10-09:
The release for ROS2 Eloquent has not yet been merged into `master` branch. 
It is yet under development and you can get latest version checking out the `eloquent` branch and recompiling. 
As it is yet under active development it may change daily before the merging into `master`.

Myzhar, 2020-10-09:
To start the node you can follow the instructions in the README file:
https://github.com/stereolabs/zed-ros2-wrapper/tree/eloquent#starting-the-zed-node

harisiva1, 2020-10-10:
i checked out eloquent branch but i dont see "zed_wrapper" package in my system.so i am not launch zed launch file.
are there some command line to sync it?

Myzhar, 2020-10-10:
Have you followed the installation instruction in the README?
https://github.com/stereolabs/zed-ros2-wrapper/tree/eloquent#build-the-package

I add here the steps to clone the `eloquent` branch:
```
$ cd ~/ros2_ws/src/ #use your current ros2 workspace folder
$ git clone https://github.com/stereolabs/zed-ros2-wrapper.git
$ cd zed-ros2-wrapper
$ git checkout eloquent
$ cd ../..
$ rosdep install --from-paths src --ignore-src -r -y
$ colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release
$ echo source $(pwd)/install/local_setup.bash >> ~/.bashrc
$ source ~/.bashrc
```

harisiva1, 2020-10-10:
git pull resolved my previous issue.
now i am able to launch the file but i don't see the topics like rect_image_raw ,cloud_point registered etc., in eloquent..so it is still under development right?

Myzhar, 2020-10-10:
Many topic names are changes to match the names used in ROS1.
You can see the full list of available topics with the command:
`$ ros2 topic list` or using RQT

harisiva1, 2020-10-10:
yes i have looked at all the topic earlier using the cmd as you said above and neither zed is turning on(no light in zed camera) nor i can see the image(camera output,depth image,confidence image etc.,) in rviz after i launch the launch file and zed node.

Myzhar, 2020-10-10:
You can use rqt or rviz2 to view topics

[image]

harisiva1, 2020-10-10:
are the above images from rviz in **eloquent**?

Myzhar, 2020-10-10:
They are

harisiva1, 2020-10-10:
please support me..
[image]

Myzhar, 2020-10-10:
You cannot find any topic because the node is not running. 
If you read carefully the console log you can see that the lib file `libnvcuvid.so.1` is missing. This means that you have not correctly installed CUDA on your machine. 
Can you run the zed tools? Try for example `ZED_Depthviewer`

Myzhar, 2020-10-29:
@harisiva1 can you close this issue if you solved it?

harisiva1, 2020-11-04:
hi,
sorry for the late reply!
it has already worked me weeks ago ,so you can consider this issue as closed

closed 2020-11-04 by harisiva1

harisiva1, 2020-11-21:
Hi,
Which topic do i need to choose to create mapping in rviz?