stereolabs/zed-ros2-wrapper #13: about reconfiguring parameters dynamically

fukaura-jin, 2020-09-24:
hello, Can we reconfigure only the following 6 parameters directly during the execution of the node?

i want to reconfigure >**video.qos_reliability** and >**video.qos_history** directly during the execution of the node.
I would appreciate it if you could get back to me.

`>**general.mat_resize_factor:** Sets the scale factor of the output images and depth map. Note that the camera will acquire data at the dimension set by the resolution parameter; images are resized before being sent to the user
>**video.auto_exposure:** Enables/disables automatic gain and exposure
>**video.exposure:** Sets camera exposure only if auto_exposure is false
>**video.gain:** Set camera gain only if auto_exposure is false
>**depth.confidence:** Sets a threshold that filters the values of the depth or the point cloud. With a confidence threshold set to 100, all depth values will be written in the depth and the point cloud. This is set to 80 by default, which removes the least accurate values.
>**depth.max_depth:** Sets the maximum depth range`

Myzhar, 2020-09-24:
Hi @fukaura-jin 
we are working on a new version of the wrapper in the branch `eloquent`.
The version of the wrapper for ROS2 Crystal will be soon replaced.

Sincerely I'm not sure that putting QoS parameters in the list of the runtime modificable parameters will be a good choice, but I'm going to think about it.

You can retrieve the list of current dynamic parameters looking at the YAML file:
https://github.com/stereolabs/zed-ros2-wrapper/blob/eloquent/zed_wrapper/config/common.yaml

The branch is under heavy development and many changing are push in these days

fukaura-jin, 2020-09-24:
Thank you for your quick response!

I thought that **video.qos_reliability** and **video.qos_history** were the runtime modificable parameters because I got **successful** when I modified them during the execution of the node.

Myzhar, 2020-09-25:
That's strange because they were not handled in the parameter changing callback for Crystal:
https://github.com/stereolabs/zed-ros2-wrapper/blob/master/zed_wrapper/src/component/src/zed_component.cpp#L1373-L1516

fukaura-jin, 2020-09-25:
Thank you for your response!
I don't know why I got **successful**.

I try to do QoS control with zedm (ROS2) in my research.
Then, I need to modify **video.qos_** parameters during the execution of the node.

Is there any good way for this situation?
I need your help.

Myzhar, 2020-09-25:
The callback must be modified in order to handle this changing runtime.

Furthermore the publisher of the messages must be deinitialized and re-initialized again with the new QoS

Myzhar, 2020-10-29:
This feature will not be added to the ROS2 wrapper because of security issues.
I suggest you modify the node to add it

closed 2020-10-29 by Myzhar