stereolabs/zed-ros2-wrapper #15: Does zed-ros2-wrapper support ROS 2 Foxy ?

neo-knight-td, 2020-10-04:
Hello,

I am new to ROS. I have installer ROS 2 Foxy on my Ubuntu 20.04 machine and I would like to use a ZEDM stereo camera to perform SLAM. However, from what I have read, the zed-ros2-wrapper is only compatible with Eloquent and Crystal. 

Is this information verified ? Are there new releases coming soon to use ZED cameras on Ubuntu 20.04 ? 
Is there another way to use ROS and ZED on Ubuntu 20.04 ? 

Thank You !

Myzhar, 2020-10-04:
Hi @neo-knight-td 
the current `master` branchonly supports Crystal.
n the net days we are releasing a new version of the wrapper with support for ROS2 Eloquent that is based on Ubuntu 18.04 that is currently the most diffused Ubuntu release (see Nvidia Jetson boards).
To use the Eloquent version on your Ubuntu 20.04 system you can use Docker

Myzhar, 2020-10-29:
Eloquent support has been released

commit 85282e14 (2024-01-29): Merge branch '15-add-optional_opencv_calibration_file-support' into 'master'

Resolve "Add `optional_opencv_calibration_file` support"

Closes #15

See merge request sl/zed-ros2-wrapper!44
closed 2020-10-29 by Myzhar