stereolabs/zed-ros2-wrapper #17: [BUG] Failed to visualize all transforms in bag recording

laszloturanyi, 2020-11-04:
**Describe the bug**
[Eloquent] When replaying zed node data from rosbag in tf view, only odom, base_link, map frames are available, the following frames are not:
- zed2_camera_center
- zed2_left_camera_frame
- zed2_left_camera_optical_frame
- zed2_right_camera_frame
- zed2_right_cmera_optical_frame
- zed2_imu_link (it is marked as latched in info log message at startup, I don't know how it is achieved)

**Device information:**
 - OS: Ubuntu
 - OS version: 18.04
 - CPU: Intel(R) Core(TM) i7-9750H CPU @ 2.60GHz
 - GPU Nvidia:  GeForce GTX 1650
 - ZED SDK Version: 3.3.0
 - ROS Wrapper version: latest master

**To Reproduce**
Steps to reproduce the behavior:
1. Start zed_node with zed2 config (mapping, publishing odom and base link is turned on).
```
ros2 launch zed_wrapper zed2.launch.py
```
2. Start rviz2 and display tf frames.
You can see all the frames (including zed2_* frames showing up on display)
3. Record data with rosbag2
```
ros2 bag record -a -o zed_test_bag
```
4. Stop rosbag recording (with ctrl+c), zed_node, and rvzi2
 
5. Now try to replay data, start rviz2 first.

6. Start rosbag replay
```
ros2 bag play zed_test_bag
```

**Expected behavior**
Here I expect to see base_link, map, odom, and zed2_frames showing but I cannot see zed2_frames nor in the frame listing neither in the Tree. Is there any guess what coud be the problem.

However, If I run robot state publisher manually, while replaying the bag all other frames appears, too.

Step 7
```
 ros2 run robot_state_publisher robot_state_publisher <ros_ws>/src/zed-ros2-wrapper/zed_wrapper/urdf/zed2.urdf
```

**Additional context**
- There seems to be a problem (and recent changes) with robot_state_publisher writing (or not writing) / before frame names, though it should not be a problem because when running the zed node it works the same way.
- robot_description is not contained in the rosbag file (which is published only once) but I run rviz2 after starting the zed node, it shows nicely all the frames so I assume it is not necessary for build the tree.

Myzhar, 2020-11-04:
The static frames (`zed2_*`) are published by the `robot_state_publisher ` node when you use the Python launch file.
Have you checked on ROS community websites if there is a known issue with `robot_state_publisher` and bags files?

laszloturanyi, 2020-11-04:
Yes I have checked that and I have filled in the **Additional context** section with the info I could found and appeared to be somewhat related.

Could you reproduce this error?

Myzhar, 2020-11-04:
I have not tried yet, but I suppose this is a ROS2 bug more than a "ZED ROS2 Wrapper" bug and the best place to search for a solution is on ROS answers website

laszloturanyi, 2020-11-04:
Yes, I have submitted this issue on ROS answers, as well.

Myzhar, 2020-11-04:
Good, please link the post here so we can keep track of it :+1:

laszloturanyi, 2020-11-05:
Here you are:
[ROS Answers: Failed to visualize all transforms in bag recording](https://answers.ros.org/question/364761/failed-to-visualize-all-transforms-in-bag-recording/)

laszloturanyi, 2020-11-09:
Hi,

Unfortunately no one has answered this question so far...
Big up for the SVO recording though, it is really handy to use it. We also have some ROS issues with it (cannot topic echo when using playback) but otherwise it works fine.

If we could find a simple solution to record and sync data during playback from other sensors we could drop rosbag once and for all.

Myzhar, 2020-11-09:
There was a problem in ROS1 with SVO and simulated timestamp. I must check if in ROS2 the problem has been overcome.
I will update this issue with news about it as soon as I verify it.