stereolabs/zed-ros2-wrapper #18: Could NOT find CUDA (missing: CUDA_CUDART_LIBRARY) (found version "10.2") [BUG]

viciopoli01, 2020-11-10:
**Describe the bug**
Running     `colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release`
 on a Docker image based on Jetpack 4.4 on Jetson Xavier NX. I got the following error

```
Starting >>> zed_components
--- stderr: zed_components
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find CUDA (missing: CUDA_CUDART_LIBRARY) (found version "10.2")
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.10/Modules/FindCUDA.cmake:1080 (find_package_handle_standard_args)
  CMakeLists.txt:46 (find_package)
---
Failed   <<< zed_components [5.19s, exited with code 1]
```

**Device information:**
 - OS: Ubuntu
 - OS 18.04
 - CPU:  Jeston Xavier NX
 - GPU Nvidia:  Volta 
 - ZED SDK Version: 3.3
 - ROS Wrapper version: eloquent

**To Reproduce**
Steps to reproduce the behavior:

1. Create a docker image based on `nvcr.io/nvidia/l4t-base:r32.4.4`
2. Install the dependencies for the SDK.
3. Install the ZED SDK 3.3:

```
RUN apt-get update -y && apt-get install --no-install-recommends lsb-release wget less udev sudo apt-transport-https -y && \
    echo "# R32 (release), REVISION: 4" > /etc/nv_tegra_release ; \
    wget -q --no-check-certificate -O ZED_SDK_Linux_JP.run https://download.stereolabs.com/zedsdk/3.3/jp44/jetsons && \
    chmod +x ZED_SDK_Linux_JP.run ; ./ZED_SDK_Linux_JP.run silent && \
    rm -rf /usr/local/zed/resources/* \
    rm -rf ZED_SDK_Linux_JP.run && \
    rm -rf /var/lib/apt/lists/*

#This symbolic link is needed to use the streaming features on Jetson inside a container
RUN ln -sf /usr/lib/aarch64-linux-gnu/tegra/libv4l2.so.0 /usr/lib/aarch64-linux-gnu/libv4l2.so

```
4.
 ```
RUN cd src ; git clone --branch eloquent https://github.com/stereolabs/zed-ros2-wrapper.git && \
    cd .. && \
    . /opt/ros/$ROS2_DISTRO/setup.sh && \
    rosdep install --from-paths src --ignore-src -r -y && \
    colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release
```
5. Build the Docker image.

**Expected behavior**
I expect to have the wrapper package building correctly.

Myzhar, 2020-11-10:
Hi @viciopoli01 

You should link to the stubs and not the actual libs, since during the image creation, especially on jetson, the cuda libs are not mounted.
See here https://github.com/stereolabs/zed-docker#compiling-issues

Maybe you need to change the library path using `--cmake-args`

viciopoli01, 2020-11-11:
Dear @Myzhar I tried passing the arguments `--cmake-args " -DCMAKE_BUILD_TYPE=Release" " -DCMAKE_LIBRARY_PATH=/usr/local/cuda/lib64/stubs" " -DCUDA_CUDART_LIBRARY=/usr/local/cuda/lib64/stubs" " -DCMAKE_CXX_FLAGS='-Wl,--allow-shlib-undefined'"`

Anyway I end up with the follwoing:

```
--- stderr: zed_interfaces
CMake Warning:
  Manually-specified variables were not used by the project:

    CUDA_CUDART_LIBRARY

---
```
And also :

```
--- stderr: zed_components
WARNING: Target "zed_camera_component" requests linking to directory "/usr/local/cuda/lib64/stubs".  Targets may link only to libraries.  CMake is dropping the item.
/yager/ros2_zed_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp: In member function ‘bool stereolabs::ZedCamera::startObjDetect()’:
/yager/ros2_zed_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:2262:10: error: ‘struct sl::ObjectDetectionParameters’ has no member named ‘enable_body_fitting’; did you mean ‘enable_tracking’?
     od_p.enable_body_fitting = mBodyFitting;
          ^~~~~~~~~~~~~~~~~~~
          enable_tracking
/yager/ros2_zed_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:2272:51: error: ‘BAG’ is not a member of ‘sl::OBJECT_CLASS’
         mObjDetFilter.push_back(sl::OBJECT_CLASS::BAG);
                                                   ^~~
/yager/ros2_zed_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:2275:51: error: ‘ANIMAL’ is not a member of ‘sl::OBJECT_CLASS’
         mObjDetFilter.push_back(sl::OBJECT_CLASS::ANIMAL);
                                                   ^~~~~~
/yager/ros2_zed_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:2278:51: error: ‘ELECTRONICS’ is not a member of ‘sl::OBJECT_CLASS’
         mObjDetFilter.push_back(sl::OBJECT_CLASS::ELECTRONICS);
                                                   ^~~~~~~~~~~
/yager/ros2_zed_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:2281:51: error: ‘FRUIT_VEGETABLE’ is not a member of ‘sl::OBJECT_CLASS’
         mObjDetFilter.push_back(sl::OBJECT_CLASS::FRUIT_VEGETABLE);
                                                   ^~~~~~~~~~~~~~~
/yager/ros2_zed_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp: In member function ‘void stereolabs::ZedCamera::processDetectedObjects(rclcpp::Time)’:
/yager/ros2_zed_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:3968:51: error: ‘BAG’ is not a member of ‘sl::OBJECT_CLASS’
         mObjDetFilter.push_back(sl::OBJECT_CLASS::BAG);
                                                   ^~~
/yager/ros2_zed_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:3971:51: error: ‘ANIMAL’ is not a member of ‘sl::OBJECT_CLASS’
         mObjDetFilter.push_back(sl::OBJECT_CLASS::ANIMAL);
                                                   ^~~~~~
/yager/ros2_zed_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:3974:51: error: ‘ELECTRONICS’ is not a member of ‘sl::OBJECT_CLASS’
         mObjDetFilter.push_back(sl::OBJECT_CLASS::ELECTRONICS);
                                                   ^~~~~~~~~~~
/yager/ros2_zed_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:3977:51: error: ‘FRUIT_VEGETABLE’ is not a member of ‘sl::OBJECT_CLASS’
         mObjDetFilter.push_back(sl::OBJECT_CLASS::FRUIT_VEGETABLE);
                                                   ^~~~~~~~~~~~~~~
/yager/ros2_zed_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:4017:59: error: ‘class sl::ObjectData’ has no member named ‘sublabel’; did you mean ‘label’?
         objMsg->objects[idx].sublabel = sl::toString(data.sublabel).c_str();
                                                           ^~~~~~~~
                                                           label
make[2]: *** [CMakeFiles/zed_camera_component.dir/src/zed_camera/src/zed_camera_component.cpp.o] Error 1
make[1]: *** [CMakeFiles/zed_camera_component.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< zed_components [30.5s, exited with code 2]
```

Myzhar, 2020-11-11:
@viciopoli01 are you using ZED SDK v3.3?
The new ROS2 wrapper is built on top of SDK v3.3. It does not support older SDKs.

viciopoli01, 2020-11-11:
@Myzhar yes, you are right, I tried there the SDK v3.2 . Trying with the v3.3 it gives also errors. 

One related with `stereolabs_zed: Cannot locate rosdep definition for [message_runtime]`
And anotherone related with `stereolabs_zed` 

```
--- stderr: stereolabs_zed
CMake Error at CMakeLists.txt:59 (find_package):
  By not providing "Findstereolabs_zed_interfaces.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "stereolabs_zed_interfaces", but CMake did not find one.

  Could not find a package configuration file provided by
  "stereolabs_zed_interfaces" with any of the following names:

    stereolabs_zed_interfacesConfig.cmake
    stereolabs_zed_interfaces-config.cmake

  Add the installation prefix of "stereolabs_zed_interfaces" to
  CMAKE_PREFIX_PATH or set "stereolabs_zed_interfaces_DIR" to a directory
  containing one of the above files.  If "stereolabs_zed_interfaces" provides
  a separate development package or SDK, be sure it has been installed.

---
Failed   <<< stereolabs_zed [7.61s, exited with code 1]
```
The SDK is installed indeed I see the log 

```
Dependencies installation complete

ZED SDK installation complete !
```

Myzhar, 2020-11-11:
Are you using ROS2 Eloquent?

Myzhar, 2020-11-11:
Furthermore `stereolabs_zed_interfaces` is the old name for the interface package. Have you cleaned your workspace after updating your repository?

Try the following commands in the root of your ROS2 workspace
```
rm -rf install
rm -rf build
rm -rf log
```
then recompile

viciopoli01, 2020-11-11:
@Myzhar yes I am using ros2 Eloquent, for this reason I switch to the branch `sdk_v3_and_eloquent` .
Regarding the ros2 ws it is empty at the beginning (I do: `RUN mkdir -p $ROS2_ZED_WS/src` in the Dockerfile), since I am trying to have a separate ws for the ZED.

viciopoli01, 2020-11-11:
> Furthermore `stereolabs_zed_interfaces` is the old name for the interface package. Have you cleaned your workspace after updating your repository?
> 
> Try the following commands in the root of your ROS2 workspace
> 
> ```
> rm -rf install
> rm -rf build
> rm -rf log
> ```
> 
> then recompile

Compiling gives again the same error here

Myzhar, 2020-11-11:
Are you using the same console window or have you closed it before recompiling?
Maybe that some environmental variables keep the old paths

viciopoli01, 2020-11-11:
I am putting together https://github.com/stereolabs/zed-docker/tree/master/3.X/jetpack_4.X/devel this image and the ros1 one but intead of the ros1 wrapper I am installing ROS2

viciopoli01, 2020-11-11:
> Are you using the same console window or have you closed it before recompiling?
> Maybe that some environmental variables keep the old paths

I don't have old path of regarding ZED since each time I build it is like the first one.

Myzhar, 2020-11-11:
You are doing something wrong. `stereolabs_zed` does not exist in the zed-ros2-wrapper repository.
It's an old name used in the previous Crystal release.
The same is for `stereolabs_zed_interfaces`.

viciopoli01, 2020-11-11:
Now it is working, I had to source all the ROS2 ws I had.

closed 2020-11-11 by viciopoli01

Myzhar, 2020-11-11:
Good. I suggest you to use the master branch. The `eloquent` branch was created for development and has just been merged into the `master` branch. It will be deleted

harisiva1, 2020-11-22:
hi,
i am using master branch now but  i am getting the same error.and i am not able to build it.
please support!
[image]

Myzhar, 2020-11-22:
Hi @harisiva1 
I think this is a different issue unless you are using Docker. Please open a new ticket if you are not using docker