stereolabs/zed-ros2-wrapper #20: Could NOT find CUDA (missing: CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY)

harisiva1, 2020-11-22:
Hello,
I am using ROS2 ELOQUENT and i am trying to build red-ros2-wrapper and i get the below error.attached the image.
I have tried with both master branch and eloquent branch but i get the same error.
Please support!
[image]

harisiva1, 2020-11-23:
Hi,
i have tried few things so far but still get the same error.

first try:https://github.com/torch/torch7/issues/1138
 secondly i tried to install cuda10.2 once again but the local(run) file throws a error saying that install of driver component failed like here. https://forums.developer.nvidia.com/t/error-while-install-cuda-toolkit-10-2-on-ubuntu-18-04-3/109829/3
i could able to install only deb(local)
do you have some suggestions to overcome this error??
i have also tried to run zed package in ros melodic still ended up with this cuda error
[image]

Myzhar, 2020-11-23:
Hi @harisiva1 
can you post the output of the command `nvcc -V`?

harisiva1, 2020-11-23:
[image]

Myzhar, 2020-11-23:
@harisiva1 as you can see this is your problem.
Your system sees CUDA 9.1 as default, not CUDA 10.2.
I suggest you clean up your system purging CUDA and installing again CUDA 10.2.
Here you can find a bunch of valid methods to perform this task:
https://askubuntu.com/questions/530043/removing-nvidia-cuda-toolkit-and-installing-new-one

harisiva1, 2020-11-23:
[image]
but i also see cuda version 11 when i run nvidia-smi command

Myzhar, 2020-11-23:
So your system is definitively in a chaotic configuration.
At this point, I suggest you remove also the Nvidia driver and install everything from scratch.

harisiva1, 2020-11-23:
you can consider this issue as closed.
with the current configuration i have been able to run ZED launch file in both ros melodic and eloquent and perform spatial mapping.Nevertheless i will try to do as per your suggestion.
thanks!

closed 2020-11-23 by Myzhar