stereolabs/zed-ros2-wrapper #22: Jetson nano low frame rate in ROS2, verified OK in ROS1

steveanthonyjones, 2020-11-30:
Hello!
ROS2 zed2 camera produces low camera frame rates (5-8Hz) when published as ROS2 node (lower than publish setting in YAML file, various tried >10Hz) despite no CPU or GPU limitations. GPU frequency reported does oscillate between approximately 5% and 70% between reports (produced by tegrastats). Performance was verified with a fresh system installation with ROS(1), no problems at 15Hz matching YAML settings.

All software latest at time of writing.

**Device information:**
 - OS: [Ubuntu]
 - OS version: [18.04]
 - CPU: [Jeston Nano]
 - ZED SDK Version: [3.3.1]
 - ROS Wrapper version: [latest as per "getting started" instructions]
- ROS versions: ROS2 Eloquent, secondary verification on ROS Melodic (works fine)

Attached:
-Screen shot of node Hz publish frequencies and tegrastats report (ROS2)
-Ros2 doctor report
-YAML files (common, ZED2) - for reference

Thanks for your time

[image]

[ROS2 doctor report - ZED2 camera in operation.txt](https://github.com/stereolabs/zed-ros2-wrapper/files/5613339/ROS2.doctor.report.-.ZED2.camera.in.operation.txt)

[YAML.zip](https://github.com/stereolabs/zed-ros2-wrapper/files/5613346/YAML.zip)

Myzhar, 2020-11-30:
Hi @steveanthonyjones 
have you tried reducing the grab frequency to 15 FPS? The Jetson Nano is not a powerful embedded board and 30FPS can be too much for it to support also the ROS2 infrastructure.

steveanthonyjones, 2020-12-02:
Hi @Myzhar, thanks for your reply. Strangely reducing the grab frame rate to 15Hz (with publish rate also 15Hz) produces a node publish rate of approximately 0.25Hz (recall that a grab of 30, publish of 15 produced approximately 7Hz previously in ROS2). Also tested was grab:20, publish:10, node result:0.4). Curiously, reinstating the original values (and restarting the node) resulted in a publish rate of 0.4Hz (it did not return to 7Hz).  Rebuilding the wrapper resulted in a return to 7Hz.
I was concerned with the hardware limitations also, I tested the ZED2 in ROS1 on a clean Jetson/Ubuntu install and achieved 15Hz which in my mind eliminated the hardware as a limiting factor. Any further assistance would be greatly appreciated.

Myzhar, 2020-12-02:
We are investigating the problem. I will be back to you as soon as we get more information.

Myzhar, 2020-12-02:
@steveanthonyjones in the first tests that I made on the Jetson Nano with JP4.4.1 and SDK v3.3.2 I'm retrieving images at ~12Hz setting the grab framerate to 30 and the publishing rate to 15.
I was getting initially very bad results, but I noticed that the default depth mode is wrongly set to `ULTRA` instead of `PERFORMANCE`.
I'm expecting the worst performances on ROS2 with respect to ROS1 because of the underlying DDS infrastructure that is not light from a computational point of view.

Using your YAML files I get ~6Hz subscribing to `left/image_rect_color`.

However, I'm investigating further the code searching for possible bottlenecks that on Desktop and on other more powerful Jetson devices are not so evident.

Myzhar, 2020-12-02:
Like I supposed there is something that slows down the data communication at DDS level.

This is what I get printing the publishing rate at node level (activate the debug mode and remove comments in `publishVideoDepth`):
```
[zed_wrapper-2] 1606926982.535107535 [zed2.zed_node] [DEBUG] PUBLISHING PERIOD: 0.066954 sec @14.9356 Hz
[zed_wrapper-2] 1606926982.537091836 [zed2.zed_node] [DEBUG] MEAN PUBLISHING PERIOD: 0.0663077 sec @15.0812 Hz
[zed_wrapper-2] 1606926982.602994849 [zed2.zed_node] [DEBUG] PUBLISHING PERIOD: 0.066901 sec @14.9475 Hz
[zed_wrapper-2] 1606926982.603129899 [zed2.zed_node] [DEBUG] MEAN PUBLISHING PERIOD: 0.0663275 sec @15.0767 Hz
[zed_wrapper-2] 1606926982.665769772 [zed2.zed_node] [DEBUG] PUBLISHING PERIOD: 0.066233 sec @15.0982 Hz
[zed_wrapper-2] 1606926982.665898364 [zed2.zed_node] [DEBUG] MEAN PUBLISHING PERIOD: 0.0663244 sec @15.0774 Hz
[zed_wrapper-2] 1606926982.731846793 [zed2.zed_node] [DEBUG] PUBLISHING PERIOD: 0.033884 sec @29.5125 Hz
[zed_wrapper-2] 1606926982.734409941 [zed2.zed_node] [DEBUG] MEAN PUBLISHING PERIOD: 0.065243 sec @15.3273 Hz
[zed_wrapper-2] 1606926982.797835429 [zed2.zed_node] [DEBUG] PUBLISHING PERIOD: 0.099523 sec @10.0479 Hz
[zed_wrapper-2] 1606926982.797967563 [zed2.zed_node] [DEBUG] MEAN PUBLISHING PERIOD: 0.0663857 sec @15.0635 Hz
[zed_wrapper-2] 1606926982.863922398 [zed2.zed_node] [DEBUG] PUBLISHING PERIOD: 0.066712 sec @14.9898 Hz
[zed_wrapper-2] 1606926982.864050886 [zed2.zed_node] [DEBUG] MEAN PUBLISHING PERIOD: 0.0663966 sec @15.061 Hz
[zed_wrapper-2] 1606926982.929289479 [zed2.zed_node] [DEBUG] PUBLISHING PERIOD: 0.066848 sec @14.9593 Hz
[zed_wrapper-2] 1606926982.929432082 [zed2.zed_node] [DEBUG] MEAN PUBLISHING PERIOD: 0.0664116 sec @15.0576 Hz
[zed_wrapper-2] 1606926982.995322907 [zed2.zed_node] [DEBUG] PUBLISHING PERIOD: 0.066318 sec @15.0789 Hz
[zed_wrapper-2] 1606926982.997523820 [zed2.zed_node] [DEBUG] MEAN PUBLISHING PERIOD: 0.0664085 sec @15.0583 Hz
[zed_wrapper-2] 1606926983.061686751 [zed2.zed_node] [DEBUG] PUBLISHING PERIOD: 0.06695 sec @14.9365 Hz
[zed_wrapper-2] 1606926983.061826958 [zed2.zed_node] [DEBUG] MEAN PUBLISHING PERIOD: 0.0664265 sec @15.0542 Hz
[zed_wrapper-2] 1606926983.128110179 [zed2.zed_node] [DEBUG] PUBLISHING PERIOD: 0.033049 sec @30.2581 Hz
[zed_wrapper-2] 1606926983.128241792 [zed2.zed_node] [DEBUG] MEAN PUBLISHING PERIOD: 0.065314 sec @15.3107 Hz
[zed_wrapper-2] 1606926983.196238118 [zed2.zed_node] [DEBUG] PUBLISHING PERIOD: 0.033539 sec @29.816 Hz
[zed_wrapper-2] 1606926983.196404835 [zed2.zed_node] [DEBUG] MEAN PUBLISHING PERIOD: 0.0642548 sec @15.563 Hz
[zed_wrapper-2] 1606926983.259810431 [zed2.zed_node] [DEBUG] PUBLISHING PERIOD: 0.133867 sec @7.4701 Hz
[zed_wrapper-2] 1606926983.259949544 [zed2.zed_node] [DEBUG] MEAN PUBLISHING PERIOD: 0.0665752 sec @15.0206 Hz
[zed_wrapper-2] 1606926983.326512033 [zed2.zed_node] [DEBUG] PUBLISHING PERIOD: 0.066564 sec @15.0231 Hz
[zed_wrapper-2] 1606926983.326732447 [zed2.zed_node] [DEBUG] MEAN PUBLISHING PERIOD: 0.0665748 sec @15.0207 Hz
[zed_wrapper-2] 1606926983.390972357 [zed2.zed_node] [DEBUG] PUBLISHING PERIOD: 0.033648 sec @29.7194 Hz
[zed_wrapper-2] 1606926983.391103449 [zed2.zed_node] [DEBUG] MEAN PUBLISHING PERIOD: 0.0654773 sec @15.2725 Hz
[zed_wrapper-2] 1606926983.457424482 [zed2.zed_node] [DEBUG] PUBLISHING PERIOD: 0.066632 sec @15.0078 Hz
```

this is instead what I get with the command `ros2 topic hz /zed2/zed_node/left/image_rect_color`:
```
	min: 0.040s max: 0.983s std dev: 0.19490s window: 75
average rate: 4.808
	min: 0.040s max: 0.983s std dev: 0.20274s window: 78
average rate: 4.819
	min: 0.040s max: 0.983s std dev: 0.19781s window: 83
average rate: 4.801
	min: 0.040s max: 0.983s std dev: 0.19967s window: 89
average rate: 4.815
	min: 0.040s max: 0.983s std dev: 0.20362s window: 95
average rate: 4.846
	min: 0.040s max: 0.983s std dev: 0.19891s window: 102
average rate: 4.798
	min: 0.040s max: 0.983s std dev: 0.19748s window: 106
average rate: 4.779
	min: 0.040s max: 0.983s std dev: 0.19549s window: 111
average rate: 4.621
	min: 0.040s max: 0.983s std dev: 0.20381s window: 114
average rate: 4.650
	min: 0.040s max: 0.983s std dev: 0.19920s window: 120
average rate: 4.587
	min: 0.040s max: 0.983s std dev: 0.19684s window: 125
average rate: 4.654
	min: 0.040s max: 0.983s std dev: 0.19314s window: 132
average rate: 4.644
	min: 0.040s max: 0.983s std dev: 0.19389s window: 137
average rate: 4.630
	min: 0.040s max: 0.983s std dev: 0.19298s window: 143
average rate: 4.602
```

Myzhar, 2020-12-03:
An update to this issue. 
We introduced some little improvements with this MR:
https://github.com/stereolabs/zed-ros2-wrapper/pull/23
however, the change is really slight.
I checked the frequency of publishing of all topics at the code level and they are all correct, the slowing down is probably generated, like I thought, by the communication infrastructure of ROS2 that on the Jetson Nano has not enough resource to work at full speed.
If you want to test more this aspect of ROS2 I suggest you try to change the RMW middleware testing all the possible options.
Here you can find useful information:
https://index.ros.org/doc/ros2/Tutorials/Working-with-multiple-RMW-implementations/

steveanthonyjones, 2020-12-07:
Thanks for looking into this Myzhar. Is there a way to confirm what resources are actually limited?  tegrastats report didn't suggest any issues in it's scope as mentioned earlier. Are sterolabs following this up with ROS2 developers?

steveanthonyjones, 2020-12-07:
Hi Myzhar, I'm happy to report that I tested the "some little improvements" you mentioned for MR: #23. This significantly improved frame rate to a usable 12-13 Hz with default settings (publish rate set at 15Hz so still restricted but, this is operating at higher 1080 resolution). Thank you for the revisions!

Myzhar, 2020-12-07:
Hi @steveanthonyjones 
thank you for the feedback :+1:

shecker, 2021-05-05:
Hello is there any update on this issue? 
We are witnessing a severe drop in point cloud and image publishing frequency on a Jetson Xavier AGX running ROS2 foxy. The native ZED SDK samples are running fine at around 50Hz on ULTRA depth at 720p. However, once we start publishing this data via ROS2 we are stuck at around 22-25HZ and the variance between messages is very high, (drops in images, where we don't see an output for like 0.8s).

If you have any idea on how to tune the ROS2 middleware to improve this, that would be greatly appreciated!
Thanks a lot.

Myzhar, 2021-05-05:
@shecker how are you measuring the publishing frequency of the topics?

shecker, 2021-05-05:
we are using `ros2 topic hz <topic_name>` we have also tried implementing our own small zed sdk + ros2 publisher. Unfortunately it does not seem to help to get the frequency to increase.

Myzhar, 2021-05-05:
I supposed. The `ros2 topic hz` command sometimes does not provide a good estimation of the frequency of the topics. Have you tried the `rqt_topic_monitor` with the `rqt` gui?

closed 2021-08-06 by Myzhar