stereolabs/zed-ros2-wrapper #24: [Question] Difference in depth image quality between SDK 3.3.0 and SDK 3.4.0

laszloturanyi, 2021-02-15:
Hi,

We experienced some difference when using ZED SDK 3.4.0 compared to version 3.3.0 regarding the depth image quality.

See attached attached images below.
The depth quality with SDK 3.3.0 seems to be better. We used zed-ros2-wrapper with same configuration on the same svo recording. The compute platform is a Jetson AGX Xavier.

Corresponding RGB
**RGB image**
[image]

**SDK 3.3.0**
[image]

**SDK 3.4.0**
[image]

Can you tell us what could be the cause for this? We chose to revert to SDK 3.3.0.
Thanks!

Myzhar, 2021-02-15:
Hi @laszloturanyi 
Thank you for reporting. 
This seems to be a problem related to the ZED SDK. Can you please send an email to support@stereolabs.com describing it in details? Is it present also on desktop platforms and using different depth qualities?

laszloturanyi, 2021-02-16:
Hi @Myzhar, 

Thank you for your answers. I have sent an email to the address. We also tried this on a pc with ubuntu 18.04 and we could see the same issue. Depth quality was set to performance (1) and did not try it with any other setting.

Regards,
Laszlo

Myzhar, 2021-02-16:
Hi @laszloturanyi 
we identified the issue in the ZED SDK and we are solving it.
It will be fixed with the next patch release.

Thank you for reporting

closed 2021-02-16 by Myzhar

laszloturanyi, 2021-02-16:
Wow. What a quick reaction.
Thank you for your help.