stereolabs/zed-ros2-wrapper #25: Trouble building with CUDA 11.1 - CUDA_nppicom_LIBRARY not found

jknick, 2021-02-18:
Is there a dependency on CUDA 10.2 for this wrapper?

When trying to build with colcon for Ubuntu 20.04 with CUDA 11.1 we're hitting this error:
 
```Starting >>> zed_components
--- stderr: zed_components

CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDA_nppicom_LIBRARY (ADVANCED)
linked by target "zed_camera_component" in directory /home/kanke/ros2_example_ws/src/zed-ros2-wrapper/zed_components

Failed   <<< zed_components [1.72s, exited with code 1]

Summary: 5 packages finished [3.42s]
1 package failed: zed_components
1 package had stderr output: zed_components
2 packages not processed
```

In digging a little bit it seems that libnppicom.so is no longer part of CUDA, but also I don't see where the reference to this is coming from as it doesn't seem to be reference anywhere in this repo.  

Any thoughts?  Thanks!

Rayman, 2021-03-15:
I've applied this diff to my FindCUDA and it now seems to compile:

https://gitlab.kitware.com/cmake/cmake/-/merge_requests/4929

jknick, 2021-03-16:
Thanks, Rayman!

Kismetzc, 2021-05-25:
The same question. I'm using Ubuntu18.04, Cuda11.0, SDK3.5 and ros2-eloquent.

Kismetzc, 2021-05-25:
@jknick @Rayman Do you solve it?

Myzhar, 2021-05-25:
@Kismetzc I suppose this is the solution that they found:
https://github.com/stereolabs/zed-ros2-wrapper/issues/25#issuecomment-799383330

Kismetzc, 2021-05-25:
> @Kismetzc I suppose this is the solution that they found:
> [#25 (comment)](https://github.com/stereolabs/zed-ros2-wrapper/issues/25#issuecomment-799383330)

Yeah. Does this mean I should update my cmake version？

Myzhar, 2021-05-25:
Yes, it can solve the issue

Kismetzc, 2021-05-25:
> Yes, it can solve the issue

THX！I will try right now.

Kismetzc, 2021-05-25:
Thank you all !!! 
I have solved this problem.

Myzhar, 2021-05-25:
The solution works, so I close the issue

closed 2021-05-25 by Myzhar