stereolabs/zed-ros2-wrapper #26: [Question] About ROS 2 Jetson docker

cnpcshangbo, 2021-02-28:
Hello,
Do you have a ROS 2 Jetson docker image or Dockerfile for testing this zed-ros2-wrapper?
Thanks,
Bo

cnpcshangbo, 2021-02-28:
We are considering ROS 2 because ROS Melodic doesn't support Python 3 fully. We'd like to try ROS 2 to avoid using Python 2.

Myzhar, 2021-03-18:
Hi @cnpcshangbo 
there is not yet a ROS2 docker available yet, but we plan to release it soon.

cnpcshangbo, 2021-03-18:
Great. Thanks.

PureRockets, 2021-04-04:
+1 for a ROS2 docker image for the Jetson Nano!

XDGFX, 2021-07-09:
+1. I'm trying to build my own container but having some issues getting the Python API to work building on-top of the dustynv ros foxy container. If I don't get any further I'll open an issue on the relevant repo.

griz1112, 2021-07-09:
I have a working one. Just got it to build yesterday. Let me do a bit more testing and I will create a repo for it. I started with dusty nv also but you need to use one of the stereolabs images so the sdk will be installed properly. If you are having issues with numpy core dumping use this. export OPENBLAS_CORETYPE=ARMV8 I suspect this has caused a faulty install of the sdk python stuff.

XDGFX, 2021-07-09:
That fixed the coredump issue I was having! Seems to be working (the Hello Zed tutorial is all I've tested so far). Thanks!

For reference, below is my Dockerfile so far. Need to manually export OPENBLAS_CORETYPE=ARMV8 after launching the container interactively

```Dockerfile
ARG ROS_VERSION=foxy
ARG L4T_MINOR_VERSION=5.0
ARG FROM_IMAGE=dustynv/ros:$ROS_VERSION-ros-base-l4t-r32.$L4T_MINOR_VERSION
ARG OVERLAY_WS=/opt/ros/overlay_ws

# Base container
FROM $FROM_IMAGE

# After using an arg in a `FROM` line, the arg is lost.
# This will allow it to be used again.
ARG ROS_VERSION
ARG L4T_MINOR_VERSION
ARG OVERLAY_WS

ARG ZED_SDK_MAJOR=3
ARG ZED_SDK_MINOR=5
ARG JETPACK_MAJOR=4
ARG JETPACK_MINOR=5

# Update and install dependencies
RUN apt-get update && apt-get install --no-install-recommends -y \
    lsb-release \
    less \
    udev \
    wget \
    apt-transport-https \
    python3-pip

# Fix for numpy installation later on
RUN pip3 install --upgrade pip

# Install ZED SDK
RUN echo "# R32 (release), REVISION: ${L4T_MINOR_VERSION}" > /etc/nv_tegra_release ; \
    wget -q --no-check-certificate -O ZED_SDK_Linux_JP.run https://download.stereolabs.com/zedsdk/${ZED_SDK_MAJOR}.${ZED_SDK_MINOR}/jp${JETPACK_MAJOR}${JETPACK_MINOR}/jetsons && \
    chmod +x ZED_SDK_Linux_JP.run ; ./ZED_SDK_Linux_JP.run silent skip_tools && \
    rm -rf /usr/local/zed/resources/* \
    rm -rf ZED_SDK_Linux_JP.run && \
    rm -rf /var/lib/apt/lists/*

# Install ZED Python API
RUN apt-get update -y && apt-get install --no-install-recommends python3 python3-pip python3-dev gfortran gcc musl-dev python3-setuptools build-essential -y && \ 
    wget download.stereolabs.com/zedsdk/pyzed -O /usr/local/zed/get_python_api.py && \
    python3 /usr/local/zed/get_python_api.py && \
    python3 -m pip install cython wheel && \
    python3 -m pip install numpy pyopengl *.whl && \
    apt-get remove --purge build-essential -y && apt-get autoremove -y && \
    rm *.whl ; rm -rf /var/lib/apt/lists/*

# Copy source
WORKDIR $OVERLAY_WS
COPY src src

# Build
RUN /bin/bash -c "source /opt/ros/${ROS_VERSION}/install/setup.bash && colcon build"

# Source entrypoint setup, used if running container interactively to
# automatically source the ROS workspace.
ENV OVERLAY_WS $OVERLAY_WS
RUN sed --in-place --expression \
    '$isource "$OVERLAY_WS/install/setup.bash"' \
    /ros_entrypoint.sh
```

Need to manually export OPENBLAS_CORETYPE=ARMV8 after launching the container interactively, and I'm using the below run command to launch it.

```bash
sudo docker run -it --rm --gpus all --privileged --network=host ${container_name}
```

griz1112, 2021-07-09:
if you go to the stereolabs section of hub.docker.com there are prebuilt images you can start with. https://hub.docker.com/r/stereolabs/zed/  Adding one of those to my your dockerfile with the from keyword is how I built mine.  The ones I built are for ros2 foxy. You should be able to add that export with some sed commands to your .bashrc file in the container.

XDGFX, 2021-07-09:
@griz1112 Which image did you start with? From what I understand they're all either without ROS or ROS1.

Still learning about all this but everything Jetson specific seems to be based off Ubuntu 18.04, looks like this might cause issues with e.g. the ROS2 Wrapper node.

griz1112, 2021-07-09:
You can actually install ros2 foxy native on 18.04 there are a few extra packages you need to compile for it to work that aren't included in the 18.04 distro. But then you can't have ros1 on the same machine because it removes part of it. So the container route is the best way to go. Especially if you need to use the ros2 bridge for some of your nodes. There is a Dockerfile in jetson-containers that will build ros2 foxy and ros1 noetic. I took the one for foxy and modified it to use the stereolabs image here https://github.com/stereolabs/zed-docker/blob/master/3.X/jetpack_4.X/tools-devel/ Dockerfile so I would have the ZED SDK. Built a base and desktop image from that. I've tested it with a ZED ZED Mini and ZED 2i on an Xavier and Nano. Most of the packages on 20.04 are the same as 18.04.  You can build a 20.04 container that will run on a 18.04 install. I've even installed 20.04 on one of my jetson boards. However I wouldn't recommend it as its possible to brick your jetson if you aren't very careful. And I'm sure there are some differences in the kernel that could cause you some grief. Dusty wouldn't put a Dockerfile out there for foxy on 18.04 if it wasn't ok to run it that way. I've been working with jetsons for almost 6 years now. Starting with a 4 node TK1 cluster.

griz1112, 2021-07-10:
Ran into a problem testing the container with the ros2_wrapper.
[zed_wrapper-2] 1625902681.149991893 [rcl] [ERROR] Failed to parse global arguments
[zed_wrapper-2] terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidROSArgsError'
[zed_wrapper-2]   what():  failed to initialize rcl: Couldn't parse params file: '--params-file /zed/install/zed_wrapper/share/zed_wrapper/config/zedm.yaml'. Error: Error parsing a event near line 14, at /opt/ros/foxy/src/rcl/rcl_yaml_param_parser/src/parse.c:600, at /opt/ros/foxy/src/rcl/rcl/src/rcl/arguments.c:390
[ERROR] [zed_wrapper-2]: process has died [pid 2972, exit code -6, cmd '/zed/install/zed_wrapper/lib/zed_wrapper/zed_wrapper --ros-args -r __node:=zed_node -r __ns:=/zedm --params-file /zed/install/zed_wrapper/share/zed_wrapper/config/common.yaml --params-file /zed/install/zed_wrapper/share/zed_wrapper/config/zedm.yaml'].

On my intel box on 20.04 this works just fine. So not sure changing the zed config yaml files will do any good. One of the things you have to add to get foxy installed in the container is libyaml so I suspect its something to do with that. Guess I need to study up on yaml files next.

Myzhar, 2021-08-05:
@XDGFX a really useful job. Thank you :+1:

CircleOnCircles, 2021-09-15:
this is my dockerfile for ZED 2i and Jetson NX Xavier.
<https://dev.to/circleoncircles/how-to-build-zed-2i-camera-x-ros2-foxy-x-nvidia-jetson-x-ubuntu-18-04-via-docker-8o4>
it is a bit big(~3.4GB) but also work perfectly.

closed 2021-12-07 by Myzhar

AndreV84, 2022-04-06:
@Myzhar is there ros2 foxy zed docker conrainer by stereolabs by now?

Myzhar, 2022-04-06:
Hi @AndreV84, we stopped generating the ROS/ROS2 containers because it was not easy to keep them synchronized with the latest updates by the many operators involved (Stereolabs, Nvidia, ROS/ROS2, ecc).

AndreV84, 2022-04-06:
@Myzhar could you advise which value to put for zed camera in the global frame for ros2 foxy zed wrapper?
```
Usually, the "global_frame" is map, but it depends on your configuration and on what node you are using.
```

Myzhar, 2022-04-06:
@AndreV84 this is a closed issue and this question is off-topic.
Can you please write to support@stereolabs.com or post the question with more details on [our forum](https://community.stereolabs.com/)

AndreV84, 2022-08-20:
@Myzhar is it likely that you could support building in Humble ROS container[provided by nvidia] on Jetson? outside of container for Humble? asked to email too but no response