stereolabs/zed-ros2-wrapper #28: [Question] Jetson nano ip streaming

mastrcheif, 2021-03-16:
Hello,
would it be possible to open the mZed camera with an ip address? I would like to stream multiple zed2 cameras over the network and do image and depth evaluation on another pc. As a requirement it is necessary to have all components talking over ros2 but running the ros2-zed2-node on the Jetson Nano itself is not optimal.

Myzhar, 2021-03-16:
Hi @mastrcheif 
the ZED ROS Wrapper has a service to start remote streaming, but it's not yet available in the ROS2 wrapper because it was not required by other users.
We will add this feature as soon as possible.

mastrcheif, 2021-03-16:
@Myzhar this sounds like music to my ears. I thank you in advance for the implementation of that feature.

peredwardsson, 2022-05-17:
Also curious about this feature! Any news? Thanks for the great work.

Myzhar, 2022-05-17:
The feature will be released in the next big upgrade.

peredwardsson, 2022-05-17:
Awesome! ETA?

jonpol01, 2022-05-20:
> The feature will be released in the next big upgrade.

@Myzhar may I ask does this include the ros2 launcher to open a stream aswell?
currently I am tinkering the [zed_camera_component ](https://github.com/jonpol01/zed-ros2-wrapper/blob/1a40fbe6a63c45880a8f4498cdf79bf2db965b3a/zed_components/src/zed_camera/src/zed_camera_component.cpp#L2398)through a fork, and managed to to a 0 = default, 1 = stream camera..
but, having a hard time right now to have it open a stream.

Myzhar, 2022-05-20:
In the ROS1 wrapper there is a service for that. Maybe it will be the same

Howard-Ryu-Brooklyn, 2023-07-04:
@Myzhar I'm trying to use ZED in ROS2 via streaming. Is the update your mentioned compeleted? if it is, how can I use it?