stereolabs/zed-ros2-wrapper #29: [Question] ZED2 Resolution Halved

jknick, 2021-03-16:
Having trouble with this wrapper.  I see in the common.yaml file the device is being set to 720p by default, but  when I echo the topic I see:

header:
  stamp:
    sec: 1615931279
    nanosec: 259278004
  frame_id: zed2_left_camera_optical_frame
**height: 360
width: 640**
encoding: bgra8

640x360

So exactly half of 720p.  Setting this to 1080p results in a resolution of 960x540.  Again, exactly half.  I'm digging into this more, but up front I don't see what the cause is.  QOS settings?  Something dynamic?  Bug?  User error?

Help appreciated.

Myzhar, 2021-03-16:
Hi @jknick 
That's not a bug, but a setting. If you want full resolution image as output simply set this parameter to 1.0:
https://github.com/stereolabs/zed-ros2-wrapper/blob/master/zed_wrapper/config/common.yaml#L27

Myzhar, 2021-03-16:
`img_downsample_factor`, like `depth_downsample_factor`, is used to reduce bandwidth, keeping a good view result.

jknick, 2021-03-16:
And I see it now.  This works.

Much thanks!

closed 2021-03-18 by Myzhar