stereolabs/zed-ros2-wrapper #31: ZED2, ROS2-foxy: compressed image topic not available with errors

jfinken, 2021-03-24:
**Describe the bug**
When launching the `zed-ros2-wrapper` _with_ the [image_transport_plugins](https://github.com/ros-perception/image_transport_plugins/tree/ros2) installed, the below errors are observed and `compressed` image topics are not available.

**Device information:**
 - OS: Ubuntu
 - OS version: 18.04
 - CPU: i7
 - GPU Nvidia:  Quadro T1000
 - CUDA 10.2
 - ZED SDK Version: v3.4.2 for CUDA 10.2
 - ROS Wrapper version: `master`, last commit via `git log`:
```
commit 11ca6484df8aaf2cc6a0ecb524182c0d93c56c42 (HEAD -> master, origin/master, origin/HEAD)
Merge: f7bb886 ade4cb2
Author: Walter Lucetti <walter.lucetti@stereolabs.com>
Date:   Thu Dec 3 14:17:52 2020 +0100
```

**To Reproduce**
1. Build and install the zed-ros2-wrapper, for example:
```bash
$ cd ~/my-foxy-workspace/src
$ git clone https://github.com/stereolabs/zed-ros2-wrapper.git
$ cd ~/my-foxy-workspace
$ rosdep install --from-paths src --ignore-src --rosdistro foxy -y --skip-keys "console_bridge fastcdr fastrtps rti-connext-dds-5.3.1 urdfdom_headers"
$ colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release
```
2. Build and install image_transport_plugins
```bash
$ cd ~/my-foxy-workspace/src
$ git clone git@github.com:ros-perception/image_transport_plugins.git
# this needs to be on the same branch as the image_common dependency
$ cd image_transport_plugins && git checkout ros2 
# hence:
$ cd ~/my-foxy-workspace/src/image_common && git checkout ros2

$ rosdep install --from-paths src --ignore-src --rosdistro foxy -y --skip-keys "console_bridge fastcdr fastrtps rti-connext-dds-5.3.1 urdfdom_headers"
$ colcon build --symlink-install
```
3. Run the zed-ros2-wrapper, observe errors:
```bash
$ ros2 launch zed_wrapper zed2.launch.py
...
[zed_wrapper-2] 1616597777.165880326 [zed2.zed_node] [INFO]  * Right Temperature	-> zed2_right_camera_frame
[zed_wrapper-2] 1616597777.166162383 [zed2.zed_node] [INFO] *** PUBLISHED TOPICS ***
[zed_wrapper-2] 1616597777.288631336 [zed2.zed_node] [ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'format' could not be set: Not supported
[zed_wrapper-2] 
[zed_wrapper-2] 1616597777.299310657 [zed2.zed_node] [INFO] Advertised on topic: /zed2/zed_node/rgb/image_rect_color
[zed_wrapper-2] 1616597777.300331471 [zed2.zed_node] [ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'format' has already been declared
[zed_wrapper-2] 
[zed_wrapper-2] 1616597777.301188885 [zed2.zed_node] [INFO] Advertised on topic: /zed2/zed_node/rgb/image_rect_gray
[zed_wrapper-2] 1616597777.301215098 [zed2.zed_node] [INFO] Advertised on topic: /zed2/zed_node/rgb/camera_info
[zed_wrapper-2] 1616597777.302073967 [zed2.zed_node] [ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'format' has already been declared
[zed_wrapper-2] 
[zed_wrapper-2] 1616597777.302909071 [zed2.zed_node] [INFO] Advertised on topic: /zed2/zed_node/rgb_raw/image_raw_color
[zed_wrapper-2] 1616597777.303637059 [zed2.zed_node] [ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'format' has already been declared
```
4. Now, manually apply [this yet-to-be-merged PR](https://github.com/ros-perception/image_transport_plugins/pull/58) from `image_transport_plugins`:
```bash
$ find image_transport_plugins/ -name compressed_publisher.cpp
  image_transport_plugins/compressed_image_transport/src/compressed_publisher.cpp
# Download compressed_publisher.cpp containing the fixes from the MR
# Rebuild your workspace and/or the ws that contains image_transport_plugins
```
5. Re-run the zed-ros2-wrapper, observer _different_ errors:
```bash
$ ros2 launch zed_wrapper zed2.launch.py
...
[zed_wrapper-2] 1616598630.319304068 [zed2.zed_node] [INFO]  * Right Temperature	-> zed2_right_camera_frame
[zed_wrapper-2] 1616598630.319344545 [zed2.zed_node] [INFO] *** PUBLISHED TOPICS ***
[zed_wrapper-2] 1616598630.382450727 [zed2.zed_node] [ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'zed_node.rgb.image_rect_color.format' could not be set: Not supported
[zed_wrapper-2] 
[zed_wrapper-2] 1616598630.388444699 [zed2.zed_node] [INFO] Advertised on topic: /zed2/zed_node/rgb/image_rect_color
[zed_wrapper-2] 1616598630.389348695 [zed2.zed_node] [ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'zed_node.rgb.image_rect_gray.format' could not be set: Not supported
[zed_wrapper-2] 
[zed_wrapper-2] 1616598630.390243071 [zed2.zed_node] [INFO] Advertised on topic: /zed2/zed_node/rgb/image_rect_gray
[zed_wrapper-2] 1616598630.390296178 [zed2.zed_node] [INFO] Advertised on topic: /zed2/zed_node/rgb/camera_info
[zed_wrapper-2] 1616598630.391076111 [zed2.zed_node] [ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'zed_node.rgb_raw.image_raw_color.format' could not be set: Not supported
[zed_wrapper-2] 
[zed_wrapper-2] 1616598630.391579021 [zed2.zed_node] [INFO] Advertised on topic: /zed2/zed_node/rgb_raw/image_raw_color
[zed_wrapper-2] 1616598630.392199161 [zed2.zed_node] [ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'zed_node.rgb_raw.image_raw_gray.format' could not be set: Not supported

```

**Expected behavior**
We expect to see valid compressed image topics, e.g.
```
$ ros2 topic list
/zed2/zed_node/rgb/image_rect_color
/zed2/zed_node/rgb/image_rect_color/compressed
/zed2/zed_node/rgb/image_rect_color/compressedDepth
```

**Additional context**
Walter from Stereolabs reports that the parameters around JPEG compression do not exist in ROS2-eloquent when running:
```bash
# in eloquent...
$ ros2 param list
```

Myzhar, 2021-03-24:
This is a ROS2 Foxy bug that they have just solved but it has not yet been merged in the repository.
To avoid the problems you must compile the `image_transport` package by source.

However the `master` branch of the ZED ROS2 Wrapper is not ready to work with ROS2 Foxy, we are currently working on a Foxy release, you can checkout the `dev_foxy` branch.

Myzhar, 2021-03-24:
Recompiling the `image_transport_plugins` package the new error 
` [ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'zed_node.xxxxxx.yyyyyyyyyyyyy.format' could not be set: Not supported`
comes out.
I suppose that this is caused by the wrong handling of the params namespace by the [applied patch](https://github.com/ros-perception/image_transport_plugins/pull/58).

I'm working on it to try to fix the problem, waiting for a solution by the ROS developers

Myzhar, 2021-03-24:
I reverted to the "original" `image_transport_plugins` package available with `sudo apt install ros-foxy-image-transport-plugins` to verify what happens when advertizing the image topics and I noticed this:
```
[zed_wrapper-2] 1616604958.106023291 [zed2.zed_node] [INFO] *** PUBLISHED TOPICS ***
[zed_wrapper-2] 1616604958.210126809 [zed2.zed_node] [ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'format' could not be set: Not supported
[zed_wrapper-2] 
[zed_wrapper-2] 1616604958.215524369 [zed2.zed_node] [INFO] Advertised on topic: /zed2/zed_node/rgb/image_rect_color
[zed_wrapper-2] 1616604958.216225843 [zed2.zed_node] [ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'format' has already been declared
[zed_wrapper-2] 
[zed_wrapper-2] 1616604958.216975911 [zed2.zed_node] [INFO] Advertised on topic: /zed2/zed_node/rgb/image_rect_gray
[zed_wrapper-2] 1616604958.216997175 [zed2.zed_node] [INFO] Advertised on topic: /zed2/zed_node/rgb/camera_info
[zed_wrapper-2] 1616604958.217617106 [zed2.zed_node] [ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'format' has already been declared
[zed_wrapper-2] 
[zed_wrapper-2] 1616604958.218326287 [zed2.zed_node] [INFO] Advertised on topic: /zed2/zed_node/rgb_raw/image_raw_color
[zed_wrapper-2] 1616604958.219129885 [zed2.zed_node] [ERROR] Failed to load plugin image_transport/compressed_pub, error string: parameter 'format' has already been declared
```
at the very beginning, the error is `not supported` then it becomes `parameter 'format' has already been declared`, indeed if you look in the params list there is a `format` parameter.

I can say that there are two problems generated by the same cause: the `image_transport` plugin does not correctly handle the namespace of the node.

I continue to investigate the problem to try to understand if it is possible to fix it

jfinken, 2021-03-24:
Yes, I have tried all of the above as well (as per the Description):
- compile `image_transport_plugins` from source
- applying [PR58](https://github.com/ros-perception/image_transport_plugins/pull/58) manually, then recompiling.
- observe all of the above errors with both attempts.

> I can say that there are two problems generated by the same cause: the image_transport plugin does not correctly handle the namespace of the node.  I continue to investigate the problem to try to understand if it is possible to fix it

Great thank you so much Walter.

laszloturanyi, 2021-04-23:
This has been merged.
https://github.com/ros-perception/image_transport_plugins/pull/58
I tried to compile image_transport_plugins from source and it worked fine.

Myzhar, 2021-04-23:
Good news, thank you

jfinken, 2021-04-24:
Note that I have previously applied [PR58](https://github.com/ros-perception/image_transport_plugins/pull/58) manually and still observed the issue with the ZED2/zed_ros2_wrapper/ROS2-foxy.  Do you observe the same @Myzhar?  

Thanks in advance.

closed 2021-12-07 by Myzhar