stereolabs/zed-ros2-wrapper #35: [BUG] Weird Behaviour when replaying an ZED2 SVO in `realtime`

kevinanschau, 2021-05-05:
**Describe the bug**
A clear and concise description of what the bug is.

**Device information:**
 - OS: container based on Jetpack 4.4
 - CPU: Jetson Xavier NX
 - GPU Nvidia: Jetson Xavier NX
 - ZED SDK Version: 3.3.3 for Jetpack 4.4
 - ROS Wrapper version: latest master (678bb3b)

**To Reproduce**
Steps to reproduce the behavior:
1. Use the following parameters:
    ```yaml
    /**:
      general:
        debug_mode: true
        sdk_verbose: true
        svo_file: path/to/svo_file
        svo_loop: false
        svo_realtime: true
        camera_model: zed2
        camera_name: zed2
        camera_flip: true
        resolution: 2
        self_calib: false
        grab_frame_rate: 30
        pub_frame_rate: 15.0
      video:
        img_downsample_factor: 0.5
      depth:
        quality: 1
        min_depth: 1.0
        max_depth: 10.0
        depth_downsample_factor: 0.5
        point_cloud_freq: 15.0
      pos_tracking:
        publish_tf: false
        publish_map_tf: false
        base_frame: base_link
    ```
2. Launch with
    ```bash
    ros2 launch zed_wrapper zed2.launch.py
    ```
3. See the log below for when running on a Jetson Xavier NX. When I do the same on my personal laptop, the node initially publishes topics but stops publishing at random. The SVO file is still being replayed, though, as I was able to see from the `SVO reached the end. The node has been stopped` message in the logs. Note, when I change `svo_realtime` to `True`, everything works just fine.

**Expected behavior**
The node starts without camera error.

**ROS log**
```bash
[robot_state_publisher-1] Initialize urdf model from file: /workspaces/eloquent/install/zed_wrapper/share/zed_wrapper/urdf/zed2.urdf
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link zed2_camera_center had 2 children
[robot_state_publisher-1] Link zed2_left_camera_frame had 1 children
[robot_state_publisher-1] Link zed2_left_camera_optical_frame had 0 children
[robot_state_publisher-1] Link zed2_right_camera_frame had 1 children
[robot_state_publisher-1] Link zed2_right_camera_optical_frame had 0 children
[robot_state_publisher-1] got segment base_link
[robot_state_publisher-1] got segment zed2_camera_center
[robot_state_publisher-1] got segment zed2_left_camera_frame
[robot_state_publisher-1] got segment zed2_left_camera_optical_frame
[robot_state_publisher-1] got segment zed2_right_camera_frame
[robot_state_publisher-1] got segment zed2_right_camera_optical_frame
[robot_state_publisher-1] Adding fixed segment from base_link to zed2_camera_center
[robot_state_publisher-1] Adding fixed segment from zed2_camera_center to zed2_left_camera_frame
[robot_state_publisher-1] Adding fixed segment from zed2_left_camera_frame to zed2_left_camera_optical_frame
[robot_state_publisher-1] Adding fixed segment from zed2_camera_center to zed2_right_camera_frame
[robot_state_publisher-1] Adding fixed segment from zed2_right_camera_frame to zed2_right_camera_optical_frame
[... 229 lines cut ...]
[zed_wrapper-2] 1620226851.352797063 [zed24388529.zed_node] [INFO] Published static TF: 'zed2_imu_link' -> 'zed2_left_camera_frame'
[zed_wrapper-2] 1620226851.422183929 [zed24388529.zed_node] [DEBUG] Point Cloud thread started
[zed_wrapper-2] 1620226851.422624250 [zed24388529.zed_node] [DEBUG] Grab thread started
[zed_wrapper-2] 1620226851.433411675 [zed24388529.zed_node] [INFO] *** Starting Positional Tracking ***
[zed_wrapper-2] 1620226851.433633691 [zed24388529.zed_node] [INFO]  * Waiting for valid static transformations...
[zed_wrapper-2] 1620226851.433733596 [zed24388529.zed_node] [DEBUG] Getting static TF from 'zed2_left_camera_frame' to 'base_link'
[zed_wrapper-2] 1620226851.433980732 [zed24388529.zed_node] [INFO] Static transform Sensor to Base [zed2_left_camera_frame -> base_link]
[zed_wrapper-2] 1620226851.434067581 [zed24388529.zed_node] [INFO]  * Translation: {0.000,-0.060,0.000}
[zed_wrapper-2] 1620226851.434171517 [zed24388529.zed_node] [INFO]  * Rotation: {0.000,-0.000,0.000}
[zed_wrapper-2] 1620226851.434283709 [zed24388529.zed_node] [DEBUG] Getting static TF from 'zed2_left_camera_frame' to 'zed2_camera_center'
[zed_wrapper-2] 1620226851.434531646 [zed24388529.zed_node] [INFO] Static transform Sensor to Camera Center [zed2_left_camera_frame -> zed2_camera_center]
[zed_wrapper-2] 1620226851.434658078 [zed24388529.zed_node] [INFO]  * Translation: {0.000,-0.060,0.000}
[zed_wrapper-2] 1620226851.435311648 [zed24388529.zed_node] [INFO]  * Rotation: {0.000,-0.000,0.000}
[zed_wrapper-2] 1620226851.435883426 [zed24388529.zed_node] [DEBUG] Getting static TF from 'zed2_camera_center' to 'base_link'
[zed_wrapper-2] 1620226851.435994050 [zed24388529.zed_node] [INFO] Static transform Camera Center to Base [zed2_camera_center -> base_link]
[zed_wrapper-2] 1620226851.436051555 [zed24388529.zed_node] [INFO]  * Translation: {0.000,0.000,0.000}
[zed_wrapper-2] 1620226851.436111523 [zed24388529.zed_node] [INFO]  * Rotation: {0.000,-0.000,0.000}
[zed_wrapper-2] 1620226851.488482977 [zed24388529.zed_node] [DEBUG] Video Depth callback called
[zed_wrapper-2] 1620226851.536341938 [zed24388529.zed_node] [DEBUG] Time required to get valid static transforms: 0.002 sec
[zed_wrapper-2] 1620226851.536644243 [zed24388529.zed_node] [INFO] Initial ZED left camera pose (ZED pos. tracking): 
[zed_wrapper-2] 1620226851.536779699 [zed24388529.zed_node] [INFO]  * T: [0,0.06,0]
[zed_wrapper-2] 1620226851.536919668 [zed24388529.zed_node] [INFO]  * Q: [0,0,0,1]
[zed_wrapper-2] 1620226854.915591036 [zed24388529.zed_node] [ERROR] Camera error: CAMERA NOT INITIALIZED
[zed_wrapper-2] 1620226854.915790972 [zed24388529.zed_node] [DEBUG] Grab thread finished
```

Myzhar, 2021-05-05:
Hi @kevinanschau 
I'm sorry, but it's not clear when the error `Camera error: CAMERA NOT INITIALIZED` appears.
Can you explain better what happens on the Xavier NX and what happens instead on the laptop PC?

kevinanschau, 2021-05-05:
Hi @Myzhar
Sure. I previously recorded an SVO file with my ZED2 camera. Now I wanted to replay said SVO file with the `zed_wrapper` node on a Jetson Xavier NX. I set the `general.svo_file` to the SVO file path and also set the parameter `general.svo_realtime` to `True` because I wanted a replay in real-time. Unfortunately, this results in the `Camera error: CAMERA NOT INITIALIZED`. 

On my laptop PC, though, when `general.svo_realtime` is `True`, the node starts without any error but stops publishing messages after some time (randomly).

In contrast, when I set `general.svo_realtime` to `False`, the node starts without any issues on both devices and replays the SVO till the end.

closed 2022-04-26 by github-actions[bot]