stereolabs/zed-ros2-wrapper #39: [Feature]Update ROS2 to a supported ROS2 platform?

PureRockets, 2021-05-25:
Your requirement for Eloquent https://index.ros.org/doc/ros2/Installation/Eloquent/ is EOL

Myzhar, 2021-05-25:
@PureRockets we know about EOL of Eloquent, that's why we are working on a Foxy release that you can get in the `dev_foxy` branch:
https://github.com/stereolabs/zed-ros2-wrapper/tree/dev_foxy

cardboardcode, 2021-07-04:
@PureRockets @Myzhar 

Hi, just a FYI. 

### I can verify that the **zed-ros2-wrapper** under **[dev_foxy](https://github.com/stereolabs/zed-ros2-wrapper/tree/dev_foxy)** works. :rocket: :+1: 

See below for the steps I took to set it up.

## **Environment**
1. Ubuntu **20.04**
2. **[CUDA 11.0](https://developer.nvidia.com/cuda-11.0-download-archive)** 
3. **[CUDNN 8.1.1](https://developer.nvidia.com/rdp/cudnn-archive)**
4. Kernel Image - **5.4.0**-47-generic
5. ROS2 **[Foxy](https://index.ros.org/doc/ros2/Installation/Foxy/)**
6. Nvidia Driver **460.80**
7. ZED SDK **3.5.0**

## **Setup** 
Download and Install [ZED-SDK 3.5.0 for Ubuntu 20.04](https://download.stereolabs.com/zedsdk/3.5/cu110/ubuntu20).

## **Quick Verify**
Run the following command to verify your ZED-SDK has been installed properly.
```bash
ZED_Depth_Viewer
```
This executable can be found under /usr/local/zed/tools. You should see a window pop and show the **images and pointcloud** seen from your connected ZED camera.

Download and Build **zed-ros2-wrapper** and **zed-ros2-examples** ROS2 packages onto your own workstation
```bash
cd $HOME
mkdir -p ~/zed_ros2_ws/src && cd ~/zed_ros2_ws/src
git clone https://github.com/stereolabs/zed-ros2-wrapper.git --single-branch --branch dev_foxy --depth 1
git clone https://github.com/stereolabs/zed-ros2-examples.git --single-branch --branch dev_foxy --depth 1
cd ~/zed_ros2_ws/
source /opt/ros/foxy/setup.bash
colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release
```

## **Run & Verify**
To know that it is working, run the following command:
```bash
cd ~/zed_ros2_ws/src
source /opt/ros/foxy/setup.bash
source install/setup.bash
ros2 launch zed_display_rviz2 display_zed.launch.py
```
You should then see RViz2 window pop up and show **images and pointclouds** from your connected ZED camera.

Myzhar, 2021-07-04:
Hi @PureRockets 
Thank you for the feedback. 
We have not yet merged the `dev_foxy` branch into the `master` branch to guarantee compatibility with Nvidia Jetson users that cannot "easily" migrate to Foxy because of Ubuntu 18.04.
Currently we are developing on the `dev_foxy` branch and we will merge it as soon as Nvidia will release a Jetpack with Ubuntu 20.04 support (Q1 2022?)
We tried to merge the Eloquent and the Foxy branches (like we did for Noetic and Melodic for ROS1), but there are too big differences (mainly on the launch part) and the final result is really confusing.

an99990, 2021-07-16:
Hi,
Just wanted to confirm that the dev_foxy works. I have done exactly like @cardboardcode mentionned and it works fine.

Myzhar, 2021-07-16:
@an99990 since a few days the stable version of the `Foxy` release has been merged in the `master` branch.
The `devel_foxy` branch is under development to add a few missing features: diagnostic, xacro, ecc

an99990, 2021-07-16:
Thank you for the update.
Do you guys think about doing an implementation of yolo with the zed and
ros2 foxy ?

Myzhar, 2021-07-16:
It's not planned. The ZED SDK has its own Object Detection module and it's up to the users to develop their own custom nodes that do not use the ZED SDK.

closed 2021-08-06 by Myzhar