stereolabs/zed-ros2-wrapper #40: [BUG] Unable to subscribe and display /zed/zed_node/left/image_rect_color Images

cardboardcode, 2021-07-04:
## **Describe the bug**
Subscribing to the `sensor_msgs/msg/Image` topic named **/zed/zed_node/left/image_rect_color** images fails and does not show images are being published.

**Device information:**
 - **OS**: Ubuntu
 - **OS** version: 20.04
 - **CPU**: i7
 - **GPU Nvidia**:  GTX Geforce 1060
 - **ZED SDK Version**: 3.5.0
 - **ROS Wrapper version**: latest **dev_foxy**
 - **ZED Camera Model**: ZED

## **To Reproduce**
Steps to reproduce the behavior:
##### To Reproduce: Setup
```bash
cd $HOME
mkdir -p ~/zed_ros2_ws/src && cd ~/zed_ros2_ws/src
git clone https://github.com/stereolabs/zed-ros2-wrapper.git --single-branch --branch dev_foxy --depth 1
git clone https://github.com/stereolabs/zed-ros2-examples.git --single-branch --branch dev_foxy --depth 1
cd ~/zed_ros2_ws/
source /opt/ros/foxy/setup.bash
colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release
# Launch zed_display_rviz2
ros2 launch zed_display_rviz2 display_zed.launch.py
```
**Observed** that RViz2 is indeed able to display published images from **/zed/zed_node/left/image_rect_color**. See screenshot attached.

### Run **zed_tutorial_video** example subscription node taken from [this ZED tutorial link](https://www.stereolabs.com/docs/ros2/video/#running-the-tutorial). 
```bash
source /opt/ros/foxy/setup.bash
cd ~/zed_ros2_ws/
source install/setup.bash
ros2 run zed_tutorial_video zed_tutorial_video --ros-args -r left_image:=/zed/zed_node/left/image_rect_color -r right_image:=/zed/zed_node/right/image_rect_color
```
**zed_tutorial_video** is unresponsive and does not show anything.

### Run `ros2 topic echo` to check if images are being published.
```bash
source /opt/ros/foxy/setup.bash
ros2 topic echo /zed/zed_node/left/image_rect_color Images
```
**Observed** that nothing is being echoed on this topic, even though we confirmed earlier that RVIz2 is able to display the images being published on the same topic. Refer to `.gif` animation below.

## **Expected behavior**
Expected to run the following commands and observe the **sensor_ msgs/msg/Image** raw information being published onto terminal while **zed_display_rviz2** ROS2 package is running:
```bash
source /opt/ros/foxy/setup.bash
ros2 topic echo /zed/zed_node/left/image_rect_color

```

## **Screenshots**
[image]
[image]

## **Additional context**
Attempting to write my own ROS2 node that subscribes to the images published on this topic to use for subsequent 2D image processing.

cardboardcode, 2021-07-04:
Attached below is the terminal output upon running the command `$ ros2 launch zed_display_rviz2 display_zed.launch.py`:

<details>

```bash
cardboardvoice@bastion:~/ros_workspace/zed_ros2_ws$ bash run_zed_camera.bash 
[INFO] [launch]: All log files can be found below /home/cardboardvoice/.ros/log/2021-07-04-15-46-21-707813-bastion-19916
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [19918]
[INFO] [zed_wrapper-2]: process started with pid [19920]
[INFO] [rviz2-3]: process started with pid [19922]
[robot_state_publisher-1] 1625384781.782304523 [zed.zed_state_publisher] [WARN] No robot_description parameter, but command-line argument available.  Assuming argument is name of URDF file.  This backwards compatibility fallback will be removed in the future.
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link zed_camera_center had 2 children
[robot_state_publisher-1] Link zed_left_camera_frame had 1 children
[robot_state_publisher-1] Link zed_left_camera_optical_frame had 0 children
[robot_state_publisher-1] Link zed_right_camera_frame had 1 children
[robot_state_publisher-1] Link zed_right_camera_optical_frame had 0 children
[robot_state_publisher-1] 1625384781.783259491 [zed.zed_state_publisher] [INFO] got segment base_link
[robot_state_publisher-1] 1625384781.783272828 [zed.zed_state_publisher] [INFO] got segment zed_camera_center
[robot_state_publisher-1] 1625384781.783277583 [zed.zed_state_publisher] [INFO] got segment zed_left_camera_frame
[robot_state_publisher-1] 1625384781.783281676 [zed.zed_state_publisher] [INFO] got segment zed_left_camera_optical_frame
[robot_state_publisher-1] 1625384781.783285743 [zed.zed_state_publisher] [INFO] got segment zed_right_camera_frame
[robot_state_publisher-1] 1625384781.783289415 [zed.zed_state_publisher] [INFO] got segment zed_right_camera_optical_frame
[zed_wrapper-2] 1625384781.789399166 [zed.zed_node] [INFO] ********************************
[zed_wrapper-2] 1625384781.789462103 [zed.zed_node] [INFO]       ZED Camera Component 
[zed_wrapper-2] 1625384781.789469611 [zed.zed_node] [INFO] ********************************
[zed_wrapper-2] 1625384781.789474704 [zed.zed_node] [INFO]  * namespace: /zed
[zed_wrapper-2] 1625384781.789479890 [zed.zed_node] [INFO]  * node name: zed_node
[zed_wrapper-2] 1625384781.789483866 [zed.zed_node] [INFO] ********************************
[zed_wrapper-2] 1625384781.789512103 [zed.zed_node] [INFO] *** GENERAL parameters ***
[zed_wrapper-2] 1625384781.789534464 [zed.zed_node] [INFO]  * Camera model: zed - ZED
[zed_wrapper-2] 1625384781.789549530 [zed.zed_node] [INFO]  * SDK Verbose: 1
[zed_wrapper-2] 1625384781.789563612 [zed.zed_node] [INFO]  * SVO: 
[zed_wrapper-2] 1625384781.789575774 [zed.zed_node] [INFO]  * SVO Loop: FALSE
[zed_wrapper-2] 1625384781.789586041 [zed.zed_node] [INFO]  * SVO Realtime: FALSE
[zed_wrapper-2] 1625384781.789597404 [zed.zed_node] [INFO]  * Camera name: zed
[zed_wrapper-2] 1625384781.789614644 [zed.zed_node] [INFO]  * Camera ID: 0
[zed_wrapper-2] 1625384781.789626354 [zed.zed_node] [INFO]  * Camera SN: 0
[zed_wrapper-2] 1625384781.789638256 [zed.zed_node] [INFO]  * Camera timeout [sec]: 5
[zed_wrapper-2] 1625384781.789650035 [zed.zed_node] [INFO]  * Camera reconnection temptatives: 5
[zed_wrapper-2] 1625384781.789661683 [zed.zed_node] [INFO]  * Camera framerate: 30
[zed_wrapper-2] 1625384781.789675082 [zed.zed_node] [INFO]  * GPU ID: -1
[zed_wrapper-2] 1625384781.789687758 [zed.zed_node] [INFO]  * Camera resolution: 2 - HD720
[zed_wrapper-2] 1625384781.789699141 [zed.zed_node] [INFO]  * Camera self calibration: TRUE
[zed_wrapper-2] 1625384781.789709615 [zed.zed_node] [INFO]  * Camera flip: FALSE
[zed_wrapper-2] 1625384781.789721960 [zed.zed_node] [INFO]  * [DYN] Publish framerate [Hz]: 15 
[zed_wrapper-2] 1625384781.789731509 [zed.zed_node] [INFO] *** VIDEO parameters ***
[zed_wrapper-2] 1625384781.789745016 [zed.zed_node] [INFO]  * Use old extrinsic parameters: 0
[zed_wrapper-2] 1625384781.789756856 [zed.zed_node] [INFO]  * [DYN] Image downsample factor: 0.5 
[zed_wrapper-2] 1625384781.789769387 [zed.zed_node] [INFO]  * [DYN] Brightness: 4
[zed_wrapper-2] 1625384781.789780962 [zed.zed_node] [INFO]  * [DYN] Contrast: 4
[zed_wrapper-2] 1625384781.789791994 [zed.zed_node] [INFO]  * [DYN] Hue: 0
[zed_wrapper-2] 1625384781.789803007 [zed.zed_node] [INFO]  * [DYN] Saturation: 4
[zed_wrapper-2] 1625384781.789815899 [zed.zed_node] [INFO]  * [DYN] Sharpness: 4
[zed_wrapper-2] 1625384781.789827011 [zed.zed_node] [INFO]  * [DYN] Gamma: 8
[zed_wrapper-2] 1625384781.789838077 [zed.zed_node] [INFO]  * [DYN] Auto Exposure/Gain: TRUE
[zed_wrapper-2] 1625384781.789849334 [zed.zed_node] [INFO]  * [DYN] Exposure: 80
[zed_wrapper-2] 1625384781.789860558 [zed.zed_node] [INFO]  * [DYN] Gain: 80
[zed_wrapper-2] 1625384781.789870893 [zed.zed_node] [INFO]  * [DYN] Auto White Balance: TRUE
[zed_wrapper-2] 1625384781.789884062 [zed.zed_node] [INFO]  * [DYN] White Balance Temperature: 42
[zed_wrapper-2] 1625384781.789897435 [zed.zed_node] [INFO]  * Video QoS History: KEEP_LAST
[zed_wrapper-2] 1625384781.789908664 [zed.zed_node] [INFO]  * Video QoS History depth: 1
[zed_wrapper-2] 1625384781.789919963 [zed.zed_node] [INFO]  * Video QoS Reliability: Unknown QoS value
[zed_wrapper-2] 1625384781.789931112 [zed.zed_node] [INFO]  * Video QoS Durability: VOLATILE
[... cut ...]
[zed_wrapper-2] 1625384783.102567518 [zed.zed_node] [INFO]  * Rotation: {0.000,-0.000,0.000}
[zed_wrapper-2] 1625384783.202645976 [zed.zed_node] [INFO] Initial ZED left camera pose (ZED pos. tracking): 
[zed_wrapper-2] 1625384783.202709534 [zed.zed_node] [INFO]  * T: [0,0.06,0]
[zed_wrapper-2] 1625384783.202737813 [zed.zed_node] [INFO]  * Q: [0,0,0,1]
[zed_wrapper-2] 1625384783.274375093 [zed.zed_node] [INFO] Odometry aligned to last tracking pose
```
</details>

Hope this sheds more light on the issue.

Myzhar, 2021-07-04:
ROS2 requires a bit of attention when subscribing to topics. The Quality of Service (QoS) profiles for publisher and subscriber must be compatible. 
You can read more in the online documentation:
https://www.stereolabs.com/docs/ros2/zed-node/#published-topics

cardboardcode, 2021-07-04:
Hi @Myzhar, thanks for prompt clarification.

Checking out this section, [QOS Profiles](https://www.stereolabs.com/docs/ros2/zed-node/#qos-profiles) and noting this relevant section:
```bash
In order for a connection to be made, all of the policies that affect compatibility must be compatible. For instance, if a publisher-subscriber pair has compatible reliability QoS profiles, but incompatible durability QoS profiles, the connection will not be made.
```

Will continue debugging on my side and update once I have resolved it. No actions required.

Myzhar, 2021-07-04:
Can you try to subscribe to the pose or odom topic in a second console window to see if the image topics are published.
I remember to have fixed a similar issue, but I guess that the fix has not yet been propagated from the `dev_foxy` branch to the `master` branch (see https://github.com/stereolabs/zed-ros2-wrapper/issues/31)

cardboardcode, 2021-07-04:
> Can you try to subscribe to the pose or odom topic in a second console window to see if the image topics are published.
> I remember to have fixed a similar issue, but I guess that the fix has not yet been propagated from the `dev_foxy` branch to the `master` branch (see #31)

Subscribing to `/zed/zed_node/pose` by running the commands produce the following terminal output:
```bash
source /opt/ros/foxy/setup.bash
ros2 topic echo /zed/zed_node/pose
```
## Results
```bash 
---
header:
  stamp:
    sec: 1625387715
    nanosec: 434199578
  frame_id: map
pose:
  position:
    x: 0.00027193161196812304
    y: 6.463974714278342e-05
    z: 0.00016781841342351252
  orientation:
    x: 0.0
    y: 0.0002441406286379786
    z: -0.0001726334911185649
    w: 0.9999999552965145
---
```

Subscribing to `/zed/zed_node/odom` by running the commands produce the following terminal output:
```bash
source /opt/ros/foxy/setup.bash
ros2 topic echo /zed/zed_node/pose
```

## Results

<details>

```bash
---
header:
  stamp:
    sec: 1625387812
    nanosec: 442366578
  frame_id: odom
child_frame_id: base_link
pose:
  pose:
    position:
      x: 0.004224781797693515
      y: 0.009861260627486601
      z: 0.007599379259835867
    orientation:
      x: -0.002352972971777034
      y: -0.0011238760447997464
      z: -0.0023442516879424043
      w: 0.9999938524335289
  covariance:
  - 2.2090580387157388e-07
  - -2.606017446282749e-08
  - -1.1616376305312315e-08
  - -3.019323013830899e-09
  - 8.043928856338312e-10
  - -7.591679285212649e-09
  - -2.6060169133756972e-08
  - 2.2970036184233322e-07
  - -5.4664841542262366e-08
  - -4.452547130995299e-08
  - -2.3207821442383647e-08
  - -1.2158380968685378e-07
  - -1.1616370088063377e-08
  - -5.46648522004034e-08
  - 3.1767746122568497e-07
  - 7.791960143777032e-08
  - 1.954197728082363e-07
  - 2.40120936467747e-08
  - -3.0193181288495907e-09
  - -4.452546420452563e-08
  - 7.791963696490711e-08
  - 1.3098947704293096e-07
  - 2.781641761373521e-08
  - 1.7875056101956943e-08
  - 8.04398603282408e-10
  - -2.320784453502256e-08
  - 1.9542002860362118e-07
  - 2.781642649551941e-08
  - 1.453214224511612e-07
  - 9.875265938319444e-09
  - -7.591679285212649e-09
  - -1.2158368178916135e-07
  - 2.4012079435919986e-08
  - 1.7875045443815907e-08
  - 9.87525083928631e-09
  - 7.876673180362559e-08
twist:
  twist:
    linear:
      x: 0.0
      y: 0.0
      z: 0.0
    angular:
      x: 0.0
      y: 0.0
      z: 0.0
  covariance:
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
---
```
</details>

## :warning: 
Subscribing to `/zed/zed_node/left/image_rect_color` still shows **no image** being published on it.

cardboardcode, 2021-07-04:
> Hi @Myzhar, thanks for prompt clarification.
> 
> Checking out this section, [QOS Profiles](https://www.stereolabs.com/docs/ros2/zed-node/#qos-profiles), and noting this relevant section:
> 
> > In order for a connection to be made, all of the policies that affect compatibility must be compatible. For instance, if a publisher-subscriber pair has compatible reliability QoS profiles, but incompatible durability QoS profiles, the connection will not be made.
> 
> Will continue debugging on my side and update once I have resolved it. No actions required.

### Attempted Solution [Making QOS Policies Compatible]
The following shows an example simple ROS2 subscriber node that subscribes to `/zed/zed_node/left/image_rect_color` with a compatible QOS Policy.

### ZED QOS Policy Description
[[Referenced from ZED docs]()]
> History: KEEP_LAST
Depth: 1
Reliability: RELIABLE
Durability: VOLATILE

Running `$ ros2 topic info /zed/zed_node/left/image_rect_color --verbose` produces this terminal output:

```bash 
Type: sensor_msgs/msg/Image

Publisher count: 1

Node name: zed_node
Node namespace: /zed
Topic type: sensor_msgs/msg/Image
Endpoint type: PUBLISHER
GID: 01.0f.d4.06.d3.64.00.00.01.00.00.00.00.00.34.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RMW_QOS_POLICY_RELIABILITY_BEST_EFFORT
  Durability: RMW_QOS_POLICY_DURABILITY_VOLATILE
  Lifespan: 2147483651294967295 nanoseconds
  Deadline: 2147483651294967295 nanoseconds
  Liveliness: RMW_QOS_POLICY_LIVELINESS_AUTOMATIC
  Liveliness lease duration: 2147483651294967295 nanoseconds

Subscription count: 0
``` 
## Writing simple ROS2 Subscriber Node with custom QOS Policy
Follow the instructions below to create this.
1. Reference this [ROS2 Simple Subscriber Tutorial](https://docs.ros.org/en/foxy/Tutorials/Writing-A-Simple-Py-Publisher-And-Subscriber.html) to create a simple ROS2 package.

2. Replace `subscriber_member_function.cpp`'s content with the code attached below.

```cpp
#include <functional>
#include <memory>

#include "rmw/types.h"
#include "rclcpp/rclcpp.hpp"
#include "std_msgs/msg/string.hpp"
#include "sensor_msgs/msg/image.hpp"

static const rmw_qos_profile_t my_custom_qos_profile =
{
    RMW_QOS_POLICY_HISTORY_KEEP_LAST,
    1,
    RMW_QOS_POLICY_RELIABILITY_RELIABLE,
    RMW_QOS_POLICY_DURABILITY_VOLATILE,
    RMW_QOS_DEADLINE_DEFAULT,
    RMW_QOS_LIFESPAN_DEFAULT,
    RMW_QOS_POLICY_LIVELINESS_SYSTEM_DEFAULT,
    RMW_QOS_LIVELINESS_LEASE_DURATION_DEFAULT,
    false
};

class MinimalSubscriber : public rclcpp::Node
{
public:
  MinimalSubscriber()
  : Node("minimal_subscriber")
  {
    auto qos = rclcpp::QoS(
    rclcpp::QoSInitialization::from_rmw(my_custom_qos_profile),
    my_custom_qos_profile);

    auto callback =
      [this](const sensor_msgs::msg::Image::SharedPtr msg) {
        RCLCPP_INFO(this->get_logger(), "Image Received.");
      };

    subscription_ = create_subscription<sensor_msgs::msg::Image>("/zed/zed_node/left/image_rect_color", qos, callback);
  }

private:
  rclcpp::Subscription<sensor_msgs::msg::Image>::SharedPtr subscription_;
};

int main(int argc, char * argv[])
{
  rclcpp::init(argc, argv);
  rclcpp::spin(std::make_shared<MinimalSubscriber>());
  rclcpp::shutdown();
  return 0;
}

```
3. Replace the `CMakeLists.txt`'s content with the following content:
```txt
cmake_minimum_required(VERSION 3.5)
project(cpp_pubsub)

# Default to C99
if(NOT CMAKE_C_STANDARD)
  set(CMAKE_C_STANDARD 99)
endif()

# Default to C++14
if(NOT CMAKE_CXX_STANDARD)
  set(CMAKE_CXX_STANDARD 14)
endif()

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

# find dependencies
find_package(ament_cmake REQUIRED)
find_package(rclcpp REQUIRED)
find_package(std_msgs REQUIRED)
find_package(sensor_msgs REQUIRED)
find_package(rmw REQUIRED)

include_directories(
  include
)

if(BUILD_TESTING)
  find_package(ament_lint_auto REQUIRED)

  ament_lint_auto_find_test_dependencies()
endif()

add_executable(listener src/subscriber_member_function.cpp)
ament_target_dependencies(listener rclcpp std_msgs sensor_msgs rmw)

install(TARGETS
  listener
  DESTINATION lib/${PROJECT_NAME})

ament_package()

```

4. Run the commands below to build and run this simple ROS2 package.
```bash
cd <file path to this package root directory>
source /opt/ros/foxy/setup.bash
colcon build
source install/setup.bash
ros2 run cpp_pubsub listener 
```

## **Results/Conclusion**
Even with compatible QOS policies, the connection still cannot be made and thus unable to subscribe to `/zed/zed_node/left/image_rect_color`.

@Myzhar Please advise and let me know if you require further details.

Myzhar, 2021-07-04:
I will test it in the next hours and I will be back to you with a solution 👍

kquigley29, 2021-07-05:
Hi. I'm having the exact same problem. Would be great to know if there are any updates :)

Myzhar, 2021-07-05:
@kquigley29 I'm working on the fix. It should be available in the `master` branch (Eloquent) in a few hours.
It is already available for the Foxy distribution in the `dev_foxy` branch

Myzhar, 2021-07-05:
@cardboardcode @kquigley29 I just merged an important update in the master branch that should solve this issue and adds a few other features.
Can you please test if and confirm that your issue is solved? Thank you

cardboardcode, 2021-07-05:
@Myzhar 

It **still does not work** for **Foxy distribution**. :sob: 

I redid the steps to reproduce the error but with the latest **dev_foxy** branch codebase for **zed_ros2_wrapper**. 
<details>
<summary>Steps Taken</summary>

## Setup

```bash
cd $HOME
mkdir -p ~/zed_ros2_ws/src && cd ~/zed_ros2_ws/src
git clone https://github.com/stereolabs/zed-ros2-wrapper.git --single-branch --branch dev_foxy --depth 1
git clone https://github.com/stereolabs/zed-ros2-examples.git --single-branch --branch dev_foxy --depth 1
cd ~/zed_ros2_ws/
source /opt/ros/foxy/setup.bash
colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release
# Launch zed_wrapper
ros2 launch zed_wrapper zed.launch.py
```
## Verify if issue is fixed.

```bash
#Open up new terminal
source /opt/ros/foxy/setup.bash
ros2 topic echo /zed/zed_node/left/image_rect_color
```
</details>

## Results :warning: 
`ros2 topic echo /zed/zed_node/left/image_rect_color` still shows no image being published. 

## Others
I can confirm that I am using the latest commit on `dev_foxy` branch. 

The commit number is [1debf24f5ae2aa5f51758f8d57882628c0257df5](https://github.com/stereolabs/zed-ros2-wrapper/commit/1debf24f5ae2aa5f51758f8d57882628c0257df5).
[image]

Myzhar, 2021-07-05:
@cardboardcode I just noticed that you are using a ZED camera. I tested it with a ZED2 and it worked correctly.
I will test it with a ZED in the next few hours and I will fix the eventual issue with it.

griz1112, 2021-07-12:
I had the same issue change the reliability from Reliable to Best Effort in rviz2

cardboardcode, 2021-07-13:
Hi @griz1112,

Thanks for the suggestion. Gonna try it out and verify if it works for my setup.

griz1112, 2021-07-13:
If you do a ros2 topic list do you see the topics? If not do ros2 daemon stop  ros2 daemon start.

an99990, 2021-07-19:
Hi @griz1112 
I am able to see the topics but like @cardboardcode.
I have the same issue with the ZED2 @Myzhar.
[image]

Please advise.

griz1112, 2021-07-19:
Try resetting the daemon. Ros2 daemon stop ros2 daemon start

an99990, 2021-07-19:
Thank you for the reply @griz1112.

I just tried. Doesnt change anything.
[image]

[image]

I just noticed that ros2 topic list isnt always the same: 
[image]

griz1112, 2021-07-20:
Do you have ROS_DOMAIN_ID set on both machines?

Myzhar, 2021-07-20:
@an99990 I just tested the `master` branch and everything works as expected. The `dev_foxy` branch is no more available and it has been merged into the `master` branch that now is Foxy compatible and working.
A new `devel_foxy` branch is currently under development, but it's not stable because we are adding new features.

an99990, 2021-07-20:
Hi @Myzhar,
I just build from master and I am still unavailable to `ros2 topic echo /zed2/zed_node/left/image_rect_color`
message from the camera when it opens:
```
_node] [INFO]  * OpenNI mode (16bit point cloud): FALSE
[zed_wrapper-2] 1626783204.972984088 [zed2.zed_node] [INFO]  * [DYN] Point cloud rate [Hz]: 10
[zed_wrapper-2] 1626783204.972997564 [zed2.zed_node] [INFO]  * [DYN] Depth Confidence: 50
[zed_wrapper-2] 1626783204.973015056 [zed2.zed_node] [INFO]  * [DYN] Depth Texture Confidence: 100
[zed_wrapper-2] 1626783204.973026047 [zed2.zed_node] [INFO]  * Depth QoS History: KEEP_LAST
[zed_wrapper-2] 1626783204.973042257 [zed2.zed_node] [INFO]  * Depth QoS History depth: 1
[zed_wrapper-2] 1626783204.973053288 [zed2.zed_node] [INFO]  * Depth QoS Reliability: BEST_EFFORT
[zed_wrapper-2] 1626783204.973065471 [zed2.zed_node] [INFO]  * Depth QoS Durability: VOLATILE
[zed_wrapper-2] 1626783204.973071853 [zed2.zed_node] [INFO] *** POSITIONAL TRACKING parameters ***
[zed_wrapper-2] 1626783204.973105696 [zed2.zed_node] [INFO]  * Positional tracking enabled: TRUE
[zed_wrapper-2] 1626783204.973119673 [zed2.zed_node] [INFO]  * Base frame id: base_link
[zed_wrapper-2] 1626783204.973135582 [zed2.zed_node] [INFO]  * Map frame id: map
[zed_wrapper-2] 1626783204.973151222 [zed2.zed_node] [INFO]  * Odometry frame id: odom
[zed_wrapper-2] 1626783204.973167162 [zed2.zed_node] [INFO]  * Broadcast Odometry TF: TRUE
[zed_wrapper-2] 1626783204.973181499 [zed2.zed_node] [INFO]  * Broadcast Pose TF: TRUE
[zed_wrapper-2] 1626783204.973194854 [zed2.zed_node] [INFO]  * Broadcast Static IMU TF [not for ZED]: TRUE
[zed_wrapper-2] 1626783204.973213358 [zed2.zed_node] [INFO]  * [DYN] Path publishing rate: 2
[zed_wrapper-2] 1626783204.973225822 [zed2.zed_node] [INFO]  * Path history lenght: -1
[zed_wrapper-2] 1626783204.973242864 [zed2.zed_node] [INFO]  * Initial pose: [0,0,0,0,0,0,]
[zed_wrapper-2] 1626783204.973259254 [zed2.zed_node] [INFO]  * Area Memory: TRUE
[zed_wrapper-2] 1626783204.973271267 [zed2.zed_node] [INFO]  * Area Memory DB: 
[zed_wrapper-2] 1626783204.973289090 [zed2.zed_node] [INFO]  * IMU Fusion [not for ZED]: TRUE
[zed_wrapper-2] 1626783204.973306603 [zed2.zed_node] [INFO]  * Floor Alignment: FALSE
[zed_wrapper-2] 1626783204.973318716 [zed2.zed_node] [INFO]  * Init Odometry with first valid pose data: TRUE
[zed_wrapper-2] 1626783204.973331089 [zed2.zed_node] [INFO]  * 2D mode: FALSE
[zed_wrapper-2] 1626783204.973350495 [zed2.zed_node] [INFO]  * Pose/Odometry QoS History: KEEP_LAST
[zed_wrapper-2] 1626783204.973361516 [zed2.zed_node] [INFO]  * Pose/Odometry QoS History depth: 1
[zed_wrapper-2] 1626783204.973372106 [zed2.zed_node] [INFO]  * Pose/Odometry QoS Reliability: BEST_EFFORT
[zed_wrapper-2] 1626783204.973383768 [zed2.zed_node] [INFO]  * Pose/Odometry QoS Durability: VOLATILE
[zed_wrapper-2] 1626783204.973390360 [zed2.zed_node] [INFO] *** SENSORS STACK parameters ***
[zed_wrapper-2] 1626783204.973401541 [zed2.zed_node] [INFO]  * Sensors Camera Sync: FALSE
[zed_wrapper-2] 1626783204.973414816 [zed2.zed_node] [INFO]  * Sensors publishing rate: 200 Hz
[zed_wrapper-2] 1626783204.973431036 [zed2.zed_node] [INFO]  * Sensors QoS History: KEEP_LAST
[zed_wrapper-2] 1626783204.973440675 [zed2.zed_node] [INFO]  * Sensors QoS History depth: 1
[zed_wrapper-2] 1626783204.973452086 [zed2.zed_node] [INFO]  * Sensors QoS Reliability: BEST_EFFORT
[zed_wrapper-2] 1626783204.973463557 [zed2.zed_node] [INFO]  * Sensors QoS Durability: VOLATILE
[zed_wrapper-2] 1626783204.973471733 [zed2.zed_node] [INFO] *** Spatial Mapping parameters ***
[zed_wrapper-2] 1626783204.973483455 [zed2.zed_node] [INFO]  * Spatial Mapping Enabled: FALSE
[zed_wrapper-2] 1626783204.973498984 [zed2.zed_node] [INFO]  * Spatial Mapping resolution [m]: 0.1
[zed_wrapper-2] 1626783204.973515094 [zed2.zed_node] [INFO]  * 3D Max Mapping range [m]: 20
[zed_wrapper-2] 1626783204.973528569 [zed2.zed_node] [INFO]  * Map publishing rate [Hz]: 0.5
[zed_wrapper-2] 1626783204.973544780 [zed2.zed_node] [INFO]  * Sensors QoS History: KEEP_LAST
[zed_wrapper-2] 1626783204.973559357 [zed2.zed_node] [INFO]  * Sensors QoS History depth: 1
[zed_wrapper-2] 1626783204.973570017 [zed2.zed_node] [INFO]  * Sensors QoS Reliability: BEST_EFFORT
[zed_wrapper-2] 1626783204.973580667 [zed2.zed_node] [INFO]  * Sensors QoS Durability: VOLATILE
[zed_wrapper-2] 1626783204.973595725 [zed2.zed_node] [INFO] *** OBJECT DETECTION parameters ***
[zed_wrapper-2] 1626783204.973607808 [zed2.zed_node] [INFO]  * Object Detection enabled: FALSE
[zed_wrapper-2] 1626783204.973622726 [zed2.zed_node] [INFO]  * OD min. confidence: 50
[zed_wrapper-2] 1626783204.973637343 [zed2.zed_node] [INFO]  * Object Detection model: 0 - MULTI CLASS BOX FAST
[zed_wrapper-2] 1626783204.973649436 [zed2.zed_node] [INFO]  * MultiClassBox people: TRUE
[zed_wrapper-2] 1626783204.973667430 [zed2.zed_node] [INFO]  * MultiClassBox vehicles: TRUE
[zed_wrapper-2] 1626783204.973682278 [zed2.zed_node] [INFO]  * MultiClassBox bags: TRUE
[zed_wrapper-2] 1626783204.973699179 [zed2.zed_node] [INFO]  * MultiClassBox animals: TRUE
[zed_wrapper-2] 1626783204.973717073 [zed2.zed_node] [INFO]  * MultiClassBox electronics: TRUE
[zed_wrapper-2] 1626783204.973736800 [zed2.zed_node] [INFO]  * MultiClassBox fruits and vegetables: TRUE
[zed_wrapper-2] 1626783204.973748963 [zed2.zed_node] [INFO]  * Skeleton fitting: FALSE
[zed_wrapper-2] 1626783204.973765003 [zed2.zed_node] [INFO]  * Obj. Det. QoS History: KEEP_LAST
[... cut ...]
[zed_wrapper-2] 1626783216.738441408 [zed2.zed_node] [INFO]  * Rotation: {0.000,-0.000,0.000}
[zed_wrapper-2] 1626783216.838521661 [zed2.zed_node] [INFO] Initial ZED left camera pose (ZED pos. tracking): 
[zed_wrapper-2] 1626783216.838570754 [zed2.zed_node] [INFO]  * T: [0,0.06,0]
[zed_wrapper-2] 1626783216.838589449 [zed2.zed_node] [INFO]  * Q: [0,0,0,1]
[zed_wrapper-2] 1626783216.961827614 [zed2.zed_node] [INFO] Odometry aligned to last tracking pose
```
Steps : 
I build from master following the installation instruction
ros2 launch zed_wrapper zed2.launch.py

Myzhar, 2021-07-20:
@an99990 I just replicated the steps above and everything works correctly.
Please try by deleting the `install` `build` and `log` folders before compiling.
After compiling close the terminal and open it again, then start the nodes

an99990, 2021-07-20:
by deleting install i cant source `~/.bashrc` and cant call ros2 launch zed_wrapper zed2.launch.py
I am confused with the steps you are asking me.
I try deleting everything recompiling , deleting ..

XDGFX, 2021-08-06:
I'm having the same issue.

Using the Dockerfile mentioned in https://github.com/stereolabs/zed-ros2-wrapper/issues/48#issuecomment-892700306, running on a Jetson Xavier NX.

Steps to reproduce:
- Start container privileged
- Launch using default params `ros2 launch zed_wrapper zed2.launch.py`

Logs look fine:
```log
root@d05ae10340fd:/# ros2 launch zed_wrapper zed2.launch.py 
[INFO] [launch]: All log files can be found below /root/.ros/log/2021-08-06-13-58-03-135339-d05ae10340fd-69
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [73]
[INFO] [zed_wrapper-2]: process started with pid [74]
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link zed2_camera_center had 4 children
[robot_state_publisher-1] Link zed2_baro_link had 0 children
[robot_state_publisher-1] Link zed2_left_camera_frame had 2 children
[robot_state_publisher-1] Link zed2_left_camera_optical_frame had 0 children
[robot_state_publisher-1] Link zed2_temp_left_link had 0 children
[robot_state_publisher-1] Link zed2_mag_link had 0 children
[robot_state_publisher-1] Link zed2_right_camera_frame had 2 children
[robot_state_publisher-1] Link zed2_right_camera_optical_frame had 0 children
[robot_state_publisher-1] Link zed2_temp_right_link had 0 children
[robot_state_publisher-1] 1628258284.473839233 [zed2.zed_state_publisher] [INFO] got segment base_link
[robot_state_publisher-1] 1628258284.474306472 [zed2.zed_state_publisher] [INFO] got segment zed2_baro_link
[robot_state_publisher-1] 1628258284.474384745 [zed2.zed_state_publisher] [INFO] got segment zed2_camera_center
[robot_state_publisher-1] 1628258284.474437162 [zed2.zed_state_publisher] [INFO] got segment zed2_left_camera_frame
[robot_state_publisher-1] 1628258284.474485898 [zed2.zed_state_publisher] [INFO] got segment zed2_left_camera_optical_frame
[robot_state_publisher-1] 1628258284.474535083 [zed2.zed_state_publisher] [INFO] got segment zed2_mag_link
[robot_state_publisher-1] 1628258284.474583628 [zed2.zed_state_publisher] [INFO] got segment zed2_right_camera_frame
[robot_state_publisher-1] 1628258284.474675725 [zed2.zed_state_publisher] [INFO] got segment zed2_right_camera_optical_frame
[robot_state_publisher-1] 1628258284.474752782 [zed2.zed_state_publisher] [INFO] got segment zed2_temp_left_link
[robot_state_publisher-1] 1628258284.474827887 [zed2.zed_state_publisher] [INFO] got segment zed2_temp_right_link
[zed_wrapper-2] 1628258284.609498586 [zed2.zed_node] [INFO] ********************************
[zed_wrapper-2] 1628258284.610072451 [zed2.zed_node] [INFO]       ZED Camera Component 
[zed_wrapper-2] 1628258284.610151748 [zed2.zed_node] [INFO] ********************************
[zed_wrapper-2] 1628258284.610207301 [zed2.zed_node] [INFO]  * namespace: /zed2
[zed_wrapper-2] 1628258284.610257765 [zed2.zed_node] [INFO]  * node name: zed_node
[zed_wrapper-2] 1628258284.610305062 [zed2.zed_node] [INFO] ********************************
[zed_wrapper-2] 1628258284.610870318 [zed2.zed_node] [INFO] *** DEBUG parameters ***
[zed_wrapper-2] 1628258284.610975120 [zed2.zed_node] [INFO] *** GENERAL parameters ***
[zed_wrapper-2] 1628258284.611556024 [zed2.zed_node] [INFO]  * Camera model: zed2 - ZED 2
[zed_wrapper-2] 1628258284.612071488 [zed2.zed_node] [INFO]  * SDK Verbose: 1
[zed_wrapper-2] 1628258284.612644392 [zed2.zed_node] [INFO]  * SVO: 
[zed_wrapper-2] 1628258284.612979981 [zed2.zed_node] [INFO]  * SVO Loop: FALSE
[zed_wrapper-2] 1628258284.613374899 [zed2.zed_node] [INFO]  * SVO Realtime: FALSE
[zed_wrapper-2] 1628258284.613681879 [zed2.zed_node] [INFO]  * Camera name: zed2
[zed_wrapper-2] 1628258284.614314784 [zed2.zed_node] [INFO]  * Camera ID: 0
[zed_wrapper-2] 1628258284.614642085 [zed2.zed_node] [INFO]  * Camera SN: 0
[zed_wrapper-2] 1628258284.615059243 [zed2.zed_node] [INFO]  * Camera timeout [sec]: 5
[zed_wrapper-2] 1628258284.615361232 [zed2.zed_node] [INFO]  * Camera reconnection temptatives: 5
[zed_wrapper-2] 1628258284.615661940 [zed2.zed_node] [INFO]  * Camera framerate: 30
[zed_wrapper-2] 1628258284.615969048 [zed2.zed_node] [INFO]  * GPU ID: -1
[zed_wrapper-2] 1628258284.616286973 [zed2.zed_node] [INFO]  * Camera resolution: 2 - HD720
[zed_wrapper-2] 1628258284.616666947 [zed2.zed_node] [INFO]  * Camera self calibration: TRUE
[zed_wrapper-2] 1628258284.616977319 [zed2.zed_node] [INFO]  * Camera flip: FALSE
[zed_wrapper-2] 1628258284.617282220 [zed2.zed_node] [INFO]  * [DYN] Publish framerate [Hz]: 15 
[zed_wrapper-2] 1628258284.617375245 [zed2.zed_node] [INFO] *** VIDEO parameters ***
[zed_wrapper-2] 1628258284.617673169 [zed2.zed_node] [INFO]  * Use old extrinsic parameters: 0
[zed_wrapper-2] 1628258284.618012854 [zed2.zed_node] [INFO]  * [DYN] Image downsample factor: 0.5 
[zed_wrapper-2] 1628258284.618339163 [zed2.zed_node] [INFO]  * [DYN] Brightness: 4
[zed_wrapper-2] 1628258284.618680256 [zed2.zed_node] [INFO]  * [DYN] Contrast: 4
[zed_wrapper-2] 1628258284.618977028 [zed2.zed_node] [INFO]  * [DYN] Hue: 0
[zed_wrapper-2] 1628258284.619313321 [zed2.zed_node] [INFO]  * [DYN] Saturation: 4
[zed_wrapper-2] 1628258284.619606701 [zed2.zed_node] [INFO]  * [DYN] Sharpness: 4
[zed_wrapper-2] 1628258284.619928274 [zed2.zed_node] [INFO]  * [DYN] Gamma: 8
[zed_wrapper-2] 1628258284.620232503 [zed2.zed_node] [INFO]  * [DYN] Auto Exposure/Gain: TRUE
[zed_wrapper-2] 1628258284.620604316 [zed2.zed_node] [INFO]  * [DYN] Exposure: 80
[... 180 lines cut ...]
[... cut ...]
    local-name: image_common/camera_calibration_parsers
    uri: https://github.com/ros2-gbp/image_common-release.git
    version: release/foxy/camera_calibration_parsers/2.3.0-1
- git:
    local-name: image_common/camera_info_manager
    uri: https://github.com/ros2-gbp/image_common-release.git
    version: release/foxy/camera_info_manager/2.3.0-1
- git:
    local-name: image_common/image_transport
    uri: https://github.com/ros2-gbp/image_common-release.git
    version: release/foxy/image_transport/2.3.0-1
```

Myzhar, 2021-08-06:
This is an old issue that was fixed in the latest versions of the ZED ROS2 wrapper.
Are you using the latest version available in the `master` branch?

XDGFX, 2021-08-06:
I just re-built the container to make sure it wasn't caching an old version of the repo but same issue.

Myzhar, 2021-08-06:
Is the problem "inside" or "outside" the container?
Have you checked the network configuration?

XDGFX, 2021-08-06:
Inside this time :)

I've tried host and bridge networking in testing, but shouldn't affect it as inside the container I can echo `/zed2/zed_node/stereo/image_rect_color`

XDGFX, 2021-08-12:
After some more testing it seems that this isn't confined to image topics, I also can't echo `/zed2/zed_node/left/camera_info`.

Maybe this is a separate issue? It can detect the publishers and topic type when using `ros2 topic info`

```
root@d34f757e4b5a:/# ros2 topic info /zed2/zed_node/left/camera_info --verbose
Type: sensor_msgs/msg/CameraInfo

Publisher count: 2

Node name: zed_node
Node namespace: /zed2
Topic type: sensor_msgs/msg/CameraInfo
Endpoint type: PUBLISHER
GID: 01.0f.70.b7.57.00.00.00.01.00.00.00.00.00.32.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RMW_QOS_POLICY_RELIABILITY_RELIABLE
  Durability: RMW_QOS_POLICY_DURABILITY_VOLATILE
  Lifespan: 2147483651294967295 nanoseconds
  Deadline: 2147483651294967295 nanoseconds
  Liveliness: RMW_QOS_POLICY_LIVELINESS_AUTOMATIC
  Liveliness lease duration: 2147483651294967295 nanoseconds

Node name: zed_node
Node namespace: /zed2
Topic type: sensor_msgs/msg/CameraInfo
Endpoint type: PUBLISHER
GID: 01.0f.70.b7.57.00.00.00.01.00.00.00.00.00.34.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RMW_QOS_POLICY_RELIABILITY_RELIABLE
  Durability: RMW_QOS_POLICY_DURABILITY_VOLATILE
  Lifespan: 2147483651294967295 nanoseconds
  Deadline: 2147483651294967295 nanoseconds
  Liveliness: RMW_QOS_POLICY_LIVELINESS_AUTOMATIC
  Liveliness lease duration: 2147483651294967295 nanoseconds

Subscription count: 0
```

I've pushed the container to Dockerhub if it's any help: https://hub.docker.com/repository/docker/xdgfx/zed_ros2_node_testing. arm only as I'm running on a Jetson.

I'm open to try different thing to see if I can get it working, maybe something doesn't build correctly with the current setup?

grassjelly, 2021-09-14:
Hi,

I think I'm having the same issue on Zed Camera. Is there a fix available yet? 

Thank you!

grassjelly, 2021-09-14:
> Hi,
> 
> I think I'm having the same issue on Zed Camera. Is there a fix available yet?
> 
> Thank you!

This solved my issue: https://github.com/stereolabs/zed-ros2-wrapper#image-transport-and-topic-subscriptions. Thanks

closed 2021-12-07 by Myzhar

xjw1996, 2022-05-30:
Until now, the problem of not being able to display images remains unresolved.(Inside the container)
I use a zed2i camera.
Ros2 version for foxy.
I clone the latest version of the repository.(Latest version of master)
The topic is /zed2i/zed_node/right_raw/image_raw_color
[image]
.

Myzhar, 2022-05-30:
Have you installed `image_common` 3.0.0 as described in the Readme?

xjw1996, 2022-05-31:
Mr. Myzhar
Thank you for your patience and quick reply!
I installed image_common 3.0.0 as you said, and it compiled successfully！
Hope you have a great new day at work!
[image]
[image]

cardboardcode, 2022-06-04:
@Myzhar 
Thank you for the fix. Apologies for the extremely late follow-up. **I can verify that the fix has resolved this issue on my side as well, as of 4th June 2022.** 

## Solution :v:
Follow the instructions below to set up, run and display the ZED2 Camera:

### Dependencies
1. **Ubuntu** `20.04` [Operating System] [[Link](https://releases.ubuntu.com/20.04/)]
2. **CUDA** `11.5` [Nvidia GPU] [[Link](https://developer.nvidia.com/cuda-11-5-0-download-archive)]
3. **CUDNN** `8.2.0` [Nvidia GPU] [Rely on the installation of the ZED SDK below to do this for you automatically.]
4. **ZED SDK** `3.7.4` [SDK][[Link](https://download.stereolabs.com/zedsdk/3.7/cu115/ubuntu20)]
4. ROS2 Foxy

## :warning:
Please ensure that, when installing the ZED SDK, **PLEASE INSTALL THE AI MODULES FOR DEPTH TO WORK**.

### Build
```bash
cd $HOME
mkdir -p ~/zed_ros2_ws/src
cd ~/zed_ros2_ws/src
source /opt/ros/foxy/setup.bash

# Install and build recommended image-common==3.0.0
git clone https://github.com/ros-perception/image_common.git --branch ros2
cd image_common
git reset --hard a62ab876da2599627fe6a86bbc79040c62509422 # Reset to Release Tag: 3.0.0
cd ~/zed_ros2_ws/
colcon build --symlink-install
source install/setup.bash

# Install and build zed_ros2_wrapper
cd ~/zed_ros2_ws/src
git clone  --recursive https://github.com/stereolabs/zed-ros2-wrapper.git --branch master 
cd zed-ros2-wrapper 
git reset --hard 55a4302c6365ab2742e61f4cbfbd34960d3955b2 # Reset to the Release Tag: Foxy-v3.7
cd ~/zed_ros2_ws/
sudo apt-get install ros-foxy-launch --upgrade # Ensure that the latest version of ros-foxy-launch is being used to prevent errors concerning type_utils.py.
rosdep install --from-paths src --ignore-src -r -y
colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release

# Install and build zed_ros2_wrapper
cd ~/zed_ros2_ws/src
git clone https://github.com/stereolabs/zed-ros2-examples.git --branch master
rosdep install --from-paths src --ignore-src -r -y
cd ~/zed_ros2_ws
colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release

``` 
### Run
```bash
cd ~/zed_ros2_ws
source install/setup.bash
ros2 launch zed_display_rviz2 display_zed2.launch.py
```

### Verify 
After completing steps under **Run** section, you should see `RViz2` interface displaying something similar to what is shown below:
[image]

In contrast to the original issue given at the top of this thread, the topic to subscribe has changed over time. **Verified that this issue has been resolved** by running the following terminal command and seeing the following image result appear:

```bash
source /opt/ros/foxy/setup.bash
ros2 topic echo /zed2/zed_node/rgb/image_rect_color
# It is working because I am able to subscribe and display the image stream from ZED2 Camera.
```
## Others
You might notice that the pointcloud generated in the screenshot above is quite bad. Note that calibration has not been done. Please reduce the brightness of your room and do the calibration by using the provided inbuilt tool:

```bash
cd /usr/local/zed/tools
./ZED_Calibration
# Follow the on-screen instructions to calibrate and have better depth estimation from ZED2 Camera.
```
[image]

##### Hope these instructions prove helpful to you, whoever you may be. Please have a nice day. :blush: