stereolabs/zed-ros2-wrapper #44: [BUG]

griz1112, 2021-07-11:
**Describe the bug**
A clear and concise description of what the bug is.

**Device information:**
 - OS: [e.g. Ubuntu]
 - OS version: 18.04
 - CPU: [e.g. i7, Jeston Nano, Intel desktop
 - GPU Nvidia:  jetson, GTX1660
 - ZED SDK Version: 3.5

 - ROS Wrapper version: zed-ros2-wrapper foxy-dev

**To Reproduce**

Steps to reproduce the behavior:
1. Go to zed_wrapper
2. ros2 launch zed_wrapper zedm.launch.py

**Expected behavior**
This error only shows up with the zedm. The ZED1 and ZED2i work fine on both machines.

**Screenshots**

[image]

**Additional context**
Fresh install on both machines. Running Ubuntu 20.04 and ros2 foxy on the Desktop and Ubuntu 18.04 and ros2 foxy in a stereolabs container with ros2 installed as well on the nano. I ran the .yaml file through a asii dump looking for weird characters none were found. Strange problem it should work the other two cameras are fine.

Myzhar, 2021-07-12:
Fixed in `dev_foxy` with latest commit https://github.com/stereolabs/zed-ros2-wrapper/commit/bc2d531d15c64c4d74d7265b9c18b6820681714a

closed 2021-07-12 by Myzhar

griz1112, 2021-07-12:
Thanks Myzhar.