stereolabs/zed-ros2-wrapper #48: [Question] Building foxy zed-ros2-wrapper on 18.04, file INSTALL cannot find...

XDGFX, 2021-08-03:
Hi all,

I'm trying to set up zed-ros2-wrapper on a Jetson running ROS foxy. I'm using the dustynv base container, and installing the ZED SDK, Python API, and wrapper on top.

I don't want to rely on the eloquent branch as all my other containers are running Foxy, and I don't want to risk strange issues down the line with communication between ROS versions.

My Dockerfile is shown below
```Dockerfile
ARG ROS_VERSION=foxy
ARG L4T_MINOR_VERSION=5.0
ARG FROM_IMAGE=dustynv/ros:$ROS_VERSION-ros-base-l4t-r32.${L4T_MINOR_VERSION}
ARG OVERLAY_WS=/opt/ros/overlay_ws

# Base container
FROM $FROM_IMAGE

# After using an arg in a `FROM` line, the arg is lost.
# This will allow it to be used again.
ARG ROS_VERSION
ARG L4T_MINOR_VERSION
ARG OVERLAY_WS

ARG ZED_SDK_MAJOR=3
ARG ZED_SDK_MINOR=5
ARG JETPACK_MAJOR=4
ARG JETPACK_MINOR=5

# Fix for core dump errors when ZED SDK is installed
ENV OPENBLAS_CORETYPE ARMV8

# Update and install dependencies
RUN apt-get update && apt-get install --no-install-recommends -y \
    lsb-release \
    less \
    udev \
    wget \
    apt-transport-https \
    python3-pip

# Fix for numpy installation later on
RUN pip3 install --upgrade pip

# Install ZED SDK
RUN echo "# R32 (release), REVISION: ${L4T_MINOR_VERSION}" > /etc/nv_tegra_release ; \
    wget -q --no-check-certificate -O ZED_SDK_Linux_JP.run https://download.stereolabs.com/zedsdk/${ZED_SDK_MAJOR}.${ZED_SDK_MINOR}/jp${JETPACK_MAJOR}${JETPACK_MINOR}/jetsons && \
    chmod +x ZED_SDK_Linux_JP.run ; ./ZED_SDK_Linux_JP.run silent skip_tools && \
    rm -rf /usr/local/zed/resources/* \
    rm -rf ZED_SDK_Linux_JP.run && \
    rm -rf /var/lib/apt/lists/*

# Install ZED Python API
RUN apt-get update -y && apt-get install --no-install-recommends python3 python3-pip python3-dev gfortran gcc musl-dev python3-setuptools build-essential -y && \
    wget download.stereolabs.com/zedsdk/pyzed -O /usr/local/zed/get_python_api.py && \
    python3 /usr/local/zed/get_python_api.py && \
    python3 -m pip install cython wheel && \
    python3 -m pip install numpy pyopengl *.whl && \
    rm *.whl ; rm -rf /var/lib/apt/lists/*

# Copy source
WORKDIR $OVERLAY_WS
COPY src src

# Build
RUN /bin/bash -c "source /opt/ros/${ROS_VERSION}/install/setup.bash && colcon build --cmake-args=-DCMAKE_BUILD_TYPE=Release"
```

My `src` folder contains the release version of the wrapper, as well as the ros diagnostics package as I got errors that it was missing before (https://github.com/ros/diagnostics/tree/foxy).

I'm getting the following error when building:

```
Step 24/24 : RUN /bin/bash -c "source /opt/ros/${ROS_VERSION}/install/setup.bash && colcon build --cmake-args=-DCMAKE_BUILD_TYPE=Release"
 ---> Running in 4a555d185645
Starting >>> diagnostic_updater
Starting >>> zed_interfaces
Starting >>> diagnostic_aggregator
[Processing: diagnostic_aggregator, diagnostic_updater, zed_interfaces]
[Processing: diagnostic_aggregator, diagnostic_updater, zed_interfaces]
[Processing: diagnostic_aggregator, diagnostic_updater, zed_interfaces]
Finished <<< diagnostic_updater [1min 54s]
Starting >>> self_test
[Processing: diagnostic_aggregator, self_test, zed_interfaces]
Finished <<< diagnostic_aggregator [2min 44s]
[Processing: self_test, zed_interfaces]
[Processing: self_test, zed_interfaces]
Finished <<< zed_interfaces [3min 49s]
Starting >>> zed_components
Finished <<< self_test [2min 20s]
--- stderr: zed_components
CMake Error at cmake_install.cmake:61 (file):
  file INSTALL cannot find
  "/opt/ros/overlay_ws/src/zed-ros2-wrapper-foxy-v3.5/zed_components/DIRECTORY".

make: *** [install] Error 1
---
Failed   <<< zed_components [2min 29s, exited with code 2]
[Processing: zed_components]
[Processing: zed_components]
[Processing: zed_components]
[Processing: zed_components]

Summary: 4 packages finished [6min 19s]
  1 package failed: zed_components
  1 package had stderr output: zed_components
  2 packages not processed
The command '/bin/bash -c /bin/bash -c "source /opt/ros/${ROS_VERSION}/install/setup.bash && colcon build --cmake-args=-DCMAKE_BUILD_TYPE=Release"' returned a non-zero code: 2
```

I'm not very familiar with C++ and CMakeLists so not too sure what is going wrong. Any help is much appreciated!

Myzhar, 2021-08-03:
Dustin created also a Docker container for ROS2 Foxy that includes the ZED wrapper:
https://github.com/dusty-nv/jetson-containers/blob/master/Dockerfile.ros.slam
You can use that as example

XDGFX, 2021-08-04:
Ah beauty! I adapted that container (cutting out all the stuff I didn't need) and it works perfectly! Many thanks!

My adapted dockerfile for reference if anyone else comes across this:

```Dockerfile
ARG BASE_IMAGE=dustynv/ros:foxy-ros-base-l4t-r32.5.0
FROM ${BASE_IMAGE}

SHELL ["/bin/bash", "-c"]
ENV SHELL /bin/bash

ENV DEBIAN_FRONTEND=noninteractive
ARG MAKEFLAGS=-j$(nproc)
ENV LANG=en_US.UTF-8
ENV PYTHONIOENCODING=utf-8
RUN locale-gen en_US en_US.UTF-8 && update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8

ENV PYTORCH_PATH="/usr/local/lib/python3.6/dist-packages/torch"
ENV LD_LIBRARY_PATH="${PYTORCH_PATH}/lib:${LD_LIBRARY_PATH}"

ARG ROS_ENVIRONMENT=${ROS_ROOT}/install/setup.bash

# Required packages
RUN apt-get update && \
    apt-get install -y --no-install-recommends \
    software-properties-common \
    apt-transport-https \
    ca-certificates \
    gnupg \
    && rm -rf /var/lib/apt/lists/* \
    && apt-get clean

# Fix for "Package 'test_pluginlib' must declare at least one maintainer"
RUN rm -r /opt/ros/foxy/build/pluginlib/prefix/share/test_pluginlib

#
# ZED SDK / zed-ros2-wrapper
# https://github.com/stereolabs/zed-docker/blob/master/3.X/jetpack_4.X/devel/Dockerfile
#
ARG ZED_SDK_URL="https://download.stereolabs.com/zedsdk/3.5/jp45/jetsons"
ARG ZED_SDK_RUN="ZED_SDK_Linux_JP.run"

RUN cd /tmp && \
    wget --quiet --show-progress --progress=bar:force:noscroll --no-check-certificate ${ZED_SDK_URL} -O ${ZED_SDK_RUN} && \
    chmod +x ${ZED_SDK_RUN} && \
    ./${ZED_SDK_RUN} silent skip_tools && \
    rm -rf /usr/local/zed/resources/* && \
    rm -rf ${ZED_SDK_RUN} && \
    rm -rf /var/lib/apt/lists/* && \
    apt-get clean

# Manually pull dependencies for zed-ros2-wrapper
RUN source ${ROS_ENVIRONMENT} && \
    cd ${ROS_ROOT} && \
    mkdir src/slam && \
    rosinstall_generator --deps --exclude-path ${ROS_ROOT}/src --rosdistro ${ROS_DISTRO} \
    diagnostic_updater \
    xacro \
    > ros2.${ROS_DISTRO}.zed.rosinstall && \
    cat ros2.${ROS_DISTRO}.zed.rosinstall && \
    vcs import src/slam < ros2.${ROS_DISTRO}.zed.rosinstall && \
    apt-get update && \
    rosdep install --from-paths src/slam --ignore-src --rosdistro ${ROS_DISTRO} -y --skip-keys "Pangolin libopencv-dev libopencv-contrib-dev libopencv-imgproc-dev python-opencv python3-opencv" && \
    rm -rf /var/lib/apt/lists/* && \
    apt-get clean && \
    colcon build --symlink-install --base-paths src/slam

RUN source ${ROS_ENVIRONMENT} && \
    cd ${ROS_ROOT} && \
    git clone https://github.com/stereolabs/zed-ros2-wrapper src/slam/zed-ros2-wrapper && \
    apt-get update && \
    rosdep install --from-paths src/slam --ignore-src --rosdistro ${ROS_DISTRO} -y --skip-keys "rtabmap find_object_2d Pangolin libopencv-dev libopencv-contrib-dev libopencv-imgproc-dev python-opencv python3-opencv" && \
    rm -rf /var/lib/apt/lists/* && \
    apt-get clean && \
    colcon build --symlink-install --base-paths src/slam/zed-ros2-wrapper --cmake-args=-DCMAKE_BUILD_TYPE=Release
```

closed 2021-08-04 by XDGFX