stereolabs/zed-ros2-wrapper #53: [BUG] No module named 'em'

an99990, 2021-09-02:
**Describe the bug**
I tried to build the repo and got No module named 'em'

**Device information:**
 - OS: Ubuntu
 - OS version: 20.04
 - ZED SDK Version: 3.5
 - ROS Wrapper version: latest master

**To Reproduce**
Steps to reproduce the behavior:
installation instructions given in repo
i then install emy via pip but i still get the error
**Screenshots**
[image]

Myzhar, 2021-09-02:
Have you tried with 
`$ sudo apt-get install python3-empy`
see https://answers.ros.org/question/257757/importerror-no-module-named-em-error/

Myzhar, 2021-09-02:
Be sure to be using Python3:
https://answers.ros.org/question/337135/catkin_make-no-module-named-catkin_pkg/

an99990, 2021-09-02:
yes thank you, i have managed to install the packages and this is the error that does not involve any module
[image]

please help, thanks

an99990, 2021-09-02:
bot sure what happend but i changed the python version and now im good.
Thank you again @Myzhar

closed 2021-09-02 by an99990