stereolabs/zed-ros2-wrapper #55: Unable to build Zed SDK in Nvidia Jetson Docker

peredwardsson, 2021-09-21:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

When building the Zed ROS2 wrapper in a docker image based on an Nvidia docker, the wrapper can not find Cuda, and complains that the version that it finds (10.2) is wrong, even though it explicitly states that it has found the correct version (10.2).

Previous issue https://github.com/stereolabs/zed-ros2-wrapper/issues/18 solves this by downloading SDK v3.3 and sourcing ROS2 Foxy correctly. I fail to see how that would make sense as this dockerfile is essentially untouched ROS-wise before this stage. We are using SDK version 3.5, as is recommended on https://github.com/stereolabs/zed-ros2-wrapper.

Previously, we have mounted folders from host computers into the docker environment and built from that workspace, which has worked fine. We encounter this problem because we are slimming our workspace, and hence moving the Zed stuff to the docker build process.

### Steps to Reproduce

1. On a Jetson AGX Xavier, create a new docker image based on nvcr.io/nvidia/l4t-base:r32.5.0
2. Install Zed SDK 3.5 for Jetpack via
```
RUN apt-get update -y && apt-get install --no-install-recommends lsb-release wget less udev sudo  build-essential cmake -y && \
    wget -q --no-check-certificate -O ZED_SDK_Linux_JP.run https://download.stereolabs.com/zedsdk/3.5/jp45/jetsons && \
    chmod +x ZED_SDK_Linux_JP.run ; ./ZED_SDK_Linux_JP.run silent skip_tools && \
    rm -rf /usr/local/zed/resources/* \
    rm -rf ZED_SDK_Linux_JP.run && \
    rm -rf /var/lib/apt/lists/*
```
3. Hack ROS2 together so that you can run Foxy on a Jetson via https://github.com/dusty-nv/jetson-containers/blob/master/Dockerfile.ros.foxy.
4. Download release version of Zed ROS2 Wrapper from https://github.com/stereolabs/zed-ros2-wrapper/archive/refs/tags/foxy-v3.5.tar.gz
5. `colcon build --merge-install --packages-select-regex zed.* --cmake-args "-DCMAKE_BUILD_TYPE=Release" "-DCMAKE_LIBRARY_PATH=/usr/local/cuda/lib64/stubs" "-DCUDA_CUDART_LIBRARY=/usr/local/cuda/lib64/stubs" "-DCMAKE_CXX_FLAGS='-Wl,--allow-shlib-undefined'"` as per https://github.com/stereolabs/zed-ros2-wrapper/issues/18
...

### Expected Result

Package finds appropriate dependencies and builds happily.

### Actual Result

```
Starting >>> zed_interfaces
Finished <<< zed_interfaces [1.90s]                     
Starting >>> zed_components
--- stderr: zed_components                         
CMake Error at /usr/share/cmake-3.21/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find CUDA (missing: CUDA_CUDART_LIBRARY) (found suitable version
  "10.2", minimum required is "10.2")
Call Stack (most recent call first):
  /usr/share/cmake-3.21/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.21/Modules/FindCUDA.cmake:1264 (find_package_handle_standard_args)
  /usr/local/zed/zed-config.cmake:72 (find_package)
  CMakeLists.txt:39 (find_package)

---
Failed   <<< zed_components [0.79s, exited with code 1]

```

It finds 10.2, yet thinks that it is another version than 10.2. What? 

### ZED Camera model

ZED2

### Environment

```shell
OS: Ubuntu 18.04, from docker image above
CPU: ARM
GPU: Nvidia Jetson AGX Xavier
Other info: ROS2, Foxy. Many packages are built before Zed is built, but they have been left out for brevity. Do tell me if you want a full list.
```

Myzhar, 2021-09-21:
Hi @peredwardsson 
I think this can help you:
https://github.com/stereolabs/zed-ros2-wrapper/issues/48#issuecomment-892700306

peredwardsson, 2021-10-01:
Got it working. Thank you very much.

closed 2021-10-01 by peredwardsson