stereolabs/zed-ros2-wrapper #56: Zed2, ROS2, Jetson: Some topics output, but most do not.

peredwardsson, 2021-09-22:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

On starting the node, all topics that are advertised are displayed. Only a fraction of them actually publish anything. Critically, depth images or stereo images are not published, while IMU and zed_node/stereo/image_rect_color are published.

### Steps to Reproduce

In a Docker environment running on Jetson AGX Xavier, build ZED SDK and Zed Wrapper:
```
ARG BASE_IMAGE=nvcr.io/nvidia/l4t-base:r32.5.0
FROM ${BASE_IMAGE}
[...]
RUN apt-get update -y && apt-get install --no-install-recommends lsb-release wget less udev sudo  build-essential cmake -y && \
    wget -q --no-check-certificate -O ZED_SDK_Linux_JP.run https://download.stereolabs.com/zedsdk/3.5/jp45/jetsons && \
    chmod +x ZED_SDK_Linux_JP.run ; ./ZED_SDK_Linux_JP.run silent skip_tools && \
    rm -rf /usr/local/zed/resources/* \
    rm -rf ZED_SDK_Linux_JP.run && \
    rm -rf /var/lib/apt/lists/*
[...]
ARG ROS_ENVIRONMENT=${ROS_ROOT}/install/setup.bash
RUN source ${ROS_ENVIRONMENT} && \
    cd ${ROS_ROOT} && \
    mkdir src/slam && \
    rosinstall_generator --exclude-path ${ROS_ROOT}/src --rosdistro ${ROS_DISTRO} \
    diagnostic_updater \
    xacro \
    > ros2.${ROS_DISTRO}.zed.rosinstall && \
    cat ros2.${ROS_DISTRO}.zed.rosinstall && \
    vcs import src/slam < ros2.${ROS_DISTRO}.zed.rosinstall && \
    apt-get update && \
    rosdep install --from-paths src/slam --ignore-src --rosdistro ${ROS_DISTRO} -y --skip-keys "Pangolin libopencv-dev libopencv-contrib-dev libopencv-imgproc-dev python-opencv python3-opencv" && \
    rm -rf /var/lib/apt/lists/* && \
    apt-get clean && \
    colcon build --merge-install --base-paths src/slam

RUN source ${ROS_ENVIRONMENT} && \
    cd ${ROS_ROOT} && \
    git clone https://github.com/stereolabs/zed-ros2-wrapper src/slam/zed-ros2-wrapper && \
    apt-get update && \
    rosdep install --from-paths src/slam --ignore-src --rosdistro ${ROS_DISTRO} -y --skip-keys "rtabmap find_object_2d Pangolin libopencv-dev libopencv-contrib-dev libopencv-imgproc-dev python-opencv python3-opencv" && \
    rm -rf /var/lib/apt/lists/* && \
    apt-get clean && \
    colcon build --merge-install --base-paths src/slam/zed-ros2-wrapper --cmake-args=-DCMAKE_BUILD_TYPE=Release
```
2. Start node with the following config:
```
/**:
    ros__parameters:
        general:
            svo_file:                   ''
            svo_loop:                   false                               # Enable loop mode when using an SVO as input source
            svo_realtime:               false                               # if true SVO will be played trying to respect the original framerate eventually skipping frames, otherwise every frame will be processed respecting the `pub_frame_rate` setting
            debug_mode:                 false
            camera_timeout_sec:         5
            camera_max_reconnect:       5
            camera_flip:                false
            zed_id:                     0
            serial_number:              0
            resolution:                 2                                   # '0': HD2K, '1': HD1080, '2': HD720, '3': VGA
            sdk_verbose:                true
            grab_frame_rate:            30                                  # ZED SDK internal grabbing rate
            pub_frame_rate:             15.0                                # [DYNAMIC] - frequency of publishing of visual images and depth images
            gpu_id:                     0
            camera_model:               'zed2'
            camera_name:                'zed2'

[... 413 lines cut ...]
[...]
```
4. Run `ros2 topic hz /sensors/zed_node/depth/depth_registered`. Observe lack of output.
5. Run `ros2 topic hz  /sensors/zed_node/rgb/image_rect_color`. Revel at the correct amount of output.

### Expected Result

I expect that any advertised topic produce data. In particular, that stereo, depth, rectified and associated camera infos are published. 

### Actual Result

Some topics produce data, but most do not. 

### ZED Camera model

ZED2

### Environment

```shell
OS: Ubuntu 18.04, from docker image above
CPU: ARM
GPU: Nvidia Jetson AGX Xavier
Other info: ROS2, Foxy. Many packages are built before Zed is built, but they have been left out for brevity. Do tell me if you want a full list.
```

Myzhar, 2021-09-22:
Hi @peredwardsson 
I just tried `ros2 launch zed_wrapper zed2.launch.py` as double check and it works as expected.
Are you starting it inside the container?
Can you run the command `ros2 doctor` while the node is running to see if there is something wrong in your configuration?

peredwardsson, 2021-09-23:
Yes, running it inside of container. I believe it has worked previously, so I also suspect some configuration issue. `ros2 doctor` output:
```
# ros2 doctor
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/__init__.py: 108: UserWarning: Fail to call PackageCheck class functions.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /actuators/esc/motor_current_axis0.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /actuators/esc/motor_current_axis1.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /actuators/esc/odom_odrive.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /actuators/esc/vbus_voltage.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /actuators/esc/vel_estimate.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 56: UserWarning: Subscriber without publisher detected on /actuators/odom_odrive.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /cloud.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /cloud_ground.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /cloud_map.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /cloud_obstacles.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 56: UserWarning: Subscriber without publisher detected on /cmd_vel.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /diagnostics.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /global_path.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /global_path_nodes.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 56: UserWarning: Subscriber without publisher detected on /global_pose.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 56: UserWarning: Subscriber without publisher detected on /goal.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 56: UserWarning: Subscriber without publisher detected on /goal_node.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /goal_out.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /goal_reached.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 56: UserWarning: Subscriber without publisher detected on /gps/fix.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /grid_prob_map.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 56: UserWarning: Subscriber without publisher detected on /imu.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /info.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 56: UserWarning: Subscriber without publisher detected on /initialpose.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /labels.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /landmarks.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /local_grid_empty.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /local_grid_ground.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /local_grid_obstacle.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /local_path.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /local_path_nodes.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 56: UserWarning: Subscriber without publisher detected on /localization/joint_states.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /localization/odometry/filtered.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /localization/robot_description.
[... cut ...]
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /sensors/zed_node/temperature/left.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /sensors/zed_node/temperature/right.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 56: UserWarning: Subscriber without publisher detected on /user_data_async.

All 3 checks passed
```

Myzhar, 2021-09-23:
It's strange that the depth topic is not present in the warning list.
Can you please post the output of the command
`$ ros2 topic list`?

peredwardsson, 2021-09-23:
It was probably an issue with `image_transport` which is used to create some of the publishers. Observing [the source code](https://github.com/stereolabs/zed-ros2-wrapper/blob/master/zed_components/src/zed_camera/src/zed_camera_component.cpp#L1928), I noticed that topics that had publishers created with that package's publisher creation functions. Reinstalling `image_transport` seems to have fixed it.

closed 2021-09-23 by peredwardsson

Myzhar, 2021-09-23:
`image_transport` is a problem until they do not add the fix in the official release