stereolabs/zed-ros2-wrapper #59: Colcon build fails: 'set_on_parameters_set_callback’ was not declared in this scope

Hlanden, 2021-11-04:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

# Problem Description
Hi, 

I'm unable to build the ros2 wrapper using the given colcon command in the installation guide. I'm getting the error: 

_zed_camera_component.cpp:295:3: error: ‘set_on_parameters_set_callback’ was not declared in this scope_

I've done some research on the topic and it turns out that the function set_on_parameters_set_callback is deprecated in ros21 foxy: https://github.com/ros2/rclcpp/issues/1129

I've tried replacing it with the new function called add_on_set_parameters_callback as described [here](https://docs.ros2.org/foxy/api/rclcpp/classrclcpp_1_1node__interfaces_1_1NodeParameters.html#a1937592341036268cf1e7deb2abc5566), but it did not help. 

# Log Output
```
/home/jorgen/Work/NTNU/ProsjektOppgave/Code/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp: In member function ‘void stereolabs::ZedCamera::initParameters()’:
/home/jorgen/Work/NTNU/ProsjektOppgave/Code/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:295:3: error: ‘set_on_parameters_set_callback’ was not declared in this scope
  295 |   set_on_parameters_set_callback(std::bind(&ZedCamera::callback_paramChange, this, _1));
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /opt/ros/galactic/include/rclcpp/client.hpp:40,
                 from /opt/ros/galactic/include/rclcpp/callback_group.hpp:23,
                 from /opt/ros/galactic/include/rclcpp/any_executable.hpp:20,
                 from /opt/ros/galactic/include/rclcpp/memory_strategy.hpp:25,
                 from /opt/ros/galactic/include/rclcpp/memory_strategies.hpp:18,
                 from /opt/ros/galactic/include/rclcpp/executor_options.hpp:20,
                 from /opt/ros/galactic/include/rclcpp/executor.hpp:36,
                 from /opt/ros/galactic/include/rclcpp/executors/multi_threaded_executor.hpp:26,
                 from /opt/ros/galactic/include/rclcpp/executors.hpp:21,
                 from /opt/ros/galactic/include/rclcpp/rclcpp.hpp:156,
                 from /home/jorgen/Work/NTNU/ProsjektOppgave/Code/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/include/zed_camera_component.hpp:31,
                 from /home/jorgen/Work/NTNU/ProsjektOppgave/Code/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:34:
/home/jorgen/Work/NTNU/ProsjektOppgave/Code/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp: In member function ‘void stereolabs::ZedCamera::getGeneralParams()’:
/home/jorgen/Work/NTNU/ProsjektOppgave/Code/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:371:31: warning: too many arguments for format [-Wformat-extra-args]
  371 |     RCLCPP_WARN(get_logger(), "'pub_frame_rate' cannot be bigger than 'grab_frame_rate'", paramName.c_str());
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/zed_camera_component.dir/build.make:76: CMakeFiles/zed_camera_component.dir/src/zed_camera/src/zed_camera_component.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:79: CMakeFiles/zed_camera_component.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

### Steps to Reproduce

1. Source Ros2 Foxy
2. Run the commands shown in the README file of this repo

### Expected Result

Successful colcon build

### Actual Result

Colcon build finishes with error

### ZED Camera model

ZED2i

### Environment

```shell
OS: Ubuntu 20.04
GPU: Nvidia GTX 660 
ZED SKD: 3.5 (also tried 3.6, gave same error) 
Ros2 distro: Foxy 
Cuda: 11.0
```

closed 2021-11-04 by Hlanden

Myzhar, 2021-11-04:
@Hlanden if you solved the issue can you please provide more info in case other users will face it?

supergirl-os, 2022-01-15:
I have met the same issue with you. How did you solve it?

Myzhar, 2022-01-15:
Are you using Galactic?

supergirl-os, 2022-01-15:
Yes. But I have solved this question. 
Just change the SDK verstion to 3.5 and use the related relased code source. 
Then change 'set_on_parameters_set_callback' to 'add_on_set_parameters_callback'.

Myzhar, 2022-01-15:
Why did you change the SDK to 3.5? The latest v3.6.3 should work and you can get improved stability

supergirl-os, 2022-01-15:
When I changed the function name, I got the error.

[image]
So I try to change the version in order to colcon build.

Myzhar, 2022-01-15:
When you change the SDK version you must always clean the ROS2 workspace cache, otherwise, CMake continues to search for older library versions.
You must delete the `build`, `install`, and `log` folders. I usually use this simple script:
```
#!/bin/bash

rm -rf install
rm -rf build
rm -rf log
```

Hlanden, 2022-01-17:
Sorry for the late reply. 

I had to reinstall Ubuntu 20.04 on my computer for other reasons, and it suddenly worked after the reinstallation...
I'm not really sure why I got this bug...