stereolabs/zed-ros2-wrapper #60: zed-components fails when built with colcon

felrock, 2021-11-12:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Building zed-ros2-wrapper(Foxy - SDK v3.5.x tar ball) with colcon generates and error that a file DIRECTORY is missing. Build process is run in a docker build senario. Building with the latest commit on master was succesful so the issue seems to be isolated to the 3.5 foxy release.

### Steps to Reproduce

Build the github release for foxy v3.5 with colcon/ament_cmake

### Expected Result

There is no file named DIRECTORY and i don't know why CMake looks for it really. The expected results is a successful build.

### Actual Result

Build fails

### ZED Camera model

ZED2

### Environment

```shell
Tegra for linux, jetpack 4.6
Jetson AGX Xavior
Docker
zed-ros2-wrapper release 3.5 for foxy
```

### Anything else?

Console error:

CMake Error at cmake_install.cmake:69 (file):
  file INSTALL cannot find
  "/opt/ros/foxy/src/zed-ros2-wrapper-foxy-v3.5/zed_components/DIRECTORY": No
  such file or directory.

---
Failed   <<< zed_components [2min 58s, exited with code 1]

Myzhar, 2021-11-12:
This has been fixed with this commit:
https://github.com/stereolabs/zed-ros2-wrapper/commit/0527f71a94da504c833dfd7bb812bc4d75b157cc
Let me know if you can compile by modifying the `zed_components/CMakeLists.txt` file as illustrated

felrock, 2021-11-12:
Yes it was possible! Would it be possible to make a cherry pick that change and edit the 3.5 foxy release? If not it's ok to just checkout that commit id instead. Thanks!

Myzhar, 2021-11-12:
A new release with the fix has been tagged:
https://github.com/stereolabs/zed-ros2-wrapper/releases/tag/foxy-v3.5.1

closed 2021-12-07 by Myzhar