stereolabs/zed-ros2-wrapper #63: Zed components: error: ‘set_on_parameters_set_callback’ was not declared in this scope

benbarron, 2021-11-12:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I'm getting the following error when building zed_components

### Steps to Reproduce

Steps from my docker file 

`ENV LOGNAME root
ENV DEBIAN_FRONTEND noninteractive

RUN apt-get update -y && apt-get install --no-install-recommends lsb-release wget less udev sudo apt-transport-https -y && \
    wget -q --no-check-certificate -O ZED_SDK_Linux_JP.run https://download.stereolabs.com/zedsdk/3.6/jp45/jetsons && \
    chmod +x ZED_SDK_Linux_JP.run ; ./ZED_SDK_Linux_JP.run silent runtime_only && \
    rm -rf /usr/local/zed/resources/* \
    rm -rf ZED_SDK_Linux_JP.run && \
    rm -rf /var/lib/apt/lists/*

RUN cd ${ROS_ROOT} && \
    mkdir ${ROS_ROOT}/src && \
    wget https://github.com/stereolabs/zed-ros2-wrapper/archive/refs/tags/foxy-v3.5.zip -q --no-check-certificate -O zed-ros2.zip && \
    unzip zed-ros2.zip && \
    mv zed-ros2-wrapper-foxy-v3.5 ${ROS_ROOT}/src/zed_ros2 

RUN cd ${ROS_ROOT} && \
    . ${ROS_ROOT}/install/setup.bash && \
    colcon build --merge-install --cmake-args=-DCMAKE_BUILD_TYPE=Release`

### Expected Result

Expected the package to build without error 

### Actual Result

got this error 
`Starting >>> zed_components
--- stderr: zed_components
/opt/ros/galactic/src/zed_ros2/zed_components/src/zed_camera/src/zed_camera_component.cpp: In member function ‘void stereolabs::ZedCamera::initParameters()’:
/opt/ros/galactic/src/zed_ros2/zed_components/src/zed_camera/src/zed_camera_component.cpp:295:3: error: ‘set_on_parameters_set_callback’ was not declared in this scope
   set_on_parameters_set_callback(std::bind(&ZedCamera::callback_paramChange, this, _1));
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/ros/galactic/src/zed_ros2/zed_components/src/zed_camera/src/zed_camera_component.cpp:295:3: note: suggested alternative: ‘add_on_set_parameters_callback’
   set_on_parameters_set_callback(std::bind(&ZedCamera::callback_paramChange, this, _1));
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   add_on_set_parameters_callback
make[2]: *** [CMakeFiles/zed_camera_component.dir/src/zed_camera/src/zed_camera_component.cpp.o] Error 1
make[1]: *** [CMakeFiles/zed_camera_component.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< zed_components [1min 8s, exited with code 2]`

### ZED Camera model

ZED2i

### Environment

```shell
OS: ubuntu 18.04 w/ jetpack 4.5.1
HW: nvidia jetson agx
```

Myzhar, 2021-11-12:
Hi @benbarron 
You are on a Jetson Xavier AGX running Ubuntu 18.04, unless you are not compiling ROS2 Foxy from sources  there is no support for Foxy on Ubuntu 18.04.
There is a working docker container for Jetson, you can retrieve it by reading this useful thread:
https://github.com/stereolabs/zed-ros2-wrapper/issues/26

benbarron, 2021-11-12:
@Myzhar is foxy the only ros2 distro that this library supports? I'm using galactic and yes I'm building everything from source in my base docker image. The distro mismatch could be the issue... do you know of any plans to support galactic in the near future.

Myzhar, 2021-11-12:
Galactic is not an LTS version and its lifecycle will finish next November 2022. We have not tested it yet with the ROS2 wrapper and I cannot say what are the compatibility problems.
Surely we will add support for the next LTS for Ubuntu 22.04 (Humble) that will be released on May:
https://docs.ros.org/en/foxy/Releases.html

closed 2021-12-07 by Myzhar