stereolabs/zed-ros2-wrapper #64: frame does not exist

an99990, 2021-11-15:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I didnt change anything from the repo since i started using it. I just discovered an issue recently.

### Steps to Reproduce

`ros2 launch zed_display_rviz2 display_zed2.launch.py`

### Expected Result

no errors

### Actual Result

[INFO] [launch]: All log files can be found below /home/tres/.ros/log/2021-11-15-12-19-05-535142-halodi-Precision-5550-36572
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [36575]
[INFO] [zed_wrapper-2]: process started with pid [36577]
[INFO] [rviz2-3]: process started with pid [36579]
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link zed2_camera_center had 4 children
[robot_state_publisher-1] Link zed2_baro_link had 0 children
[robot_state_publisher-1] Link zed2_left_camera_frame had 2 children
[robot_state_publisher-1] Link zed2_left_camera_optical_frame had 0 children
[robot_state_publisher-1] Link zed2_temp_left_link had 0 children
[robot_state_publisher-1] Link zed2_mag_link had 0 children
[robot_state_publisher-1] Link zed2_right_camera_frame had 2 children
[robot_state_publisher-1] Link zed2_right_camera_optical_frame had 0 children
[robot_state_publisher-1] Link zed2_temp_right_link had 0 children
[robot_state_publisher-1] 1636996745.695192530 [zed2.zed_state_publisher] [INFO] got segment base_link
[robot_state_publisher-1] 1636996745.695252072 [zed2.zed_state_publisher] [INFO] got segment zed2_baro_link
[robot_state_publisher-1] 1636996745.695260228 [zed2.zed_state_publisher] [INFO] got segment zed2_camera_center
[robot_state_publisher-1] 1636996745.695264848 [zed2.zed_state_publisher] [INFO] got segment zed2_left_camera_frame
[robot_state_publisher-1] 1636996745.695269142 [zed2.zed_state_publisher] [INFO] got segment zed2_left_camera_optical_frame
[robot_state_publisher-1] 1636996745.695273360 [zed2.zed_state_publisher] [INFO] got segment zed2_mag_link
[robot_state_publisher-1] 1636996745.695277372 [zed2.zed_state_publisher] [INFO] got segment zed2_right_camera_frame
[robot_state_publisher-1] 1636996745.695281292 [zed2.zed_state_publisher] [INFO] got segment zed2_right_camera_optical_frame
[robot_state_publisher-1] 1636996745.695285200 [zed2.zed_state_publisher] [INFO] got segment zed2_temp_left_link
[robot_state_publisher-1] 1636996745.695289091 [zed2.zed_state_publisher] [INFO] got segment zed2_temp_right_link
[zed_wrapper-2] 1636996745.701086770 [zed2.zed_node] [INFO] ********************************
[zed_wrapper-2] 1636996745.701151587 [zed2.zed_node] [INFO]       ZED Camera Component 
[zed_wrapper-2] 1636996745.701160037 [zed2.zed_node] [INFO] ********************************
[zed_wrapper-2] 1636996745.701165453 [zed2.zed_node] [INFO]  * namespace: /zed2
[zed_wrapper-2] 1636996745.701170642 [zed2.zed_node] [INFO]  * node name: zed_node
[zed_wrapper-2] 1636996745.701174834 [zed2.zed_node] [INFO] ********************************
[zed_wrapper-2] 1636996745.701207525 [zed2.zed_node] [INFO] *** DEBUG parameters ***
[zed_wrapper-2] 1636996745.701218807 [zed2.zed_node] [INFO] *** GENERAL parameters ***
[zed_wrapper-2] 1636996745.701242717 [zed2.zed_node] [INFO]  * Camera model: zed2 - ZED 2
[zed_wrapper-2] 1636996745.701259867 [zed2.zed_node] [INFO]  * SDK Verbose: 1
[zed_wrapper-2] 1636996745.701274045 [zed2.zed_node] [INFO]  * SVO: 
[zed_wrapper-2] 1636996745.701286253 [zed2.zed_node] [INFO]  * SVO Loop: TRUE
[zed_wrapper-2] 1636996745.701301484 [zed2.zed_node] [INFO]  * SVO Realtime: FALSE
[zed_wrapper-2] 1636996745.701315233 [zed2.zed_node] [INFO]  * Camera name: zed2
[zed_wrapper-2] 1636996745.701331333 [zed2.zed_node] [INFO]  * Camera ID: 0
[zed_wrapper-2] 1636996745.701344472 [zed2.zed_node] [INFO]  * Camera SN: 0
[zed_wrapper-2] 1636996745.701358189 [zed2.zed_node] [INFO]  * Camera timeout [sec]: 5
[zed_wrapper-2] 1636996745.701371338 [zed2.zed_node] [INFO]  * Camera reconnection temptatives: 5
[zed_wrapper-2] 1636996745.701386504 [zed2.zed_node] [INFO]  * Camera framerate: 30
[zed_wrapper-2] 1636996745.701400022 [zed2.zed_node] [INFO]  * GPU ID: -1
[zed_wrapper-2] 1636996745.701414293 [zed2.zed_node] [INFO]  * Camera resolution: 2 - HD720
[zed_wrapper-2] 1636996745.701426834 [zed2.zed_node] [INFO]  * Camera self calibration: TRUE
[zed_wrapper-2] 1636996745.701438419 [zed2.zed_node] [INFO]  * Camera flip: FALSE
[zed_wrapper-2] 1636996745.701453428 [zed2.zed_node] [INFO]  * [DYN] Publish framerate [Hz]: 15 
[zed_wrapper-2] 1636996745.701463702 [zed2.zed_node] [INFO] *** VIDEO parameters ***
[zed_wrapper-2] 1636996745.701476542 [zed2.zed_node] [INFO]  * Use old extrinsic parameters: 0
[... 132 lines cut ...]
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.11/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.11/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.11/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.11/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.11/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.11/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.11/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.11/src/buffer_core.cpp

### ZED Camera model

ZED2

### Environment

```shell
OS:Linux
```

Myzhar, 2021-11-15:
It's not a bug. It happens when RVIZ2 starts and the ZED node is not yet ready.
In this case Rviz issues a warning about missing frames, but it starts working correctly as soon as the ZED node starts broadcasting the TF frames

an99990, 2021-11-15:
but i dont see anything in rviz, i keep getting this warning indefinitely 
[rviz2-3] Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[rviz2-3] at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.11/src/buffer_core.cpp

Myzhar, 2021-11-16:
Sorry, I missed the following two important log lines:
```
[zed_wrapper-2] 1636996693.399577264 [zed2.zed_node] [WARN] Error opening camera: CAMERA NOT DETECTED
[zed_wrapper-2] 1636996693.399613872 [zed2.zed_node] [INFO] Please verify the USB3 connection
```
Can you correctly open the camera by using one of our tools? 
For example, try to launch `ZED_Explorer`

an99990, 2021-11-16:
The things is that i want out a svo file in common.yaml. so even if I'm just reading a video, i need to have the zed plugged in ?

Myzhar, 2021-11-16:
No, you don't. But you need to set up correctly the input:
`$ ros2 launch zed_display_rviz2 display_zed2.launch.py :=<SVO_filename>`

an99990, 2021-11-16:
i get this error when i try your command line
bash: syntax error near unexpected token `newline'
i type in
ros2 launch zed_display_rviz2 display_zed2.launch.py :=<svo/ccw2.svo>

Myzhar, 2021-11-16:
You must remove the brackets:
`ros2 launch zed_display_rviz2 display_zed2.launch.py svo_path:=svo/ccw2.svo`

an99990, 2021-11-16:
Thank you it worked :)

closed 2021-11-16 by an99990