stereolabs/zed-ros2-wrapper #65: Disable/ignore magnetometer

Sollimann, 2021-11-24:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

We need to be able to disable the magnetometer for the zed 2i-camera. It would be nice if there were some way to do that when you configure the camera setup.

### Use-Case

We're using the zed 2i camera in strong magnetic fields, it would be nice if we could turn of zed 2i's reliance to the magnetometer as it will most likely mess up the zed 2i estimates for positioning.

Myzhar, 2021-11-24:
@Sollimann you don't need to disable it. Magnetometer data are not fused in the ZED Pose estimation algorithm

Sollimann, 2021-11-24:
@Myzhar Great! Thank you!

closed 2021-12-07 by Myzhar