stereolabs/zed-ros2-wrapper #66: ros2-foxy build err

AndreV84, 2021-11-29:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

```
cat log.txt 
In file included from /home/user/your_ws/install/zed_components/include/zed_components/sl_tools.h:50,
                 from /home/user/your_ws/install/zed_components/include/zed_components/zed_camera_component.hpp:29,
                 from /home/user/your_ws/zed-ros2-wrapper/zed_wrapper/src/zed_wrapper.cpp:26:
/usr/local/zed/include/sl/Camera.hpp:784:13: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
             };
             ^
/usr/local/zed/include/sl/Camera.hpp:937:13: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
             };
             ^
/usr/local/zed/include/sl/Camera.hpp:1114:13: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
             };
             ^
/usr/local/zed/include/sl/Camera.hpp:1278:13: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
             };
             ^
/usr/local/zed/include/sl/Camera.hpp:1282:13: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
             };
             ^
/usr/local/zed/include/sl/Camera.hpp:1286:13: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
             };
             ^
/usr/local/zed/include/sl/Camera.hpp:1526:13: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
             };
             ^
/usr/local/zed/include/sl/Camera.hpp:1530:13: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
             };
             ^
/usr/local/zed/include/sl/Camera.hpp:1534:13: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
             };
             ^
/usr/local/zed/include/sl/Camera.hpp: In function ‘cudaError sl::__cudaSafeCall(cudaError, const char*, const char*, int)’:
/usr/local/zed/include/sl/Camera.hpp:2048:89: warning: unused parameter ‘file’ [-Wunused-parameter]
 daError __cudaSafeCall(cudaError err, const char *func, const char *file, const int line) {
                                                         ~~~~~~~~~~~~^~~~

/usr/local/zed/include/sl/Camera.hpp:2048:105: warning: unused parameter ‘line’ [-Wunused-parameter]
 daSafeCall(cudaError err, const char *func, const char *file, const int line) {
                                                               ~~~~~~~~~~^~~~

In file included from /home/user/your_ws/install/zed_components/include/zed_components/sl_tools.h:50,
                 from /home/user/your_ws/install/zed_components/include/zed_components/zed_camera_component.hpp:29,
                 from /home/user/your_ws/zed-ros2-wrapper/zed_wrapper/src/zed_wrapper.cpp:26:
/usr/local/zed/include/sl/Camera.hpp: At global scope:
/usr/local/zed/include/sl/Camera.hpp:5335:12: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
     inline const void /*@cond SHOWHIDDEN*/SL_SDK_EXPORT/*@endcond*/ getZEDSDKBuildVersion(int &major, int& minor, int& patch) {
            ^~~~~
In file included from /home/user/your_ws/install/zed_components/include/zed_components/sl_tools.h:50,
                 from /home/user/your_ws/install/zed_components/include/zed_components/zed_camera_component.hpp:29,
                 from /home/user/your_ws/zed-ros2-wrapper/zed_wrapper/src/zed_wrapper.cpp:26:
/usr/local/zed/include/sl/Camera.hpp:5724:2: warning: extra ‘;’ [-Wpedantic]
 };
  ^
```

### Steps to Reproduce

1.
2.
3.
...

### Expected Result

built node

### Actual Result

err

### ZED Camera model

ZED2

### Environment

```shell
foxy ros2
jetson aarch64
20.04 ubuntu
```

AndreV84, 2021-11-29:
tried with
https://stereolabs.sfo2.cdn.digitaloceanspaces.com/zedsdk/3.6/ZED_SDK_Tegra_JP46_v3.6.1.run

AndreV84, 2021-11-29:
the p;ose works

ros2 topic hz /zed2/zed_node/pose
average rate: 15.156

the rect grey seems empty

 ros2 topic hz /zed2/zed_node/left/image_rect_gray

AndreV84, 2021-11-29:
```
 colcon build --symlink-install
Starting >>> image_transport
Starting >>> rcpputils
Starting >>> zed_interfaces
Starting >>> isaac_ros_visual_odometry_interfaces
Finished <<< rcpputils [3.35s]                                                                                                
Starting >>> camera_calibration_parsers
Finished <<< image_transport [5.13s]                                                                    
Finished <<< isaac_ros_visual_odometry_interfaces [7.10s]                                                                     
Starting >>> isaac_ros_visual_odometry
Finished <<< camera_calibration_parsers [4.42s]                                                                         
Starting >>> camera_info_manager
Finished <<< zed_interfaces [8.70s]                                                                            
Starting >>> zed_components
--- stderr: camera_info_manager                                                                                        
/home/opal/your_ws/src/image_common/camera_info_manager/src/camera_info_manager.cpp:46:10: fatal error: rcpputils/env.hpp: No such file or directory
 #include "rcpputils/env.hpp"
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/camera_info_manager.dir/build.make:63: CMakeFiles/camera_info_manager.dir/src/camera_info_manager.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/camera_info_manager.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< camera_info_manager [3.83s, exited with code 2]
Aborted  <<< isaac_ros_visual_odometry [5.32s]                                                                       
Aborted  <<< zed_components [5.99s]                             

Summary: 5 packages finished [15.5s]
  1 package failed: camera_info_manager
  2 packages aborted: isaac_ros_visual_odometry zed_components
  1 package had stderr output: camera_info_manager
  3 packages not processed
```
package ros-foxy-rcpputils is installed;
trying without docker at all

Myzhar, 2021-11-30:
Hi @AndreV84 
there are no errors in the first build log that you pasted, but only warnings that do not stop the process.
The log above instead it's a failure in compiling `camera_info_manager.cpp` that is a ROS2 file, not a Stereolabs file.
Please check that you are using the correct branch of the `image_common' package for ROS2 Foxy.

Please re-open the issue if you are facing problems with building the ZED ROS2 Wrapper.

closed 2021-11-30 by Myzhar

AndreV84, 2021-11-30:
@Myzhar 
Thank you for your reply.
However, the only reason I waas trying to build the camera_info manager
is that ros2 foxy zed2 wrapper won't show any image output on camera toppics, but only pose.
So I followed the proposd image transport solution but it did not work

AndreV84, 2021-11-30:
how do I get the zed2 ros2 foxy zed camera left/ right topics? Is there a working solution?
this topic is empty
```
ros2 topic hz /zed2/zed_node/left/image_rect_gray

AndreV84, 2021-11-30:
@Myzhar 
so I tried these proposed steps which did not resolve the empty topic issue
```
$ cd <colcon_workspace>/src # Access the source folder of your colcon workspace
$ git clone https://github.com/ros-perception/image_common.git -b ros2 # clone the "ros" branch of the "image_common" repository
$ cd <colcon_workspace> # Go back to the root of your colcon workspace
$ colcon build --symlink-install # Compile everything and install

Myzhar, 2021-11-30:
OK, I understand the problem... that's a bug of the function `CameraPublisher::getNumSubscribers` that always returns zero.
You can try by using this version of `image_common' that was surely working:
https://github.com/ros-perception/image_common/releases/tag/3.0.0

AndreV84, 2021-11-30:
@Myzhar 
Thanks
It worked