stereolabs/zed-ros2-wrapper #68: Unable to Clone Repository in Dockerfile

kevinanschau, 2021-12-08:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Since the introduction of the `zed-ros2-interfaces` submodule, I cannot `git clone` the repository in a Dockerfile anymore. I suppose the issue has something to do with the SSH URL of the submodule. I also tried the `zed-ros-wrapper`, which also has a submodule but with an HTTPS URL, and that works.

My `Dockerfile`

```dockerfile
ARG ZED_SDK_VERSION
ARG L4T_VERSION
ARG ROS_DISTRO

FROM stereolabs/zed:${ZED_SDK_VERSION}-runtime-l4t-${L4T_VERSION} AS zed

FROM dustynv/ros:${ROS_DISTRO}-ros-base-l4t-${L4T_VERSION} as ros2

ENV ROS_WS="/workspaces/${ROS_DISTRO}"

FROM ros2 AS build

RUN apt-get update && \
    apt-get install -y --no-install-recommends \
        build-essential \
        libturbojpeg-dev \
        libusb-1.0-0-dev \
        libv4l-dev && \
    rm -rf /var/lib/apt/lists/*

# Get ZED ROS 2 wrapper source and dependencies that rosdep cannot find.
ARG ZED_SDK_VERSION
WORKDIR "${ROS_WS}/src"
RUN apt-get update && \
    apt-get install -y --no-install-recommends git && \
    git clone -b "${ROS_DISTRO}-v${ZED_SDK_VERSION}" --recurse https://github.com/stereolabs/zed-ros2-wrapper.git && \
    git clone -b "${ROS_DISTRO}" https://github.com/ros/diagnostics.git && \
    git clone -b ros2 https://github.com/ros-perception/image_common.git && \
    git clone -b ros2 https://github.com/ros/xacro.git

COPY --from=zed /usr/local/zed /usr/local/zed/
ENV LD_LIBRARY_PATH="${LD_LIBRARY_PATH}:/usr/local/zed/lib"

WORKDIR ${ROS_WS}
RUN apt-get update && \
    . "${ROS_ROOT/foxy/foxy\/install}/setup.sh" && \
    rosdep update --rosdistro "${ROS_DISTRO}" && \
    rosdep install --default-yes --ignore-src --from-paths . --rosdistro "${ROS_DISTRO}" && \
    colcon build --event-handlers console_direct+ --executor sequential --packages-up-to zed_ros2 --cmake-args \
        " -DCMAKE_BUILD_TYPE=Release" \
        " -DCMAKE_LIBRARY_PATH=/usr/local/cuda/lib64/stubs" \
        " -DCUDA_CUDART_LIBRARY=/usr/local/cuda/lib64/stubs" \
        " -DCMAKE_CXX_FLAGS='-Wl,--allow-shlib-undefined'" && \
    rm -rf /var/lib/apt/lists/* "${ROS_WS}/build"

# This symbolic link is needed to use the streaming features on Jetson inside a container
RUN ln -sf /usr/lib/aarch64-linux-gnu/tegra/libv4l2.so.0 /usr/lib/aarch64-linux-gnu/libv4l2.so

FROM build AS runtime

WORKDIR /workspaces
COPY tools/docker/entrypoint.sh .
ENTRYPOINT [ "./entrypoint.sh" ]
```

and the `entrypoint.sh` file

```bash
#!/bin/bash
# shellcheck disable=SC1090

source "${ROS_ROOT/foxy/foxy\/install}/setup.bash"
source "${ROS_WS}/install/setup.bash"

exec "$@"

```

### Steps to Reproduce

Build with
```bash
docker build --build-arg ZED_SDK_VERSION=3.6 --build-arg L4T_VERSION=r32.6.1 --build-arg ROS_DISTRO=foxy --platform linux/arm64 --tags test .
```

### Expected Result

Docker builds the image without failures.

### Actual Result

I get the following error

```bash
[...]
Cloning into 'zed-ros2-wrapper'...
#10 132.3 Note: checking out '7cbd090186a838a93b0077fc861b887ba55aefc3'.
#10 132.3 
#10 132.3 You are in 'detached HEAD' state. You can look around, make experimental
#10 132.3 changes and commit them, and you can discard any commits you make in this
#10 132.3 state without impacting any branches by performing another checkout.
#10 132.3 
#10 132.3 If you want to create a new branch to retain commits you create, you may
#10 132.3 do so (now or later) by using -b with the checkout command again. Example:
#10 132.3 
#10 132.3   git checkout -b <new-branch-name>
#10 132.3 
#10 134.4 Submodule 'zed-ros2-interfaces' (git@github.com:stereolabs/zed-ros2-interfaces.git) registered for path 'zed-ros2-interfaces'
#10 134.8 Cloning into '/workspaces/foxy/src/zed-ros2-wrapper/zed-ros2-interfaces'...
#10 134.8 fatal: unable to fork
#10 134.9 fatal: clone of 'git@github.com:stereolabs/zed-ros2-interfaces.git' into submodule path '/workspaces/foxy/src/zed-ros2-wrapper/zed-ros2-interfaces' failed
#10 134.9 Failed to clone 'zed-ros2-interfaces'. Retry scheduled
#10 135.1 Cloning into '/workspaces/foxy/src/zed-ros2-wrapper/zed-ros2-interfaces'...
#10 135.1 fatal: unable to fork
#10 135.1 fatal: clone of 'git@github.com:stereolabs/zed-ros2-interfaces.git' into submodule path '/workspaces/foxy/src/zed-ros2-wrapper/zed-ros2-interfaces' failed
#10 135.1 Failed to clone 'zed-ros2-interfaces' a second time, aborting
------
executor failed running [/bin/bash -c apt-get update &&     apt-get install -y --no-install-recommends git &&     git clone -b "${ROS_DISTRO}-v${ZED_SDK_VERSION}" --recurse https://github.com/stereolabs/zed-ros2-wrapper.git &&     git clone -b "${ROS_DISTRO}" https://github.com/ros/diagnostics.git &&     git clone -b ros2 https://github.com/ros-perception/image_common.git &&     git clone -b ros2 https://github.com/ros/xacro.git]: exit code: 1
```

### ZED Camera model

ZED2i

### Environment

```shell
OS: Ubuntu 20.04
```

### Anything else?

The git clone command works for [`zed-ros-wrapper`](https://github.com/stereolabs/zed-ros-wrapper), though. This is the reason why I suspect the submodule's URL.

Myzhar, 2021-12-09:
Hi @kevinanschau 
I tested the command `$ git clone --recurse-submodules https://github.com/stereolabs/zed-ros2-wrapper.git` and everything seems to work correctly:

```
$ git clone --recurse-submodules https://github.com/stereolabs/zed-ros2-wrapper.git
Cloning into 'zed-ros2-wrapper'...
remote: Enumerating objects: 1444, done.
remote: Counting objects: 100% (657/657), done.
remote: Compressing objects: 100% (436/436), done.
remote: Total 1444 (delta 371), reused 436 (delta 202), pack-reused 787
Receiving objects: 100% (1444/1444), 1.86 MiB | 1.65 MiB/s, done.
Resolving deltas: 100% (777/777), done.
Submodule 'zed-ros2-interfaces' (git@github.com:stereolabs/zed-ros2-interfaces.git) registered for path 'zed-ros2-interfaces'
Cloning into '/home/walter/temp/zed-ros2-wrapper/zed-ros2-interfaces'...
remote: Enumerating objects: 48, done.        
remote: Counting objects: 100% (48/48), done.        
remote: Compressing objects: 100% (34/34), done.        
remote: Total 48 (delta 13), reused 38 (delta 10), pack-reused 0        
Receiving objects: 100% (48/48), 713.85 KiB | 1.39 MiB/s, done.
Resolving deltas: 100% (13/13), done.
Submodule path 'zed-ros2-interfaces': checked out '5fab23c62ae6199021c35e6802d2c87ac87276a9'
```
you are using this command that is slightly different:
`git clone -b "${ROS_DISTRO}-v${ZED_SDK_VERSION}" --recurse https://github.com/stereolabs/zed-ros2-wrapper.git`

What are the values of the `ROS_DISTRO` and `ZED_SDK_VERSION` variables?
Why do you use `--recurse` and not `--recurse-submodules`? I looked in the documentation and `--recurse` seems to not exist.

kevinanschau, 2021-12-10:
Thank you for your reply, @Myzhar.

I also tested your command locally and it works. But when I use the same command in the Dockerfile, it doesn't, and it outputs the same error I got above.

I run this command

```bash
docker build --build-arg ZED_SDK_VERSION=3.6 --build-arg L4T_VERSION=r32.6.1 --build-arg ROS_DISTRO=foxy --platform linux/arm64 --tags test .
```
and so the command becomes

```bash
git clone -b foxy-v3.6 --recurse https://github.com/stereolabs/zed-ros2-wrapper.git
```

`--recurse` worked for me outside the Dockerfile. Don't know why, but I see that it's not in the documentation. Regardless, changing it to `--recurse-submodules` or `--recursive` didn't solve the issue.

Oddly enough, using the same command inside the Dockerfile to clone the `zed-ros-wrapper` works.

kevinanschau, 2021-12-16:
Hi @Myzhar,

commit 6ce5488 fixed this issue. Thank you very much.

closed 2021-12-16 by kevinanschau