stereolabs/zed-ros2-wrapper #70: stderr output: zed_wrapper

electop, 2021-12-28:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

1 package had stderr output: zed_wrapper when building the package of "zed_wrapper"

### Steps to Reproduce

## Procedure
```
$cd ~/ros2_ws/src/ #use your current ros2 workspace folder
$ git clone https://github.com/stereolabs/zed-ros2-wrapper.git
$ cd ..
$ rosdep install --from-paths src --ignore-src -r -y
$ colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release
```

### Expected Result

No building issues

### Actual Result

## Build Log
```
colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release
Starting >>> zed_interfaces
Starting >>> zed_tutorial_depth
Starting >>> zed_tutorial_pos_tracking
Starting >>> zed_tutorial_video
Finished <<< zed_tutorial_depth [8.00s]                                 
Finished <<< zed_tutorial_video [8.48s]                                 
Finished <<< zed_tutorial_pos_tracking [9.85s]                          
Finished <<< zed_interfaces [13.0s]                        
Starting >>> zed_components
Starting >>> rviz_plugin_zed_od
Finished <<< rviz_plugin_zed_od [10.7s]                                                            
Finished <<< zed_components [23.3s]                        
Starting >>> zed_wrapper
Starting >>> zed_rgb_convert
--- stderr: zed_wrapper                                                                            
In file included from /root/colcon_ws/install/zed_components/include/zed_components/sl_tools.h:50,
                 from /root/colcon_ws/install/zed_components/include/zed_components/zed_camera_component.hpp:29,
                 from /root/colcon_ws/src/zed-ros2-wrapper/zed_wrapper/src/zed_wrapper.cpp:26:
/usr/local/zed/include/sl/Camera.hpp:784:13: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
  784 |             };
      |             ^
/usr/local/zed/include/sl/Camera.hpp:937:13: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
  937 |             };
      |             ^
/usr/local/zed/include/sl/Camera.hpp:1114:13: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
 1114 |             };
      |             ^
/usr/local/zed/include/sl/Camera.hpp:1278:13: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
 1278 |             };
      |             ^
/usr/local/zed/include/sl/Camera.hpp:1282:13: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
 1282 |             };
      |             ^
/usr/local/zed/include/sl/Camera.hpp:1286:13: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
 1286 |             };
      |             ^
/usr/local/zed/include/sl/Camera.hpp:1526:13: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
 1526 |             };
      |             ^
/usr/local/zed/include/sl/Camera.hpp:1530:13: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
 1530 |             };
      |             ^
/usr/local/zed/include/sl/Camera.hpp:1534:13: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
 1534 |             };
      |             ^
/usr/local/zed/include/sl/Camera.hpp: In function ‘sl::Timestamp sl::operator+(sl::Timestamp, const sl::Timestamp&)’:
/usr/local/zed/include/sl/Camera.hpp:1992:16: warning: implicitly-declared ‘constexpr sl::Timestamp::Timestamp(const sl::Timestamp&)’ is deprecated [-Wdeprecated-copy]
 1992 |         return lhs;
      |                ^~~
/usr/local/zed/include/sl/Camera.hpp:1931:20: note: because ‘sl::Timestamp’ has user-provided ‘sl::Timestamp& sl::Timestamp::operator=(sl::Timestamp)’
 1931 |         Timestamp& operator=(Timestamp other) {
      |                    ^~~~~~~~
/usr/local/zed/include/sl/Camera.hpp: In function ‘sl::Timestamp sl::operator-(sl::Timestamp, const sl::Timestamp&)’:
/usr/local/zed/include/sl/Camera.hpp:1997:16: warning: implicitly-declared ‘constexpr sl::Timestamp::Timestamp(const sl::Timestamp&)’ is deprecated [-Wdeprecated-copy]
 1997 |         return lhs;
      |                ^~~
/usr/local/zed/include/sl/Camera.hpp:1931:20: note: because ‘sl::Timestamp’ has user-provided ‘sl::Timestamp& sl::Timestamp::operator=(sl::Timestamp)’
 1931 |         Timestamp& operator=(Timestamp other) {
      |                    ^~~~~~~~
/usr/local/zed/include/sl/Camera.hpp: In function ‘sl::Timestamp sl::operator/(sl::Timestamp, const sl::Timestamp&)’:
/usr/local/zed/include/sl/Camera.hpp:2002:16: warning: implicitly-declared ‘constexpr sl::Timestamp::Timestamp(const sl::Timestamp&)’ is deprecated [-Wdeprecated-copy]
 2002 |         return lhs;
      |                ^~~
/usr/local/zed/include/sl/Camera.hpp:1931:20: note: because ‘sl::Timestamp’ has user-provided ‘sl::Timestamp& sl::Timestamp::operator=(sl::Timestamp)’
 1931 |         Timestamp& operator=(Timestamp other) {
      |                    ^~~~~~~~
/usr/local/zed/include/sl/Camera.hpp: In function ‘sl::Timestamp sl::operator*(sl::Timestamp, const sl::Timestamp&)’:
/usr/local/zed/include/sl/Camera.hpp:2007:16: warning: implicitly-declared ‘constexpr sl::Timestamp::Timestamp(const sl::Timestamp&)’ is deprecated [-Wdeprecated-copy]
 2007 |         return lhs;
      |                ^~~
/usr/local/zed/include/sl/Camera.hpp:1931:20: note: because ‘sl::Timestamp’ has user-provided ‘sl::Timestamp& sl::Timestamp::operator=(sl::Timestamp)’
 1931 |         Timestamp& operator=(Timestamp other) {
      |                    ^~~~~~~~
/usr/local/zed/include/sl/Camera.hpp: In function ‘sl::Timestamp sl::getCurrentTimeStamp()’:
/usr/local/zed/include/sl/Camera.hpp:2041:16: warning: implicitly-declared ‘constexpr sl::Timestamp::Timestamp(const sl::Timestamp&)’ is deprecated [-Wdeprecated-copy]
 2041 |         return current_ts;
      |                ^~~~~~~~~~
/usr/local/zed/include/sl/Camera.hpp:1931:20: note: because ‘sl::Timestamp’ has user-provided ‘sl::Timestamp& sl::Timestamp::operator=(sl::Timestamp)’
 1931 |         Timestamp& operator=(Timestamp other) {
      |                    ^~~~~~~~
/usr/local/zed/include/sl/Camera.hpp: In function ‘cudaError sl::__cudaSafeCall(cudaError, const char*, const char*, int)’:
/usr/local/zed/include/sl/Camera.hpp:2048:89: warning: unused parameter ‘file’ [-Wunused-parameter]
 2048 | atic inline cudaError __cudaSafeCall(cudaError err, const char *func, const char *file, const int line) {
      |                                                                       ~~~~~~~~~~~~^~~~
/usr/local/zed/include/sl/Camera.hpp:2048:105: warning: unused parameter ‘line’ [-Wunused-parameter]
 2048 | Error __cudaSafeCall(cudaError err, const char *func, const char *file, const int line) {
      |                                                                         ~~~~~~~~~~^~~~
In file included from /root/colcon_ws/install/zed_components/include/zed_components/sl_tools.h:50,
                 from /root/colcon_ws/install/zed_components/include/zed_components/zed_camera_component.hpp:29,
                 from /root/colcon_ws/src/zed-ros2-wrapper/zed_wrapper/src/zed_wrapper.cpp:26:
/usr/local/zed/include/sl/Camera.hpp: At global scope:
/usr/local/zed/include/sl/Camera.hpp:5343:12: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
 5343 |     inline const void /*@cond SHOWHIDDEN*/SL_SDK_EXPORT/*@endcond*/ getZEDSDKBuildVersion(int &major, int& minor, int& patch) {
      |            ^~~~~
/usr/local/zed/include/sl/Camera.hpp:5732:2: warning: extra ‘;’ [-Wpedantic]
 5732 | };
      |  ^
---
Finished <<< zed_wrapper [4.80s]
Starting >>> zed_display_rviz2
Starting >>> zed_ros2
Finished <<< zed_ros2 [0.38s]                                             
Finished <<< zed_display_rviz2 [0.79s]                                    
Finished <<< zed_rgb_convert [7.78s]                       
Summary: 10 packages finished [44.1s]
  1 package had stderr output: zed_wrapper
```

### ZED Camera model

ZED2i

### Environment

```shell
OS : Ubuntu 20.04
ROS2 : Foxy
Docker image : stereolabs/zed:3.6-gl-devel-cuda11.4-ubuntu20.04
```

Myzhar, 2021-12-28:
@electop there are only warnings, not errors.
You can ignore them. If you start a new compile process you can see that they are not appearing anymore.

electop, 2021-12-29:
@Myzhar Thanks for your reply.
Happy new year!! : )

electop, 2021-12-29:
This issue is just warnings.
So I should close this issue.

closed 2021-12-29 by electop