stereolabs/zed-ros2-wrapper #75: RuntimeError during launch of the app

KorovkoAlexander, 2022-02-13:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hi!

I am experiencing a strange RuntimeError during the launch of the zedm.launch.py app. Could you please help me to fix it?

### Steps to Reproduce

1. download [image_common_v3.0.0](https://github.com/ros-perception/image_common/releases/tag/3.0.0) package
2. extract it to the <workspace>/src folder
3. cd <workspace>/src
4. git clone  --recursive https://github.com/stereolabs/zed-ros2-wrapper.git
5. cd ..
6. colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release
7. source install/local_setup.bash
8. ros2 launch zed_wrapper zedm.launch.py

### Expected Result

The app works without errors.

### Actual Result

akorovko@akorovkoLT:~/ros_ws$ ros2 launch zed_wrapper zedm.launch.py
[INFO] [launch]: All log files can be found below /home/akorovko/.ros/log/2022-02-13-18-44-41-206531-akorovkoLT-37118
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=TypeError("Allowed value types are bytes, bool, int, float, str, Sequence[bool], Sequence[int], Sequence[float], Sequence[str]. Got <class 'NoneType'>.If the parameter is meant to be a string, try wrapping it in launch_ros.parameter_descriptions.ParameterValue(value, value_type=str)")>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 3 more times]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 444, in execute
    self._perform_substitutions(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 399, in _perform_substitutions
    evaluated_parameters = evaluate_parameters(context, self.__parameters)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 160, in evaluate_parameters
    output_params.append(evaluate_parameter_dict(context, param))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 95, in evaluate_parameter_dict
    raise TypeError(
TypeError: Allowed value types are bytes, bool, int, float, str, Sequence[bool], Sequence[int], Sequence[float], Sequence[str]. Got <class 'NoneType'>.If the parameter is meant to be a string, try wrapping it in launch_ros.parameter_descriptions.ParameterValue(value, value_type=str)
Task exception was never retrieved
future: <Task finished name='Task-8' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=RuntimeError('Signal event received before subprocess transport available.')>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/opaque_function.py", line 75, in execute
    return self.__function(context, *self.__args, **self.__kwargs)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/execute_process.py", line 443, in __on_signal_process_event
    raise RuntimeError('Signal event received before subprocess transport available.')
RuntimeError: Signal event received before subprocess transport available.
[INFO] [robot_state_publisher-1]: process started with pid [37121]
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link zedm_base_link had 1 children
[robot_state_publisher-1] Link zedm_camera_center had 2 children
[robot_state_publisher-1] Link zedm_left_camera_frame had 1 children
[robot_state_publisher-1] Link zedm_left_camera_optical_frame had 0 children
[robot_state_publisher-1] Link zedm_right_camera_frame had 1 children
[robot_state_publisher-1] Link zedm_right_camera_optical_frame had 0 children
[robot_state_publisher-1] 1644767081.347807068 [zedm.zed_state_publisher] [INFO] got segment base_link
[robot_state_publisher-1] 1644767081.347915423 [zedm.zed_state_publisher] [INFO] got segment zedm_base_link
[robot_state_publisher-1] 1644767081.347923660 [zedm.zed_state_publisher] [INFO] got segment zedm_camera_center
[robot_state_publisher-1] 1644767081.347927872 [zedm.zed_state_publisher] [INFO] got segment zedm_left_camera_frame
[robot_state_publisher-1] 1644767081.347931911 [zedm.zed_state_publisher] [INFO] got segment zedm_left_camera_optical_frame
[robot_state_publisher-1] 1644767081.347935817 [zedm.zed_state_publisher] [INFO] got segment zedm_right_camera_frame
[robot_state_publisher-1] 1644767081.347957905 [zedm.zed_state_publisher] [INFO] got segment zedm_right_camera_optical_frame
[ERROR] [robot_state_publisher-1]: process[robot_state_publisher-1] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [robot_state_publisher-1]: sending signal 'SIGTERM' to process[robot_state_publisher-1]
[ERROR] [robot_state_publisher-1]: process has died [pid 37121, exit code -15, cmd '/opt/ros/foxy/lib/robot_state_publisher/robot_state_publisher --ros-args -r __node:=zed_state_publisher -r __ns:=/zedm --params-file /tmp/launch_params_d7ptzizc'].

### ZED Camera model

ZED Mini

### Environment

```shell
OS: Ubuntu 20.40
GPU: NVIDIA GeForce P4
CUDA version: 11.3
Zed SDK version: 3.6.5
image_common package v3.0.0
zed-ros2-wrapper master (commit 4d6fb835f7f00f1572b916602890fb01006c6694)
```

Myzhar, 2022-02-14:
It seems that you have a problem with one parameter:
`future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=TypeError("Allowed value types are bytes, bool, int, float, str, Sequence[bool], Sequence[int], Sequence[float], Sequence[str]. Got <class 'NoneType'>.If the parameter is meant to be a string, try wrapping it in launch_ros.parameter_descriptions.ParameterValue(value, value_type=str)")>`

Have you modified the `yaml` files in some way?

KorovkoAlexander, 2022-02-14:
Hi, @Myzhar !

No, I haven`t modified the yaml.
But it seems that I have a part of the solution.
Apparently, some ros2 packages were updated, and this is the source of the problem. I have tried to run the app with the ros environment, that my colleague gave me (just a zip archive with /opt/ros/foxy folder) and everything worked pretty well.
Comparing two ros environments showed that a set of ros packages have been updated.

I can help you with the versions of python packages, which worked successfully with the zed wrapper.
[python_packages.txt](https://github.com/stereolabs/zed-ros2-wrapper/files/8061607/python_packages.txt)
Perhaps, I can help you somehow providing the list of versions of deb packages?

Myzhar, 2022-02-15:
I forced the upgrade on my machine and I noticed that there are many ROS2 packages in the list.
I see if I can replicate and fix the problem

Myzhar, 2022-02-15:
After the upgrade I'm having the same problem problem, I'm searching for the cause and provide a fix.

Myzhar, 2022-02-15:
The problem is in the launch files, starting the node with `ros2 run` does not show the error.

Myzhar, 2022-02-15:
Finally I caught the problem. It seems that something changed in the handling of the empty strings because the launch file does not allow empty string for the `svo_path` parameter:
```    
    declare_svo_path_cmd = DeclareLaunchArgument(
        'svo_path',
        default_value='',
        description='Path to an input SVO file. Note: overrides the parameter `general.svo_file` in `common.yaml`.')
```

Myzhar, 2022-02-15:
Fixed with a patch in 6cd5affeaef51eab8f7beb708c73ff95e2adf6f3

Myzhar, 2022-02-15:
Hi @Void12987 thank you for the feedback

closed 2022-04-26 by github-actions[bot]