stereolabs/zed-ros2-wrapper #76: Robot State Publisher Error

THLL245, 2022-02-13:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Worked fine before hand, but after update & upgrade started to get a robot state publisher error when only launching the zed wrapper. Cant figure out how its related.

> tyler@tyler:~/ros_ws$ ros2 launch zed_wrapper zed2i.launch.py
[INFO] [launch]: All log files can be found below /home/tyler/.ros/log/2022-02-12-15-19-39-065977-tyler-28816
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name=‘Task-2’ coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=TypeError(“Allowed value types are bytes, bool, int, float, str, Sequence[bool], Sequence[int], Sequence[float], Sequence[str]. Got <class ‘NoneType’>.If the parameter is meant to be a string, try wrapping it in launch_ros.parameter_descriptions.ParameterValue(value, value_type=str)”)>
Traceback (most recent call last):
File “/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py”, line 228, in _process_one_event
await self.__process_event(next_event)
File “/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py”, line 248, in __process_event
visit_all_entities_and_collect_futures(entity, self.__context))
File “/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py”, line 45, in visit_all_entities_and_collect_futures
futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
File “/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py”, line 45, in visit_all_entities_and_collect_futures
futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
File “/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py”, line 45, in visit_all_entities_and_collect_futures
futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
[Previous line repeated 3 more times]
File “/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py”, line 38, in visit_all_entities_and_collect_futures
sub_entities = entity.visit(context)
File “/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py”, line 108, in visit
return self.execute(context)
File “/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py”, line 444, in execute
self._perform_substitutions(context)
File “/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py”, line 399, in _perform_substitutions
evaluated_parameters = evaluate_parameters(context, self.__parameters)
File “/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py”, line 160, in evaluate_parameters
output_params.append(evaluate_parameter_dict(context, param))
File “/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py”, line 95, in evaluate_parameter_dict
raise TypeError(
TypeError: Allowed value types are bytes, bool, int, float, str, Sequence[bool], Sequence[int], Sequence[float], Sequence[str]. Got <class ‘NoneType’>.If the parameter is meant to be a string, try wrapping it in launch_ros.parameter_descriptions.ParameterValue(value, value_type=str)
Task exception was never retrieved
future: <Task finished name=‘Task-8’ coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=RuntimeError(‘Signal event received before subprocess transport available.’)>
Traceback (most recent call last):
File “/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py”, line 228, in _process_one_event
await self.__process_event(next_event)
File “/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py”, line 248, in __process_event
visit_all_entities_and_collect_futures(entity, self.__context))
File “/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py”, line 38, in visit_all_entities_and_collect_futures
sub_entities = entity.visit(context)
File “/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py”, line 108, in visit
return self.execute(context)
File “/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/opaque_function.py”, line 75, in execute
return self.__function(context, *self.__args, **self.__kwargs)
File “/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/execute_process.py”, line 443, in __on_signal_process_event
raise RuntimeError(‘Signal event received before subprocess transport available.’)
RuntimeError: Signal event received before subprocess transport available.
[INFO] [robot_state_publisher-1]: process started with pid [28819]
[robot_state_publisher-1] Parsing robot urdf xml string.
[robot_state_publisher-1] Link zed2i_base_link had 1 children
[robot_state_publisher-1] Link zed2i_camera_center had 2 children
[robot_state_publisher-1] Link zed2i_left_camera_frame had 1 children
[robot_state_publisher-1] Link zed2i_left_camera_optical_frame had 0 children
[robot_state_publisher-1] Link zed2i_right_camera_frame had 1 children
[robot_state_publisher-1] Link zed2i_right_camera_optical_frame had 0 children
[robot_state_publisher-1] 1644707979.529803626 [zed2i.zed_state_publisher] [INFO] got segment base_link
[robot_state_publisher-1] 1644707979.529967897 [zed2i.zed_state_publisher] [INFO] got segment zed2i_base_link
[robot_state_publisher-1] 1644707979.529997436 [zed2i.zed_state_publisher] [INFO] got segment zed2i_camera_center
[robot_state_publisher-1] 1644707979.530038623 [zed2i.zed_state_publisher] [INFO] got segment zed2i_left_camera_frame
[robot_state_publisher-1] 1644707979.530064930 [zed2i.zed_state_publisher] [INFO] got segment zed2i_left_camera_optical_frame
[robot_state_publisher-1] 1644707979.530105125 [zed2i.zed_state_publisher] [INFO] got segment zed2i_right_camera_frame
[robot_state_publisher-1] 1644707979.530147145 [zed2i.zed_state_publisher] [INFO] got segment zed2i_right_camera_optical_frame
[... 45 lines cut ...]
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 3 more times]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 444, in execute
    self._perform_substitutions(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 399, in _perform_substitutions
    evaluated_parameters = evaluate_parameters(context, self.__parameters)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 160, in evaluate_parameters
    output_params.append(evaluate_parameter_dict(context, param))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 95, in evaluate_parameter_dict
    raise TypeError(
TypeError: Allowed value types are bytes, bool, int, float, str, Sequence[bool], Sequence[int], Sequence[float], Sequence[str]. Got <class 'NoneType'>.If the parameter is meant to be a string, try wrapping it in launch_ros.parameter_descriptions.ParameterValue(value, value_type=str)

Myzhar, 2022-02-13:
I just noticed that you have Ubuntu 20.04 on Xavier.
Are you using Docker or did you upgrade Ubuntu 18.04 to 20.04?
The Nvidia Jetson boards do not support Ubuntu 20.04 yet, you should not upgrade the OS via apt

THLL245, 2022-02-13:
I upgraded to 20.04 via apt but had no issues with it. Everything I need to run on the Jetson works well. Zed wrapper ran great for weeks now. Only change was a upgrade yesterday now getting this error. I'll see if I can revert back and try the wrapper again

Myzhar, 2022-02-15:
Fixed with a patch in https://github.com/stereolabs/zed-ros2-wrapper/commit/6cd5affeaef51eab8f7beb708c73ff95e2adf6f3
See also #75 

@THLL245  let me know if this solved the problem for you then close the issue if ok.

THLL245, 2022-02-20:
It works again thank you!

closed 2022-02-20 by THLL245