stereolabs/zed-ros2-wrapper #79: InvalidParameterValueException when launching in ROS2 Galactic

weitung, 2022-02-25:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I'm working with Zed2 cameras in ROS galactic environment. Zed-ros2-wrapper is built successfully from source, but when I ran `ros2 launch zed_wrapper zed2.launch.py`, it couldn't start the camera and raised the InvalidParameterValueException error: 
```[zed_wrapper-2] terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterValueException'```
```[zed_wrapper-2]   what():  parameter 'qos_overrides./tf.publisher.durability' could not be set: qos_overrides./tf.publisher.durability is not a dynamic parameter``` 

### Steps to Reproduce

1. Build zed-ros2-wrapper from source in ROS galactic ws
2. run `ros2 launch zed_wrapper zed2.launch.py`

### Expected Result

Zed2 camera successfully launches

### Actual Result

```ros2 launch zed_wrapper zed2.launch.py
[INFO] [launch]: All log files can be found below /root/.ros/log/2022-02-25-09-23-36-806348-nexi-desktop-142
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [146]
[INFO] [zed_wrapper-2]: process started with pid [147]
[robot_state_publisher-1] 1645781017.319432 [0] robot_stat: using network interface eth0 (udp/192.168.22.157) selected arbitrarily from: eth0, br-436a6caa4e28, docker0
[robot_state_publisher-1] Link zed2_base_link had 1 children
[robot_state_publisher-1] Link zed2_camera_center had 4 children
[robot_state_publisher-1] Link zed2_baro_link had 0 children
[robot_state_publisher-1] Link zed2_left_camera_frame had 2 children
[robot_state_publisher-1] Link zed2_left_camera_optical_frame had 0 children
[robot_state_publisher-1] Link zed2_temp_left_link had 0 children
[robot_state_publisher-1] Link zed2_mag_link had 0 children
[robot_state_publisher-1] Link zed2_right_camera_frame had 2 children
[robot_state_publisher-1] Link zed2_right_camera_optical_frame had 0 children
[robot_state_publisher-1] Link zed2_temp_right_link had 0 children
[robot_state_publisher-1] 1645781017.370210999 [zed2.zed_state_publisher] [INFO] got segment base_link
[robot_state_publisher-1] 1645781017.370530565 [zed2.zed_state_publisher] [INFO] got segment zed2_baro_link
[robot_state_publisher-1] 1645781017.370618537 [zed2.zed_state_publisher] [INFO] got segment zed2_base_link
[robot_state_publisher-1] 1645781017.370677195 [zed2.zed_state_publisher] [INFO] got segment zed2_camera_center
[robot_state_publisher-1] 1645781017.370731822 [zed2.zed_state_publisher] [INFO] got segment zed2_left_camera_frame
[robot_state_publisher-1] 1645781017.370810449 [zed2.zed_state_publisher] [INFO] got segment zed2_left_camera_optical_frame
[robot_state_publisher-1] 1645781017.370882612 [zed2.zed_state_publisher] [INFO] got segment zed2_mag_link
[robot_state_publisher-1] 1645781017.370939255 [zed2.zed_state_publisher] [INFO] got segment zed2_right_camera_frame
[robot_state_publisher-1] 1645781017.370997593 [zed2.zed_state_publisher] [INFO] got segment zed2_right_camera_optical_frame
[robot_state_publisher-1] 1645781017.371076221 [zed2.zed_state_publisher] [INFO] got segment zed2_temp_left_link
[robot_state_publisher-1] 1645781017.371125439 [zed2.zed_state_publisher] [INFO] got segment zed2_temp_right_link
[zed_wrapper-2] 1645781017.386122 [0] zed_wrappe: using network interface eth0 (udp/192.168.22.157) selected arbitrarily from: eth0, br-436a6caa4e28, docker0
[zed_wrapper-2] 1645781017.427309942 [zed2.zed_node] [INFO] ********************************
[zed_wrapper-2] 1645781017.427735081 [zed2.zed_node] [INFO]       ZED Camera Component 
[zed_wrapper-2] 1645781017.427854446 [zed2.zed_node] [INFO] ********************************
[zed_wrapper-2] 1645781017.427942258 [zed2.zed_node] [INFO]  * namespace: /zed2
[zed_wrapper-2] 1645781017.428024630 [zed2.zed_node] [INFO]  * node name: zed_node
[zed_wrapper-2] 1645781017.428089369 [zed2.zed_node] [INFO] ********************************
[zed_wrapper-2] 1645781017.428171132 [zed2.zed_node] [INFO] *** DEBUG parameters ***
[zed_wrapper-2] 1645781017.428704339 [zed2.zed_node] [INFO]  * Debug mode: FALSE
[zed_wrapper-2] 1645781017.428839897 [zed2.zed_node] [INFO] *** GENERAL parameters ***
[zed_wrapper-2] 1645781017.429243563 [zed2.zed_node] [INFO]  * Camera model: zed2 - ZED 2
[zed_wrapper-2] 1645781017.429635900 [zed2.zed_node] [INFO]  * SDK Verbose: 1
[zed_wrapper-2] 1645781017.429975819 [zed2.zed_node] [INFO]  * SVO: live
[zed_wrapper-2] 1645781017.430347035 [zed2.zed_node] [INFO]  * SVO Loop: TRUE
[zed_wrapper-2] 1645781017.430714475 [zed2.zed_node] [INFO]  * SVO Realtime: FALSE
[zed_wrapper-2] 1645781017.431018873 [zed2.zed_node] [INFO]  * Camera name: zed2
[zed_wrapper-2] 1645781017.431360584 [zed2.zed_node] [INFO]  * Camera ID: 0
[zed_wrapper-2] 1645781017.431765241 [zed2.zed_node] [INFO]  * Camera SN: 0
[zed_wrapper-2] 1645781017.432131497 [zed2.zed_node] [INFO]  * Camera timeout [sec]: 5
[zed_wrapper-2] 1645781017.432425302 [zed2.zed_node] [INFO]  * Camera reconnection temptatives: 5
[zed_wrapper-2] 1645781017.432764869 [zed2.zed_node] [INFO]  * Camera framerate: 30
[... 98 lines cut ...]
[zed_wrapper-2] 1645781017.470839077 [zed2.zed_node] [INFO]  * '/zed2/zed_node/start_svo_rec'
[zed_wrapper-2] 1645781017.471856978 [zed2.zed_node] [INFO]  * '/zed2/zed_node/stop_svo_rec'
[zed_wrapper-2] 1645781017.472852189 [zed2.zed_node] [INFO]  * '/zed2/zed_node/toggle_svo_pause'
[zed_wrapper-2] 1645781017.472938785 [zed2.zed_node] [INFO] ***** STARTING CAMERA *****
[zed_wrapper-2] 1645781017.472997347 [zed2.zed_node] [INFO] SDK Version: 3.7.0 - Build 40976_c89a574c
[zed_wrapper-2] 1645781017.492736450 [zed2.zed_node] [INFO] Parameter change callback
[zed_wrapper-2] 1645781017.492930539 [zed2.zed_node] [WARN] qos_overrides./tf.publisher.durability is not a dynamic parameter
[zed_wrapper-2] terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterValueException'
[zed_wrapper-2]   what():  parameter 'qos_overrides./tf.publisher.durability' could not be set: qos_overrides./tf.publisher.durability is not a dynamic parameter
[ERROR] [zed_wrapper-2]: process has died [pid 147, exit code -6, cmd '/opt/ros/galactic/install/lib/zed_wrapper/zed_wrapper --ros-args -r __node:=zed_node -r __ns:=/zed2 --params-file /opt/ros/galactic/install/share/zed_wrapper/config/common.yaml --params-file /opt/ros/galactic/install/share/zed_wrapper/config/zed2.yaml --params-file /tmp/launch_params_3zbkxils'].
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[robot_state_publisher-1] 1645781020.118573588 [rclcpp] [INFO] signal_handler(signal_value=2)
[INFO] [robot_state_publisher-1]: process has finished cleanly [pid 146]```

### ZED Camera model

ZED2

### Environment

```shell
Nvidia Jetson Docker Container
ROS Galactic
Image built based on: https://github.com/dusty-nv/jetson-containers
```

Myzhar, 2022-02-25:
@weitung the ZED ROS2 Wrapper is not officially supporting the ROS2 Galactic release.
Can you verify if ROS2 Galactic modified something regarding the QoS parameters and the relative override?

weitung, 2022-02-25:
@Myzhar Thanks for the clarification. I quickly scan through the change log and it seems like it has something to do with this: https://docs.ros.org/en/foxy/Releases/Release-Galactic-Geochelone.html#externally-configure-qos-at-start-up, and maybe tf2_ros did the external configuration by sending override params, which ends up not handling and drop to this else clause: https://github.com/stereolabs/zed-ros2-wrapper/blob/master/zed_components/src/zed_camera/src/zed_camera_component.cpp#L1785

weitung, 2022-02-25:
I also did a quick test by setting `result.successful = true;` in that else clause and it successfully launches.

Myzhar, 2022-02-25:
Currently, the ZED ROS2 Wrapper provides its own method to overrides the QoS profiles for all the senders.
The new Galactic feature requires a deep reworking of this part and for the moment we will not work on that.
The new feature will be adopted when the next ROS2 LTS [Humble Hawksbill](https://docs.ros.org/en/foxy/Releases/Release-Humble-Hawksbill.html) will be released in May.

Myzhar, 2022-02-25:
I add this link for future reference:
http://design.ros2.org/articles/qos_configurability.html

kquigley29, 2022-03-05:
@weitung @Myzhar I had this same problem working with galactic. I moved [this line](https://github.com/stereolabs/zed-ros2-wrapper/blob/55a4302c6365ab2742e61f4cbfbd34960d3955b2/zed_components/src/zed_camera/src/zed_camera_component.cpp#L264) to the very end of the constructor to allow all parameters to be declared and set before the parameter change callback is put in place. The node ran as expected afterwards

ashishk88, 2022-04-07:
> @weitung @Myzhar I had this same problem working with galactic. I moved [this [...]

@kquigley29 , can you see any messages being published on the camera_info topics after this change?

Marcelomm103, 2022-07-21:
Can anyone confirm to me that the wrapper is running as expected in ROS2 Galactic ?

nakai-omer, 2022-08-09:
This also occurs in ROS2 Humble

closed 2022-10-27 by Myzhar