stereolabs/zed-ros2-wrapper #81: Point cloud not working sometimes

Void12987, 2022-02-25:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Although I am able to get point cloud data from x86_64 Foxy without Docker or x86_64 Melodic without Docker, I am unable to get it from arm64 Foxy Docker on the Jetson Nano. Although on arm64 Foxy Docker, I was able to get anything else: images, depth images, etc.

### Steps to Reproduce

Dockerfile
* dustynv's foxy-slam for r32.6.1
* do [Bmoradi93's stuff](https://github.com/stereolabs/zed-ros2-wrapper/issues/31)
* run the republish command

docker-compose.yaml
* run zed2.launch.py

### Expected Result

`ros2 topic hz zed2/zed_node/point_cloud/cloud_registered` outputs anything

### Actual Result

The cloud_registered topic is not doing anything, but all other topics are working

### ZED Camera model

ZED2

### Environment

```shell
Jetson Nano 
ZED 3.6
Foxy Docker
```

Myzhar, 2022-02-26:
Hi @Void12987 
What are the values of the following parameters?
`grab_frame_rate`
`pub_frame_rate`
`point_cloud_freq`
Can you post the output of the command `ros2 doctor` while the nodes are running?
Have you changed the QoS policies of the topics?

Void12987, 2022-02-28:
@Myzhar I got the following using `ros2 param get`:

- grab_frame_rate = 30
- pub_frame_rate = 15
- point_cloud_freq = 10

I don't know what QoS policies are; it's likely they are still default.

Output of `ros2 doctor`:

```
root@nano:/# ros2 doctor
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: launch_ros has been updated to a new version. local: 0.11.2 < required: 0.11.6
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: rclcpp_lifecycle has been updated to a new version. local: 2.3.1 < required: 2.4.1
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: zstd_vendor has been updated to a new version. local: 0.3.8 < required: 0.3.9
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: rosbag2_tests has been updated to a new version. local: 0.3.8 < required: 0.3.9
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: ament_cmake_core has been updated to a new version. local: 0.9.8 < required: 0.9.9
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: rcl_yaml_param_parser has been updated to a new version. local: 1.1.11 < required: 1.1.13
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: ament_cmake has been updated to a new version. local: 0.9.8 < required: 0.9.9
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: performance_test_fixture has been updated to a new version. local: 0.0.6 < required: 0.0.8
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: ament_cmake_python has been updated to a new version. local: 0.9.8 < required: 0.9.9
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: tf2 has been updated to a new version. local: 0.13.10 < required: 0.13.13
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: ament_cmake_export_definitions has been updated to a new version. local: 0.9.8 < required: 0.9.9
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: ament_cmake_export_include_directories has been updated to a new version. local: 0.9.8 < required: 0.9.9
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: rosidl_typesupport_fastrtps_cpp has been updated to a new version. local: 1.0.3 < required: 1.0.4
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: rclpy has been updated to a new version. local: 1.0.5 < required: 1.0.8
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: actionlib_msgs has been updated to a new version. local: 2.0.4 < required: 2.0.5
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: sros2_cmake has been updated to a new version. local: 0.9.4 < required: 0.9.5
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: rmw_cyclonedds_cpp has been updated to a new version. local: 0.7.6 < required: 0.7.8
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: launch has been updated to a new version. local: 0.10.5 < required: 0.10.8
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: ament_cmake_pytest has been updated to a new version. local: 0.9.8 < required: 0.9.9
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: tf2_tools has been updated to a new version. local: 0.13.10 < required: 0.13.13
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: tf2_eigen has been updated to a new version. local: 0.13.10 < required: 0.13.13
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: std_srvs has been updated to a new version. local: 2.0.4 < required: 2.0.5
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: rosbag2_storage_default_plugins has been updated to a new version. local: 0.3.8 < required: 0.3.9
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: tf2_bullet has been updated to a new version. local: 0.13.10 < required: 0.13.13
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: rmw_implementation has been updated to a new version. local: 1.0.2 < required: 1.0.3
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: fastrtps_cmake_module has been updated to a new version. local: 1.0.3 < required: 1.0.4
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: sros2 has been updated to a new version. local: 0.9.4 < required: 0.9.5
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: rosidl_typesupport_fastrtps_c has been updated to a new version. local: 1.0.3 < required: 1.0.4
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/package.py: 113: UserWarning: ament_cmake_test has been updated to a new version. local: 0.9.8 < required: 0.9.9
[... cut ...]
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /zed2/zed_node/temperature/imu.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /zed2/zed_node/temperature/left.
/opt/ros/foxy/install/lib/python3.6/site-packages/ros2doctor/api/topic.py: 53: UserWarning: Publisher without subscriber detected on /zed2/zed_node/temperature/right.

All 4 checks passed

root@nano:/# 
root@nano:/# nathan@nano:~/sar/docker/arm64$ 
```

Myzhar, 2022-02-28:
Are you sending all this amount of data over Wi-Fi?

Void12987, 2022-02-28:
I don't believe so. Our docker-compose.yaml has `network_mode: host`, and I'm only working with 1 Jetson Nano. I also do `ros2 topic hz` within a terminal in docker.

closed 2022-04-26 by github-actions[bot]