stereolabs/zed-ros2-wrapper #85: Depth image not changing

Void12987, 2022-03-18:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Installing Zed on Foxy works for every topic except depth data and point cloud data. This might be a duplicate of https://github.com/stereolabs/zed-ros2-wrapper/issues/81, but not sure.

### Steps to Reproduce

1. Install zed_ros_wrapper as per the README instructions
2. `ros2 run image_transport republish compressed --ros-args -r in/compressed:=image_compressed -r out:=image_raw`
3. `ros2 launch zed_wrapper zed2.launch.py`

### Expected Result

`ros2 topic echo /zed2/zed_node/depth/depth_registered` has changing data

### Actual Result

it's the same black image

### ZED Camera model

ZED2

### Environment

```shell
OS: Ubuntu 20.04
CPU: x86_64
GPU Nvidia 1060
ZED SDK: 3.7
ROS Foxy
```

### Anything else?

1. If the republish node is necessary to get the data, should it be on the README?
2. If it's not necessary, what could be the reason why no data (even normal image data) can be retrieved in ROS?

Myzhar, 2022-03-18:
Hi @Void12987 
`image_transport` is faulty in ROS2 Foxy.
It's a known problem:
https://github.com/stereolabs/zed-ros2-wrapper#image-transport-plugins-and-compressed-topics

Void12987, 2022-03-18:
Hi @Myzhar, 

Although I removed the image_transport packages, is there anything I can do to get depth data working?

Void12987, 2022-03-18:
Yes `ros2 topic hz` works fine, but does your `ros2 topic echo` have changing data?

Myzhar, 2022-03-18:
Looking at the data in `ros2 topic echo` is not a good way to debug and understand if this kind of topic is correctly published.

```
- 255
- 255
- 127
- '...'
---
```
this means that not all the data are displayed for obvious reasons. Furthermore, you are trying to read floating-point values from a vector of uint8_t values... 

if you want to be sure that the depth map is changing you can use instead the `rqt` viewer by selecting `PlugIns` -> `Visualization` -> `Image View`
[image]

If you have not installed rqt:

```
sudo apt install ros-foxy-rqt-*
```

Void12987, 2022-03-18:
Below is a screenshot of using depth quality NEURAL mode, but the image is black.

[image]

Interestingly, when I try to run depthimage_to_laserscan, zed_wrapper throws an error about the depth and rgb are async.

[image]

When I run depth quality at PERFORMANCE mode, then the depth data shows up, but it's a bit choppy, not working in certain lighting conditions

[image]

Maybe my GPU is too old?

Void12987, 2022-03-18:
Thanks so much for helping; it's working now.

[image]

I thought I'd add some suggestions on updating the README; if you'd like me to open a PR, let me know and I'll do it:

1. mention to optimize using ZED_Diagnostic
2. In the installation prompts, it says cudnn is only necessary for object detection, but now that it's required for the default mode for depth detection; it could mention that too, because previously I didn't have cudnn installed
3. It wasn't clear to me that the v3.0.0 was necessary because afterwards it says to install the latest version, which didn't work, maybe we could erase that part?

Anyways thanks for the help. I hope to be creating a robotics marketplace soon for free advertisement of robotics products, and the ZED camera is the best for that!

closed 2022-03-18 by Void12987

Myzhar, 2022-03-18:
Do not hesitate to suggest README modifications.
User feedback is always the best way to improve our tools and software.

Void12987, 2022-03-18:
Cool, although I don't know how to update the instructions on the install script, I will open a PR for updating this README at some point next.