stereolabs/zed-ros2-wrapper #87: cant get ZED open on Jetson Xavier

an99990, 2022-03-22:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I am trying to launch the zed in a jetson xavier but i am getting usb issues

### Steps to Reproduce

ros2 launch zed_wrapper zed2.launch.py

### Expected Result

no issues

### Actual Result

```
root@645b9a04d775:/opt/halodi/ros_ws# ros2 launch zed_wrapper zed2.launch.py
[INFO] [launch]: All log files can be found below /root/.ros/log/2022-03-22-20-53-54-132821-645b9a04d775-121
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [zed_wrapper-1]: process started with pid [124]
[zed_wrapper-1] 1647982434.584965655 [zed2.zed_node] [INFO] ********************************
[zed_wrapper-1] 1647982434.585500603 [zed2.zed_node] [INFO]       ZED Camera Component 
[zed_wrapper-1] 1647982434.585570875 [zed2.zed_node] [INFO] ********************************
[zed_wrapper-1] 1647982434.585629596 [zed2.zed_node] [INFO]  * namespace: /zed2
[zed_wrapper-1] 1647982434.585685052 [zed2.zed_node] [INFO]  * node name: zed_node
[zed_wrapper-1] 1647982434.585763165 [zed2.zed_node] [INFO] ********************************
[zed_wrapper-1] 1647982434.585841917 [zed2.zed_node] [INFO] *** DEBUG parameters ***
[zed_wrapper-1] 1647982434.586074303 [zed2.zed_node] [INFO]  * Debug mode: FALSE
[zed_wrapper-1] 1647982434.586206208 [zed2.zed_node] [INFO] *** GENERAL parameters ***
[zed_wrapper-1] 1647982434.586427266 [zed2.zed_node] [INFO]  * Camera model: zed2 - ZED 2
[zed_wrapper-1] 1647982434.586647747 [zed2.zed_node] [INFO]  * SDK Verbose: 1
[zed_wrapper-1] 1647982434.586830245 [zed2.zed_node] [INFO]  * SVO: live
[zed_wrapper-1] 1647982434.586967270 [zed2.zed_node] [INFO]  * SVO Loop: TRUE
[zed_wrapper-1] 1647982434.587117639 [zed2.zed_node] [INFO]  * SVO Realtime: TRUE
[zed_wrapper-1] 1647982434.587326152 [zed2.zed_node] [INFO]  * Camera name: zed2
[zed_wrapper-1] 1647982434.587489577 [zed2.zed_node] [INFO]  * Camera ID: 0
[zed_wrapper-1] 1647982434.587645963 [zed2.zed_node] [INFO]  * Camera SN: 0
[zed_wrapper-1] 1647982434.587853228 [zed2.zed_node] [INFO]  * Camera timeout [sec]: 5
[zed_wrapper-1] 1647982434.588037613 [zed2.zed_node] [INFO]  * Camera reconnection temptatives: 5
[zed_wrapper-1] 1647982434.588159502 [zed2.zed_node] [INFO]  * Camera framerate: 30
[zed_wrapper-1] 1647982434.588275119 [zed2.zed_node] [INFO]  * GPU ID: -1
[zed_wrapper-1] 1647982434.588419856 [zed2.zed_node] [INFO]  * Camera resolution: 2 - HD720
[zed_wrapper-1] 1647982434.588559793 [zed2.zed_node] [INFO]  * Camera self calibration: TRUE
[zed_wrapper-1] 1647982434.588735827 [zed2.zed_node] [INFO]  * Camera flip: FALSE
[zed_wrapper-1] 1647982434.588865300 [zed2.zed_node] [INFO]  * [DYN] Publish framerate [Hz]: 15 
[zed_wrapper-1] 1647982434.588932500 [zed2.zed_node] [INFO] *** VIDEO parameters ***
[zed_wrapper-1] 1647982434.589144790 [zed2.zed_node] [INFO]  * Use old extrinsic parameters: 0
[zed_wrapper-1] 1647982434.589305079 [zed2.zed_node] [INFO]  * [DYN] Image downsample factor: 1 
[zed_wrapper-1] 1647982434.589471320 [zed2.zed_node] [INFO]  * [DYN] Brightness: 4
[zed_wrapper-1] 1647982434.589658105 [zed2.zed_node] [INFO]  * [DYN] Contrast: 4
[zed_wrapper-1] 1647982434.589776538 [zed2.zed_node] [INFO]  * [DYN] Hue: 0
[zed_wrapper-1] 1647982434.589909115 [zed2.zed_node] [INFO]  * [DYN] Saturation: 4
[zed_wrapper-1] 1647982434.590026908 [zed2.zed_node] [INFO]  * [DYN] Sharpness: 4
[zed_wrapper-1] 1647982434.590249150 [zed2.zed_node] [INFO]  * [DYN] Gamma: 8
[zed_wrapper-1] 1647982434.590419583 [zed2.zed_node] [INFO]  * [DYN] Auto Exposure/Gain: TRUE
[zed_wrapper-1] 1647982434.590547360 [zed2.zed_node] [INFO]  * [DYN] Exposure: 80
[zed_wrapper-1] 1647982434.590665345 [zed2.zed_node] [INFO]  * [DYN] Gain: 80
[zed_wrapper-1] 1647982434.590808194 [zed2.zed_node] [INFO]  * [DYN] Auto White Balance: TRUE
[zed_wrapper-1] 1647982434.590924803 [zed2.zed_node] [INFO]  * [DYN] White Balance Temperature: 42
[zed_wrapper-1] 1647982434.591079236 [zed2.zed_node] [INFO]  * Video QoS History: KEEP_LAST
[zed_wrapper-1] 1647982434.591236325 [zed2.zed_node] [INFO]  * Video QoS History depth: 1
[zed_wrapper-1] 1647982434.591380006 [zed2.zed_node] [INFO]  * Video QoS Reliability: RELIABLE
[zed_wrapper-1] 1647982434.591491111 [zed2.zed_node] [INFO]  * Video QoS Durability: VOLATILE
[zed_wrapper-1] 1647982434.591541447 [zed2.zed_node] [INFO] *** DEPTH parameters ***
[zed_wrapper-1] 1647982434.591678952 [zed2.zed_node] [INFO]  * Depth quality: 3 - ULTRA
[zed_wrapper-1] 1647982434.591805673 [zed2.zed_node] [INFO]  * Depth downsample factor: 1 
[... 104 lines cut ...]
[zed_wrapper-1] [ZED][Init][Info] Timeout in 3991msec
[zed_wrapper-1] [ZED][Init][Info] Opening Camera. Attempt #2
[zed_wrapper-1] [ZED][ERROR] Camera Open Internal Error: 4
[zed_wrapper-1] [ZED][Init][Info] Waiting 945 msec before next opening attempt.
[zed_wrapper-1] [ZED][Init][Info] Trying to force a device reboot to recover the video module.
[zed_wrapper-1] [ZED][Init][Info] Timeout in 1003msec
[zed_wrapper-1] [ZED][Init][Info] Opening Camera. Attempt #3
[zed_wrapper-1] [ZED][ERROR] Camera Open Internal Error: 4
[zed_wrapper-1] [ZED][Init][Info] Waiting 1391 msec before next opening attempt.
[zed_wrapper-1] [ZED][Init][Info] Camera opening timeout reached
[zed_wrapper-1] 1647982440.794486129 [zed2.zed_node] [WARN] Error opening camera: CAMERA FAILED TO SETUP
[zed_wrapper-1] 1647982440.794849971 [zed2.zed_node] [INFO] Please verify the USB3 connection
```

### ZED Camera model

ZED2

### Environment

```shell
OS: UBUNTU 18.04
CPU : ARM
ZED : ZED2
```

Myzhar, 2022-03-22:
This is not a bug of the ROS2 wrapper, but mostly a problem with your camera, your Jetson or the SDK configuration.
Please write an email to support@stereolabs.com attaching the report file generated by the ZED Diagnostic tool executed on the Jetson

closed 2022-03-22 by Myzhar

an99990, 2022-03-22:
I cant do ZED_Diagnostic is not an option, maybe i installed the SDK wrong.