stereolabs/zed-ros2-wrapper #89: ipc node for zed components

felrock, 2022-04-01:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

Copying images is not good. ROS2 has support for IPC, so why not use it. The changes should not be very large, maybe there is even another way to do it. What do you think?

### Anything else?

https://docs.ros.org/en/foxy/Tutorials/Intra-Process-Communication.html

Myzhar, 2022-04-01:
Hi @felrock 
there is a problem with IPC: 
it does not support topics with TRANSIENT LOCAL (latch messages) QoS:
https://github.com/stereolabs/zed-ros2-wrapper/blob/master/zed_components/src/zed_camera/src/zed_camera_component.cpp#L3402-L3411

If you try to enable IPC the nodelet fails at starting with an error related to this QoS setting.

Any suggestion?

felrock, 2022-04-04:
Hey, okey that makes sense. How about changing the QoS to volatile or using the SensorDataQoS ([like in the tutorial](https://github.com/ros2/demos/blob/foxy/intra_process_demo/include/image_pipeline/camera_node.hpp)?

Myzhar, 2022-04-04:
Yes, we could do that. A latched topic is better in that case because it's information that never changes.
But also `camera_info` is information that never changes but it's published with each camera frame...
I noted the feature request and I'm going to work on the modification in the next few days.

felrock, 2022-04-04:
Yeah, that sound really good.

felrock, 2022-04-08:
Here is another piece of code that could be interesting for this case, https://github.com/ros2/demos/blob/master/demo_nodes_cpp/src/topics/talker_loaned_message.cpp
It uses iceorxy for IPC

Myzhar, 2022-09-26:
IPC now is available in the `ros2_humble` branch and it will be merged soon into the `master` branch
@felrock The branch is compatible with Foxy, so you can test it.
See commit https://github.com/stereolabs/zed-ros2-wrapper/commit/197c55db9f40e0d6af70555a064b0e1f592ac49a

closed 2022-09-26 by Myzhar