stereolabs/zed-ros2-wrapper #94: Disabling depth makes zed_wrapper crash. Position tracking cannot be disabled and runs every time.

TwardzikTomas, 2022-06-08:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I have noticed, that whenever I try to disable the depth map computation "commo.yaml->depth:quality", the zed_node crashes without giving any appropriate error message. 
Also it seems that you cannot disable position tracking with the "common.yaml->pos_tracking_enabled" flag. 

Maybe I have misunderstood something in your documentation, there might be some setting in conflict causing all these problems and in that case I sincerely apologize for my incompetence and blindness.

Here is the my "common.yaml" file:

        general:
            svo_file:                   ''
            svo_loop:                   false                               # Enable loop mode when using an SVO as input source
            svo_realtime:               false                               
            camera_timeout_sec:         5
            camera_max_reconnect:       5
            camera_flip:                false
            zed_id:                     0
            serial_number:              0
            resolution:                 3                                   # '0': HD2K, '1': HD1080, '2': HD720, '3': VGA
            sdk_verbose:                true
            grab_frame_rate:            60                                # ZED SDK internal grabbing rate
            pub_frame_rate:             40.0                                # [DYNAMIC] - frequency of publishing of visual images and depth images
            gpu_id:                     -1

        video:
            extrinsic_in_camera_frame:  false                               # if `false` extrinsic parameter in `camera_info` will use ROS native frame (X FORWARD, Z UP) instead of the camera frame (Z FORWARD, Y DOWN) [`true` use old behavior as for version < v3.1]
            img_downsample_factor:      1.0                                 # Resample factor for image data matrices [0.01,1.0] The SDK works with native data sizes, but publishes rescaled matrices
            brightness:                 4                                   # [DYNAMIC]
            contrast:                   4                                   # [DYNAMIC]
            hue:                        0                                   # [DYNAMIC]
            saturation:                 4                                   # [DYNAMIC]
            sharpness:                  4                                   # [DYNAMIC]
            gamma:                      8                                   # [DYNAMIC] - Requires SDK >=v3.1
            auto_exposure_gain:         true                                # [DYNAMIC]
            exposure:                   80                                  # [DYNAMIC]
            gain:                       80                                  # [DYNAMIC]
            auto_whitebalance:          true                                # [DYNAMIC]
            whitebalance_temperature:   42                                  # [DYNAMIC] - [28,65] works only if `auto_whitebalance` is false
            qos_history:                1                                   # '1': KEEP_LAST - '2': KEEP_ALL
            qos_depth:                  1                                   # Queue size if using KEEP_LAST
            qos_reliability:            1                                   # '1': RELIABLE - '2': BEST_EFFORT -
            qos_durability:             1                                   # '1': TRANSIENT_LOCAL - '2': VOLATILE

        depth:
            quality:                    0                                   # '0': NONE, '1': PERFORMANCE, '2': QUALITY, '3': ULTRA - Note: if '0' all the modules that requires depth extraction are disabled by default (Pos. Tracking, Obj. Detection, Mapping, ...)
            sensing_mode:               0                                   # '0': STANDARD, '1': FILL
            depth_stabilization:        false                                # Forces positional tracking to start if 'true'
            openni_depth_mode:          false                               # 'false': 32bit float [meters], 'true': 16bit unsigned int [millimeters]
            depth_downsample_factor:    0.1                                 # Resample factor for depth data matrices [0.01,1.0] The SDK works with native data sizes, but publishes rescaled matrices (depth map, point cloud, ...)
            point_cloud_freq:           0.1                                # [DYNAMIC] - frequency of the pointcloud publishing (equal or less to `grab_frame_rate` value)
            depth_confidence:           50                                  # [DYNAMIC]
            depth_texture_conf:         100                                 # [DYNAMIC]
            remove_saturated_areas:     true                                # [DYNAMIC]
            qos_history:                1                                   # '1': KEEP_LAST - '2': KEEP_ALL
            qos_depth:                  1                                   # Queue size if using KEEP_LAST
            qos_reliability:            1                                   # '1': RELIABLE - '2': BEST_EFFORT -
            qos_durability:             2                                   # '1': TRANSIENT_LOCAL - '2': VOLATILE

        pos_tracking:
            pos_tracking_enabled:       true                               # True to enable positional tracking from start
            publish_tf:                 true                                # publish `odom -> base_link` TF
            publish_map_tf:             true                                # publish `map -> odom` TF
            base_frame:                 'base_link'                         # use the same name as in the URDF file
            map_frame:                  'map'
            odometry_frame:             'vis_odom'
[... 552 lines cut ...]
[zed_wrapper-2] 1654705871.358025516 [zed2.zed_node] [INFO]  * Translation: {0.000,-0.060,0.000}
[zed_wrapper-2] 1654705871.358153106 [zed2.zed_node] [INFO]  * Rotation: {0.000,-0.000,0.000}
[zed_wrapper-2] 1654705871.358252662 [zed2.zed_node] [INFO] Static transform Camera Center to Base [zed2_camera_center -> base_link]
[zed_wrapper-2] 1654705871.358342938 [zed2.zed_node] [INFO]  * Translation: {-0.109,0.000,-0.397}
[zed_wrapper-2] 1654705871.358389276 [zed2.zed_node] [INFO]  * Rotation: {0.000,-15.041,0.000}
[zed_wrapper-2] 1654705871.458715270 [zed2.zed_node] [INFO] Initial ZED left camera pose (ZED pos. tracking): 
[zed_wrapper-2] 1654705871.459129080 [zed2.zed_node] [INFO]  * T: [0.208164,0.06,0.354663]
[zed_wrapper-2] 1654705871.459296064 [zed2.zed_node] [INFO]  * Q: [0,0.130878,0,0.991398]
[zed_wrapper-2] 1654705871.478307962 [zed2.zed_node] [INFO] Odometry aligned to last tracking pose

Which clearly starts the position tracking.

### ZED Camera model

ZED2

### Environment

```shell
OS: nvidia l4t32.7
HardWare: Nvidia Jetson Xavier AGX
ZED SDK: 3.7
ROS version: Foxy
Running in docker image
```

Myzhar, 2022-06-08:
Hi @TwardzikTomas 
to disable positional tracking you must also set `publish_tf` to `false`.

Please note that with this configuration the TF tree is no more valid and you need other nodes to publish the required TF chain:
`map` -> `odom` -> `base_link` -> `<zed>_base_link`

TwardzikTomas, 2022-06-09:
Hello @Myzhar,
thank you for your suggestion. Nevertheless, all the position tracking topics are still advertised and once you request any of the messages, the position tracking resumes its action, but with the caveat you have mentioned, the TF tree is disconnected.

On the other note, have you been able to reproduce depth disabling crash?

Myzhar, 2022-06-09:
Not yet, I'm going to check the problem next week

TwardzikTomas, 2022-06-10:
I have a minor update.
Switching debug mode on, gives me following log before the crash:

[zed_wrapper-2] 1654866903.598805564 [zed2.zed_node] [INFO] Published static TF: 'zed2_imu_link' -> 'zed2_left_camera_frame'
[zed_wrapper-2] 1654866903.609808227 [zed2.zed_node] [DEBUG] Grab thread started
[zed_wrapper-2] 1654866903.615226804 [zed2.zed_node] [DEBUG] Sensors callback called
[zed_wrapper-2] 1654866903.615446013 [zed2.zed_node] [DEBUG] Sensors callback: counting subscribers
[zed_wrapper-2] 1654866903.616069817 [zed2.zed_node] [DEBUG] Sensors callback: Grab data and setup timestamps
[zed_wrapper-2] 1654866903.620841292 [zed2.zed_node] [DEBUG] Video Depth callback called
[ERROR] [zed_wrapper-2]: process has died [pid 12998, exit code -11, cmd '/code/ros2_ws/install/zed_wrapper/lib/zed_wrapper/zed_wrapper --ros-args -r __node:=zed_node -r __ns:=/zed2 --params-file /code/ros2_ws/install/zed_wrapper/share/zed_wrapper/config/common.yaml --params-file /code/ros2_ws/install/zed_wrapper/share/zed_wrapper/config/zed2.yaml --params-file /tmp/launch_params_9tbb9o12']

So it seems that the major issue might arise somewhere within the Video Depth CB.

Myzhar, 2022-06-10:
It should not be created if the depth extraction is disabled. I note this. Thank you

Myzhar, 2022-06-13:
Hi @TwardzikTomas 
I just pushed the [`test_no_depth`](https://github.com/stereolabs/zed-ros2-wrapper/tree/test_no_depth) branch that should fix the "no depth" behavior.

By setting `depth/quality` to `0` the depth extraction and all the sub-modules depending on it should be automatically correctly disabled by now.

Please test the branch and send me feedback. In case it will work as expected also for you I merge it into the master branch

Myzhar, 2022-06-16:
@TwardzikTomas did you have the opportunity to test the fix?

TwardzikTomas, 2022-06-16:
Hello, unfortunately I was not, I hope to get to it by the end of the week.

TwardzikTomas, 2022-06-21:
Hello, I just came to work, so I'll get right on to it. I don't want to spam you at all, I just wanted to show that this is not a dead thread.

Myzhar, 2022-06-21:
OK @TwardzikTomas I'm waiting for your feedback before merging the patch into the `master` branch

PR #95 (2022-06-22): Test no depth
Fix #94 : wrong behaviors when Depth extraction was disabled
closed 2022-06-22 by Myzhar