stereolabs/zed-ros2-wrapper #101: Ubuntu 22 over ARM, installing ZED SDK 3.7.6 falsely presumes it is Tegra Linux

nakai-omer, 2022-08-14:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hello,

We are using this project for a ROS2 Humble based robot over Ubuntu 22.04.
Trying to build a docker image for an ARM based processer, while installing the ZED SDK caused it to try to look for Linux Tegra throwing the following error:
```
head: cannot open '/etc/nv_tegra_release' for reading: No such file or directory
dpkg-query: no packages found matching nvidia-l4t-core
Detected Tegra_L4T., required exact Ubuntu22
Installing...

This installer is not compatible with ARM processor, for the Jetson Tegra check the special installers available at stereolabs.com/developers
```

There is no /etc/nv_tegra_release, becuase it is not a Tegra Linux OS. 

### Steps to Reproduce

1. Create the following Dockerfile and docker compose file
Dockerfile:
```
ARG NVIDIA_IMAGE=nvidia/cuda:11.7.0-base-ubuntu22.04

FROM $NVIDIA_IMAGE AS builder

ARG ROS_DISTRO=humble
ARG OVERLAY_WS=/opt/ros/nakai
ARG UBUNTU_RELEASE_YEAR=22
ARG CUDA_MAJOR=11
ARG CUDA_MINOR=7
ARG ZED_SDK_MAJOR=3
ARG ZED_SDK_MINOR=7

# Install external deps + ZED sdk for the depth camera
RUN pip install transforms3d \
    python-can \
    && apt-get install --no-install-recommends lsb-release wget less udev -y \
    && wget -q -O ZED_SDK_Linux_Ubuntu${UBUNTU_RELEASE_YEAR}.run https://download.stereolabs.com/zedsdk/${ZED_SDK_MAJOR}.${ZED_SDK_MINOR}/cu${CUDA_MAJOR}${CUDA_MINOR%.*}/ubuntu${UBUNTU_RELEASE_YEAR}

RUN chmod +x ZED_SDK_Linux_Ubuntu${UBUNTU_RELEASE_YEAR}.run ; ./ZED_SDK_Linux_Ubuntu${UBUNTU_RELEASE_YEAR}.run -- silent runtime_only && \
    rm ZED_SDK_Linux_Ubuntu${UBUNTU_RELEASE_YEAR}.run && \
    rm -rf /var/lib/apt/lists/*

```
docker-compose.yaml
```
version: "3.9"
services:
  nakai-robot:
    privileged: true
    network_mode: "host"
    platform: linux/arm64/v8
    container_name: nakai-robot
    image: nakairobotics/nakai:nakai
    build:
      context: .
      dockerfile: Dockerfile
```
2. Run `sudo docker compose build`

### Expected Result

ZED SDK should be installed as expected, and not guess falsely this is a Jetson Tegra OS. 

### Actual Result

head: cannot open '/etc/nv_tegra_release' for reading: No such file or directory
dpkg-query: no packages found matching nvidia-l4t-core
Detected Tegra_L4T., required exact Ubuntu22
Installing...

This installer is not compatible with ARM processor, for the Jetson Tegra check the special installers available at stereolabs.com/developers

### ZED Camera model

ZED Mini

### Environment

```shell
OS: Ubuntu 22.04
CPU: linux/arm64/v8
```

### Anything else?

Please don't tell me to use your docker images, as there is no Ubuntu 22.04 image.

AndreV84, 2022-08-20:
@nakai-omer does the wrapper node builds in your case?

nakai-omer, 2022-08-21:
@AndreV84 What do you mean by wrapper node?

AndreV84, 2022-08-21:
@nakai-omer thank you for your  response
by the wrapper node I refer to the code https://github.com/stereolabs/zed-ros2-wrapper/

nakai-omer, 2022-08-21:
@AndreV84 I can't get to this point, as it requires the sdk to be installed before

Myzhar, 2022-08-23:
This issue is not valid for three reasons:
1. we do not support generic ARM64 devices, but only Nvidia Jetson boards
2. Nvidia Jetson boards do not yet support Ubuntu 22.04
3. The ZED ROS2 Wrapper is not designed for ROS2 Humble, but for ROS2 Foxy

The support for ROS2 Humble will be released in the near future.

closed 2022-08-23 by Myzhar