stereolabs/zed-ros2-wrapper #102: URDF Barometer, Magnetometer, and Temperature Sensor Links For Zed2i

swiz23, 2022-08-19:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Similar to what was done in the ROS1 urdf [here](https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/urdf/zed_macro.urdf.xacro#L128) to include the zed2i camera for the barometer/magnetometer frames, the same should be done in the ROS2 urdf [here](https://github.com/stereolabs/zed-ros2-wrapper/blob/master/zed_wrapper/urdf/zed_macro.urdf.xacro#L128).

### Steps to Reproduce

N/A

### Expected Result

See the other sensor frames being output to the tf tree

### Actual Result

Frames are not output to the tf tree

### ZED Camera model

ZED2i

### Environment

```shell
N/A
```

swiz23, 2022-09-23:
Commenting to prevent the issue from closing.

swiz23, 2022-10-27:
Commenting to prevent issue from closing.

Myzhar, 2022-10-27:
Will be fixed soon