stereolabs/zed-ros2-wrapper #103: jetson Orin devkit DP 50.1 ubuntu 22.04 [ upgraded from 20.04] ros Humble - the node won't build

AndreV84, 2022-08-20:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

steps which are supported to build the node for humble ?

### Use-Case

-- stderr: zed_components                          
CMake Error at /workspaces/zed_ws/install/zed_interfaces/share/zed_interfaces/cmake/ament_cmake_export_targets-extras.cmake:18 (message):
  Failed to find exported target names in
  '/workspaces/zed_ws/install/zed_interfaces/share/zed_interfaces/cmake/export_zed_interfaces__rosidl_generator_cExport.cmake'
Call Stack (most recent call first):
  /workspaces/zed_ws/install/zed_interfaces/share/zed_interfaces/cmake/zed_interfacesConfig.cmake:41 (include)
  CMakeLists.txt:71 (find_package)

---
Failed   <<< zed_components [2.00s, exited with code 1]

davidbro-in, 2022-08-23:
Hey @AndreV84 I am experiencing the same error but using ROS2 Galactic in Ubuntu Bionic. The weird thing is that it was working before. Did you find any solution?

Myzhar, 2022-08-23:
ROS2 Humble is not yet supported by the ROS2 wrapper. We will work on this feature in the new few weeks.
ROS2 Galactic will not be officially supported because it's not an LTS version and the EOL date is very close.
For more information:
https://docs.ros.org/en/galactic/Releases.html

Myzhar, 2022-08-23:
PS Upgrading a Jetson board to a version of the Operating System not officially supported is not a good practice and the final result will be an unstable machine.
In this particular case, the Jetpack 5.0.1 is based on Ubuntu 20.04 and it should not be upgraded to Ubuntu 22.04 because Nvidia does not provide drivers for this version and on our side, we do not provide a ZED SDK version for Ubuntu 22.04 running on Jetson.

Myzhar, 2022-08-23:
I close the issue because it's not valid.

closed 2022-08-23 by Myzhar

davidbro-in, 2022-08-23:
@AndreV84 I fixed the issue by downgrading CMake to version 3.23.2. Apparently ament_cmake has an issue with CMake 3.24+

See: https://github.com/ament/ament_cmake/issues/396

AndreV84, 2022-12-17:
@Myzhar 
How do you add the wrapper into existing ros humble container [nvblox nav nvblox humble provided by nvidia] ?
from running container terminal it won't appear that sdk file installation results in satisfaction of requirements for the wrapper as it would fail with

```
[Processing: nvblox_performance_measurement, realsense_splitter, zed_interfaces]                                                                                           
--- stderr: zed_interfaces                                                                                                        
* Found ROS2 humble
* ROS2 humble is officially supported by this package.
---
Finished <<< zed_interfaces [2min 0s]
Starting >>> zed_components                                                                                                            
--- stderr: zed_components                                                                                                                                               
* Found ROS2 humble
* ROS2 humble is officially supported by this package.
CMake Error at CMakeLists.txt:78 (find_package):
  By not providing "FindZED.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ZED", but
  CMake did not find one.

  Could not find a package configuration file provided by "ZED" (requested
  version 3) with any of the following names:

    ZEDConfig.cmake
    zed-config.cmake

  Add the installation prefix of "ZED" to CMAKE_PREFIX_PATH or set "ZED_DIR"
  to a directory containing one of the above files.  If "ZED" provides a
  separate development package or SDK, be sure it has been installed.

---
Failed   <<< zed_components [1.56s, exited with code 1]
```

AndreV84, 2022-12-17:
zed sdk is already installed systemwide
trying to install it from container gets through some of steps but will throw 
```
To continue you have to accept the EULA. Accept  [Y/n] ?y
Installing...
Installation path: /usr/local/zed
Checking CUDA version...
OK: Found CUDA 11.4
cp: cannot create directory '/usr/local/zed/tools': Permission denied

```
at another steps

Myzhar, 2022-12-17:
The error means that you have not installed the ZED SDK.

PS please never post in a closed issue. 
If it's a bug kindly create a new issue.
If it's a question or a support request please ask on [the Stereolabs forum](https://community.stereolabs.com)

AndreV84, 2022-12-17:
> The error means that you have not installed the ZED SDK.
> 
> PS please never post in a closed issue. If it's a bug kindly create a new issue. If it's a question or a support request please ask on [the Stereolabs forum](https://community.stereolabs.com)

but it is not a closed issue:
I am still not able to run the zed ros wrapper since " Aug 20" when it was emerged
Does Zed SDK need to be installed two times? both outside the container also within the container? or only within the container? but the internal installation would say it can not acceee /usr/local/zed   - shal  Icreate new question/ issue for that concern necessarily?