stereolabs/zed-ros2-wrapper #104: Add expected performance metrics (benchmarks) for zed products in docs

jodusan, 2022-09-06:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

Although documentation is very well written I believe that benchmarking FPS, CPU, GPU usage for depth sensing and positional tracking on  Nvidia NX ( jetson box) ZED BOX for ros-2-wrapper (for example) could provide better estimate of what is expected at runtime. It seems that ZED Sdk processing is quite intensive and leaves little room to do other things (is multi camera setup on xavier NX even possible)? 

Clarifications of these sorts of things would help a lot.

Myzhar, 2022-09-06:
Hi @jodusan
interesting request. The problem is that there are so many possible combinations that it is not easy to provide a benchmark for each of them. A short list of the variables involved:
* resolution
* positional tracking
* object detection (AI model, tracking, ...)
* 3D mapping
* Jetson nvpmodel
* ...

jodusan, 2022-09-06:
I was just reading through your comments here https://github.com/stereolabs/zed-ros2-wrapper/issues/22 is there a preferred DDS for Jetson boards? or at least configuration of the current one (FastDDS?) (In my case Jetson NX). I'm getting around 100% faster performance with ROS 1 Noetic than with ROS 2 Galactic zed-ros-wrapper

jodusan, 2022-09-08:
@Myzhar  I'm trying to profile zed_ros2_wrapper node with valgrind to see where the bottleneck is, but when running zed2i.launch.py built with
`
colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=RelWithDebInfo
`
It just hangs with

"""
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [17922]
[INFO] [zed_wrapper-2]: process started with pid [17924]
[robot_state_publisher-1] Link zed2i_base_link had 1 children
[robot_state_publisher-1] Link zed2i_camera_center had 2 children
[robot_state_publisher-1] Link zed2i_left_camera_frame had 1 children
[robot_state_publisher-1] Link zed2i_left_camera_optical_frame had 0 children
[robot_state_publisher-1] Link zed2i_right_camera_frame had 1 children
[robot_state_publisher-1] Link zed2i_right_camera_optical_frame had 0 children
[robot_state_publisher-1] 1662634222.409792280 [zed2i.zed_state_publisher] [INFO] got segment base_link
[robot_state_publisher-1] 1662634222.410137661 [zed2i.zed_state_publisher] [INFO] got segment zed2i_base_link
[robot_state_publisher-1] 1662634222.410237791 [zed2i.zed_state_publisher] [INFO] got segment zed2i_camera_center
[robot_state_publisher-1] 1662634222.410313312 [zed2i.zed_state_publisher] [INFO] got segment zed2i_left_camera_frame
[robot_state_publisher-1] 1662634222.410388417 [zed2i.zed_state_publisher] [INFO] got segment zed2i_left_camera_optical_frame
[robot_state_publisher-1] 1662634222.410467106 [zed2i.zed_state_publisher] [INFO] got segment zed2i_right_camera_frame
[robot_state_publisher-1] 1662634222.410528899 [zed2i.zed_state_publisher] [INFO] got segment zed2i_right_camera_optical_frame
[INFO] [zed_wrapper-2]: process has finished cleanly [pid 17924]
"""

Any suggestions?