stereolabs/zed-ros2-wrapper #106: The ZED 2 Camera IMU Data is not publishing using the /zed2/zed_node/imu/data.

DanielHailu3, 2022-09-26:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

The ZED 2 Camera IMU Data is not publishing using the topic /zed2/zed_node/imu/data.  I run the zed2.launch.py using ros2 launch and it publishes the left and right camera feed but not the imu data including the raw data.  I am running the ZED 2 ROS 2 wrapper on Nvidia Jetson Xavier with ZED Camera SDK installed on the Xavier.

### Steps to Reproduce

1. ros2 launch zed_wrapper zed2.launch.py
2. ros2 topic hz /zed2/zed_node/imu/data 
3.  Publishes no data
4. ros2 topic echo /zed2/zed_node/imu/data 
5. Publishes no data
6. ros2 topic echo /zed2/zed_node/imu/data_raw
7. Publishes no data
8.  Note: The IMU tf is publishing and working.
...

### Expected Result

I was expecting Sensor IMU data. 

### Actual Result

No Sensor IMU Data.

### ZED Camera model

ZED2

### Environment

```shell
OS: Using Docker container running Ubuntu 22.04 on an Ubuntu 20.04 Xavier 
Hardware: Xavier AGX 
Running ROS 2 Humble
ZED SDK 3.7.7 for L4T 34.1 Beta (JetPack 5.0)
```

Myzhar, 2022-09-26:
Hi @DanielHailu3 
have you checked the compatibility of the QoS profiles?

Myzhar, 2022-09-26:
@DanielHailu3 I just tested the `ros2_humble` branch and it works as expected.
You can check the QoS profile of a topic by using the command
`ros2 topic info -v <topic_name>`
it shows the QoS settings for the publisher and the subscribers.

**Note**: the `ros2_humble` branch is currently under development and will be merged into the `master` branch soon.

closed 2022-11-01 by github-actions[bot]