stereolabs/zed-ros2-wrapper #108: Documentation doesn't provide correct git clone command

KohlhardtC, 2022-10-18:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

The [Getting Started with ROS 2 and ZED ](https://www.stereolabs.com/docs/ros2/) does not show the correct clone command in the **Build the package** section. The command is missing the recursive option. 

The correct clone command is: 
`git clone --recursive https://github.com/stereolabs/zed-ros2-wrapper.git`

On a side note, it might help some new users to remind them to source their ROS2 environment before attempting to run rosdep

### Steps to Reproduce

1. Follow the documentation to build the pacakge
2.
3.
...

### Expected Result

The package builds

### Actual Result

rosdep and colcon build fails because the recursive option is not in the clone command

### ZED Camera model

ZED2i

### Environment

```shell
Ubuntu 20.04
Jetson Orin AGX
ROS2/Foxy
```

Myzhar, 2022-10-18:
Hi @KohlhardtC 
thank you for reporting

Myzhar, 2022-10-27:
The online documentation is updated now

closed 2022-10-27 by Myzhar