stereolabs/zed-ros2-wrapper #109: Image transport compressed format can't be changed

xkaraman, 2022-10-21:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I am using Zed2 on ROS2 using the latest SDK 3.8, CUDA 11.4 on a Jetson NX2.

Launching they `zed2.launch.py` everything works including the `compressed`, `compressedDepth` transport plugins. i can `echo` both of the topics but now i would like to change the `format` of the plugin from 'jpeg` to `png`. As this a configurable parameter of `image_transport` package i should be able to do so.

Unfortunately, i can't  because as far as i can see from `ros2 run rqt_reconfigure rqt_reconfigure`. there is no parameter for the plugin options, only for the original topic ie `zed2/zed_node` `rgb.image_rect_color.format`. Even this though is not configurable with a message like this from terminal running zed launch file: 
` zed_node.rgb.image_rect_color.format is not a dynamic parameter`

### Steps to Reproduce

1. Launch zed2 launch file
2. Run rqt_reconfigure node
3. Try changing image format to png but ends with error
4.  Image format of compressed, compressedDepth and other plugins can't be seen in rqt under zed2 node

### Expected Result

I expect i can change the format of the image transport plugin or any other parameter of it

### Actual Result

Can't chagne due to ` zed_node.rgb.image_rect_color.format is not a dynamic parameter` error

### ZED Camera model

ZED2

### Environment

```shell
Jetpack LT35.1
ARM
Nvidia Jetson NX2 module
ROS Galactic
```

Myzhar, 2022-10-21:
Hi @xkaraman 
ROS2 Galactic is not LTS, so it's not officially supported

closed 2022-10-21 by Myzhar

xkaraman, 2022-10-21:
Should i expect any difference in humble? Is there any differnce on how you handle image transport plugins and their parameters?

Moreover Jetson's Jetpack that is LTS only supports (officialy) galactic due to being based on Ubuntu 20.04.

Myzhar, 2022-10-21:
The official LTS for Ubuntu 20.04 is Foxy. 
Galactic is an intermediate release that will reach the [EOL in the next month](http://docs.ros.org/en/galactic/Releases.html).
You can use ROS2 Humble on Jetson in a Docker container.

Myzhar, 2022-10-21:
There is a known problem with `image_transport` in Foxy:
https://github.com/stereolabs/zed-ros2-wrapper#known-issues

It works correctly instead in `Humble`.

Myzhar, 2022-10-21:
Ah OK, you can fix this behavior by upgrading to the latest version of the `master` branch that we released a few minutes ago.

xkaraman, 2022-10-21:
Ok i will check it out and report back if the problem persists!

Thanks for the quick respondence! :)

xkaraman, 2022-10-24:
I get this error when trying to built the latest `master` branch!
`/home/ssd-nvidia/ros2_workspaces/sensors_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:41:2: error: #error Unsupported ROS2 distro
   41 | #error Unsupported ROS2 distro
`

So, i can't built this at all at ROS2 Galactic?

Myzhar, 2022-10-24:
Hi @xkaraman 
As I told you, we do not officially support Galactic. In the new release, we added CMake commands and C++ macros to avoid the compilation for unsupported distributions.
You can remove them manually to force the compilation and then fix the eventual compilation issues.

However, by reading the Galactic and Humble changelogs, I noticed that the fixes that you need have been introduced in Humble and not in Galactic.

xkaraman, 2022-10-24:
Yes i have tried to remove the CMake checks and it seems to build alright!

I can now alter the format of the image transport. But a crucial point is that i can only do that for the `base` image topic (this was the case before too, only that i couldnt alter the parameter). There is no parameters for the `compressed` topic, `compressedDepth` or any of the plugins!
As you can see there is no `zed_node.depth.depth_registered.compressed.format` ( To be honest i am coming from ROS1 and i am sure there was a different parameter for `base`, `compressed`, `compressedDepth`, don't know if it stands true for ROS2. Correct me if i am wrong) 

[image]

Myzhar, 2022-10-24:
Yes, you are right.
I can confirm that they are all available in Humble

xkaraman, 2022-10-24:
Ok then, so the best way forward for a Nvidia Jetson (latest version is based on 20.04 Ubuntu that humble is not available through official packaging) is through Docker Humble version i guess?