stereolabs/zed-ros2-wrapper #111: error building ZED component

mmilaprat, 2022-10-25:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I receive an error during the colcon build coment.

The previous zed-ros2-wrappe version works fine. I think that the version I tested las week was foxy-v3.7. I think that the last MR introduce something that doesn't works in my environment.

### Steps to Reproduce

1. colcon build --symlink-install

### Expected Result

the component was build

### Actual Result

I get errors like these when I try to build the componet

/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp: In member function ‘bool stereolabs::ZedCamera::startCamera()’:

/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:2689:31: error: ‘class sl::Camera’ has no member named ‘setRegionOfInterest’
 2689 |     sl::ERROR_CODE err = mZed.setRegionOfInterest(roi_mask);
          |                                                   ^~~~~~~~~~~~~~~~~~~

/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp: In member function ‘bool stereolabs::ZedCamera::startPosTracking()’:

/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:2922:15: error: ‘struct sl::PositionalTrackingParameters’ has no member named ‘depth_min_range’
2922 |   trackParams.depth_min_range = mPosTrackDepthMinRange;
         |                       ^~~~~~~~~~~~~~~

/ros2_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:2923:15: error: ‘struct sl::PositionalTrackingParameters’ has no member named ‘set_gravity_as_origin’; did you mean ‘set_floor_as_origin’?
 2923 |   trackParams.set_gravity_as_origin = mSetGravityAsOrigin;

### ZED Camera model

ZED2i

### Environment

```shell
OS: Ubuntu 22.04
ROS2 Version: Foxy
Python 3.8.8
```

Myzhar, 2022-10-25:
The latest version of the master branch requires the ZED SDK v3.8.x as specified in the README file.

mmilaprat, 2022-10-26:
it's true, now works, thanks

closed 2022-10-26 by Myzhar