stereolabs/zed-ros2-wrapper #112: Build Error in zed_components

swiz23, 2022-10-26:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I'm currently trying to build this package within the most recently published docker image (stereolabs/zed:3.8-devel-cuda11.7-ubuntu22.04), but I'm running into an issue during compilation. The error is as shown below.

```
--- stderr: zed_interfaces
* Found ROS2 humble
* ROS2 humble is officially supported by this package.
---
Finished <<< zed_interfaces [22.1s]
Starting >>> zed_components
Starting >>> rviz_plugin_zed_od
--- stderr: zed_topic_benchmark
* Found ROS2 humble
* ROS2 humble is officially supported by this package.
---
Finished <<< zed_topic_benchmark [4.83s]
--- stderr: zed_components
* Found ROS2 humble
* ROS2 humble is officially supported by this package.
In file included from /opt/colcon_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/include/zed_camera_component.hpp:28,
                 from /opt/colcon_ws/src/zed-ros2-wrapper/zed_components/src/zed_camera/src/zed_camera_component.cpp:25:
/opt/colcon_ws/src/zed-ros2-wrapper/zed_components/src/tools/include/sl_tools.h:50:10: fatal error: sl/Camera.hpp: No such file or directory
   50 | #include <sl/Camera.hpp>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
In file included from /opt/colcon_ws/src/zed-ros2-wrapper/zed_components/src/tools/src/sl_tools.cpp:25:
/opt/colcon_ws/src/zed-ros2-wrapper/zed_components/src/tools/include/sl_tools.h:50:10: fatal error: sl/Camera.hpp: No such file or directory
   50 | #include <sl/Camera.hpp>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/zed_camera_component.dir/build.make:90: CMakeFiles/zed_camera_component.dir/src/zed_camera/src/zed_camera_component.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[2]: *** [CMakeFiles/zed_camera_component.dir/build.make:76: CMakeFiles/zed_camera_component.dir/src/tools/src/sl_tools.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:164: CMakeFiles/zed_camera_component.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< zed_components [3.51s, exited with code 2]
Aborted  <<< rviz_plugin_zed_od [18.1s]

Summary: 7 packages finished [40.3s]
```
Any help would be much appreciated!

### Steps to Reproduce

1. docker pull stereolabs/zed:3.8-devel-cuda11.7-ubuntu22.04
2. Install ROS2 Humble
3. Follow instructions in the repo to build zed-ros2-wrapper from source

### Expected Result

Compiles successfully

### Actual Result

Fails compilation with the previously shown error message.

### ZED Camera model

ZED2i

### Environment

```shell
OS: Ubuntu 22.04 in Docker (Host is Ubuntu 18.04.6)
CPU: Intel® Core™ i7-10875H
GPU: NVIDIA GeForce GTX 1650 Ti/PCIe/SSE2
ZED SDK version: 3.8
ROS Distro: Humble
```

Myzhar, 2022-10-26:
Hi @swiz23 
What command did you use to start the docker container?

swiz23, 2022-10-26:
It's not even at that point yet. I wrote a custom dockerfile that builds upon the stereolabs one. Within the custom file, I pull in and build the ros2 wrapper. But it's failing.

swiz23, 2022-10-26:
Sure.
```
FROM stereolabs/zed:3.8-devel-cuda11.7-ubuntu22.04

SHELL ["/bin/bash", "-c"]

# STEP 1: Prepare environment for ROS installation
RUN apt-get update && apt-get install locales \
    && locale-gen en_US en_US.UTF-8 \
    && update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8 \
    && export LANG=en_US.UTF-8

RUN apt-get install --no-install-recommends -y curl gnupg2 lsb-release \
    && curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key  -o /usr/share/keyrings/ros-archive-keyring.gpg \
    && echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(source /etc/os-release && echo $UBUNTU_CODENAME) main" | sudo tee /etc/apt/sources.list.d/ros2.list > /dev/null

# STEP 2: Install ROS Humble and dev tools
RUN apt-get update && apt-get install --no-install-recommends -y \
    ros-humble-ros-base \
    ros-humble-rmw-cyclonedds-cpp \
    ros-humble-rqt* \
    git \
    libopencv-dev \
    libusb-1.0-0-dev \
    nano \
    vim \
    python3-colcon-common-extensions \
    python3-rosdep

# STEP 3: Configure ROS environment
RUN echo "source /opt/ros/humble/setup.bash" >> ~/.bashrc \
    && echo "source /usr/share/colcon_cd/function/colcon_cd.sh" >> ~/.bashrc \
    && echo "export _colcon_cd_root=/opt/ros/humble/" >> ~/.bashrc \
    && echo "source /usr/share/colcon_argcomplete/hook/colcon-argcomplete.bash" >> ~/.bashrc \
    && echo "export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp" >> ~/.bashrc \
    && echo "# export CYCLONEDDS_URI='<CycloneDDS><Domain><General><NetworkInterfaceAddress>enx747827ead180</NetworkInterfaceAddress></General></Domain></CycloneDDS>'" >> ~/.bashrc

# STEP 4: Build the Zed ROS Wrapper
WORKDIR /opt/colcon_ws/src

RUN git clone --recursive https://github.com/stereolabs/zed-ros2-wrapper.git \
    && git clone https://github.com/stereolabs/zed-ros2-examples.git

WORKDIR /opt/colcon_ws

RUN source /opt/ros/humble/setup.bash \
    && rosdep init \
    && rosdep update \
    && rosdep install --from-paths src --ignore-src -r -y \
    && colcon build --symlink-install --cmake-args=-DCMAKE_BUILD_TYPE=Release \
    && echo "source /opt/colcon_ws/install/local_setup.bash" >> ~/.bashrc \
    && source ~/.bashrc

LABEL ci-dockerfile="ros2_humble_zed/Dockerfile" \
      ros_distro="humble" \
      maintainer="Solomon Wiznitzer <shlomo.wiz@gmail.com>"
```

adujardin, 2022-10-27:
@swiz23 There was an issue on our side, we're deploying the fix. I'll let you know when the images are updated (you should "force-pull" them once done), it shouldn't take more than a few hours.

swiz23, 2022-10-27:
Hi @adujardin, I just pulled the latest image, and that error has gone away! However, I'm seeing a new error involving cuda (towards the end of the snippet below).
```
* ROS2 humble is officially supported by this package.
---
Finished <<< zed_components [39.9s]
--- stderr: zed_components
* Found ROS2 humble
* ROS2 humble is officially supported by this package.
---
Starting >>> zed_wrapper
Starting >>> zed_rgb_convert
--- stderr: zed_wrapper
* Found ROS2 humble
* ROS2 humble is officially supported by this package.
In file included from /opt/colcon_ws/install/zed_components/include/zed_components/sl_tools.h:50,
                 from /opt/colcon_ws/install/zed_components/include/zed_components/zed_camera_component.hpp:28,
                 from /opt/colcon_ws/src/zed-ros2-wrapper/zed_wrapper/src/zed_wrapper.cpp:26:
/usr/local/zed/include/sl/Camera.hpp:822:20: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
  822 |             struct {
      |                    ^
/usr/local/zed/include/sl/Camera.hpp:975:20: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
  975 |             struct {
      |                    ^
/usr/local/zed/include/sl/Camera.hpp:1152:20: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
 1152 |             struct {
      |                    ^
/usr/local/zed/include/sl/Camera.hpp:1316:20: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
 1316 |             struct {
      |                    ^
/usr/local/zed/include/sl/Camera.hpp:1320:20: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
 1320 |             struct {
      |                    ^
/usr/local/zed/include/sl/Camera.hpp:1324:20: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
 1324 |             struct {
      |                    ^
/usr/local/zed/include/sl/Camera.hpp:1564:20: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
 1564 |             struct {
      |                    ^
/usr/local/zed/include/sl/Camera.hpp:1568:20: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
 1568 |             struct {
      |                    ^
/usr/local/zed/include/sl/Camera.hpp:1572:20: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
 1572 |             struct {
      |                    ^
/usr/local/zed/include/sl/Camera.hpp: In function ‘sl::Timestamp sl::operator+(sl::Timestamp, const sl::Timestamp&)’:
/usr/local/zed/include/sl/Camera.hpp:2032:16: warning: implicitly-declared ‘constexpr sl::Timestamp::Timestamp(const sl::Timestamp&)’ is deprecated [-Wdeprecated-copy]
 2032 |         return lhs;
      |                ^~~
/usr/local/zed/include/sl/Camera.hpp:1971:20: note: because ‘sl::Timestamp’ has user-provided ‘sl::Timestamp& sl::Timestamp::operator=(sl::Timestamp)’
 1971 |         Timestamp& operator=(Timestamp other) {
      |                    ^~~~~~~~
/usr/local/zed/include/sl/Camera.hpp: In function ‘sl::Timestamp sl::operator-(sl::Timestamp, const sl::Timestamp&)’:
/usr/local/zed/include/sl/Camera.hpp:2037:16: warning: implicitly-declared ‘constexpr sl::Timestamp::Timestamp(const sl::Timestamp&)’ is deprecated [-Wdeprecated-copy]
 2037 |         return lhs;
      |                ^~~
/usr/local/zed/include/sl/Camera.hpp:1971:20: note: because ‘sl::Timestamp’ has user-provided ‘sl::Timestamp& sl::Timestamp::operator=(sl::Timestamp)’
 1971 |         Timestamp& operator=(Timestamp other) {
      |                    ^~~~~~~~
/usr/local/zed/include/sl/Camera.hpp: In function ‘sl::Timestamp sl::operator/(sl::Timestamp, const sl::Timestamp&)’:
/usr/local/zed/include/sl/Camera.hpp:2042:16: warning: implicitly-declared ‘constexpr sl::Timestamp::Timestamp(const sl::Timestamp&)’ is deprecated [-Wdeprecated-copy]
 2042 |         return lhs;
      |                ^~~
/usr/local/zed/include/sl/Camera.hpp:1971:20: note: because ‘sl::Timestamp’ has user-provided ‘sl::Timestamp& sl::Timestamp::operator=(sl::Timestamp)’
 1971 |         Timestamp& operator=(Timestamp other) {
      |                    ^~~~~~~~
/usr/local/zed/include/sl/Camera.hpp: In function ‘sl::Timestamp sl::operator*(sl::Timestamp, const sl::Timestamp&)’:
/usr/local/zed/include/sl/Camera.hpp:2047:16: warning: implicitly-declared ‘constexpr sl::Timestamp::Timestamp(const sl::Timestamp&)’ is deprecated [-Wdeprecated-copy]
 2047 |         return lhs;
      |                ^~~
/usr/local/zed/include/sl/Camera.hpp:1971:20: note: because ‘sl::Timestamp’ has user-provided ‘sl::Timestamp& sl::Timestamp::operator=(sl::Timestamp)’
[... 42 lines cut ...]
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: undefined reference to `cuvidCreateDecoder'
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: undefined reference to `cuCtxGetCurrent'
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: undefined reference to `cuMemAllocPitch_v2'
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: undefined reference to `cuvidCtxLockDestroy'
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: undefined reference to `cuvidCreateVideoParser'
/usr/bin/ld: /opt/colcon_ws/install/zed_components/lib/libzed_camera_component.so: undefined reference to `cuCtxGetDevice'
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: undefined reference to `cuInit'
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: undefined reference to `cuCtxCreate_v2'
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: undefined reference to `cuMemcpy2DUnaligned_v2'
/usr/bin/ld: /usr/local/zed/lib/libsl_zed.so: undefined reference to `cuStreamSynchronize'
[... cut ...]
gmake[1]: *** [CMakeFiles/Makefile2:164: CMakeFiles/zed_wrapper.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< zed_wrapper [7.86s, exited with code 2]
Aborted  <<< zed_rgb_convert [12.5s]

Summary: 9 packages finished [1min 12s]
  1 package failed: zed_wrapper

```
I found that by adding `--cmake-args=-DCMAKE_CXX_FLAGS="-Wl,--allow-shlib-undefined"` to the colcon build command fixed this issue. I don't know if both of those flags are needed or just one. But maybe this should be documented somewhere?

swiz23, 2022-10-27:
Ah, cool! So I guess i only needed to ensure the cxx_flags were set, but it wasn't necessary to set the DCMAKE_LIBRARY_PATH for things to compile successfully.

closed 2022-10-27 by swiz23

swiz23, 2022-10-27:
Thanks for all the help and the quick response!